MichaIng/DietPi #5623: Tailscale crashes after  'apt-get -y update' after uninstall.

SlowRaid, 2022-07-14:
Hi, I'm new to Linux, and Linux terminal, config etc. I've been lurking on the forum and here for a while. In fact, I just started  a course in network engineering, and I don't  think I would have had the confidence to do so if I hadn't played around with DietPi. 

So a huge thanks to the devs, and everyone else who are involved with the project. 

### Creating a bug report/issue

#### Required Information

#### Details:
- Date           | Thu Jul 14 16:49:54 AEST 2022
- DietPi version | v8.6.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:           39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y update`
- Exit code      | 100
- Software title | DietPi-Software

#### Additional Information (if applicable)
- **Software title** | Tailscale
- **Was the software title installed freshly or updated/migrated?**
- I had Tailscale installed, but Tailscale was recently uninstalled using Dietpi-Software uninstall feature.
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1.  apt-get -y update
2. Err:8 https://pkgs.tailscale.com/stable/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 458CA832957F5868

#### Expected behaviour

-... Sunshine and happy fun time?

#### Actual behaviour

- Tailscale crashes on  'apt-get -y update' after uninstall. 

#### Extra details
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://download.docker.com/linux/debian bullseye InRelease
Hit:6 https://packages.grafana.com/oss/deb stable InRelease
Hit:7 https://repos.influxdata.com/debian bullseye InRelease
Get:8 https://pkgs.tailscale.com/stable/debian bullseye InRelease
Hit:9 https://archive.raspberrypi.org/debian bullseye InRelease
Err:8 https://pkgs.tailscale.com/stable/debian bullseye InRelease 
The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 458CA832957F5868
Reading package lists...
W: GPG error: https://pkgs.tailscale.com/stable/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 458CA832957F5868
E: The repository 'https://pkgs.tailscale.com/stable/debian bullseye InRelease' is not signed.
```

- ...

Joulinar, 2022-07-14:
Did you uninstalled Tailscale using `dietpi-software`? If yes, usually the soruce file should have been removed as well.

SlowRaid, 2022-07-14:
> Did you uninstalled Tailscale using `dietpi-software`? If yes, usually the soruce file should have been removed as well.

Hi Joulinar, yes I did use the uninstallation feature in dietpi-software.

Joulinar, 2022-07-14:
do following

```
rm /etc/apt/sources.list.d/dietpi-tailscale.list
apt update
```

SlowRaid, 2022-07-14:
'sudo rm dietpi-tailscale.list' fixed it. Thanks Joulinar!

MichaIng, 2022-07-14:
Fixed with #5624

MichaIng, 2022-07-14:
Live patch merged as well, available via `dietpi-update`.