MichaIng/DietPi #5630: Home assistant upgrade fail

maury77, 2022-07-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.6.1`
- Distro version | `bullseye 1`
- Kernel version | `Linux DietPi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux`

#### Additional Information (if applicable)
- Software title Home assistant upgrade from 2022.5 to 2022.7

#### Steps to reproduce

**/home/homeassistant/homeassistant-update.sh**

#### Actual behaviour
 Building wheels for collected packages: maturin
        Building wheel for maturin (pyproject.toml): started
        Building wheel for maturin (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Building wheel for maturin (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [76 lines of output]
            running bdist_wheel
            running build
            running build_py
            creating build
            creating build/lib.linux-armv6l-3.9
            creating build/lib.linux-armv6l-3.9/maturin
            copying maturin/import_hook.py -> build/lib.linux-armv6l-3.9/maturin
            copying maturin/__init__.py -> build/lib.linux-armv6l-3.9/maturin
            running build_ext
             Downloading crates ...
            error: failed to download `scroll v0.11.0`
      
            Caused by:
              unable to get packages from source
      
            Caused by:
              failed to parse manifest at `/home/homeassistant/.cargo/registry/src/github.com-1285ae84e5963aae/scroll-0.11.0/Cargo.toml`
      
            Caused by:
              **feature `edition2021` is required**
      
              **The package requires the Cargo feature called `edition2021`, but that feature is not stabilized in this version of Cargo** (1.55.0 (32da73ab1 2021-08-23))
#### Extra details

- ...

MichaIng, 2022-07-17:
Many thanks for your report.

Building `maturin` requires Rust. The pre-compiled wheel from piwheels failed to build since end of last year, which is why it is tried to be compiled in place: https://github.com/piwheels/packages/issues/254

Your logs however indicate that Rust is installed already. Did you do that manually? If so how? The version seems to be too old, probably `apt install cargo`? In this case, let's try to install the latest Rust via rustup for the service user instead:
```sh
sudo -u homeassistant dash -c "$(curl -sSfL 'https://sh.rustup.rs')" -- -y --profile minimal
dietpi-software reinstall 157
```

`maturin` however does not seem to be required for the base HA installation, probably for some specific integration. Although, you faced this when calling the update script, which should only update dependencies of the `homeassistant` module. Can you show the whole update of `/home/homeassistant/homeassistant-update.sh`, please?

maury77, 2022-07-18:
Yes I had tried to install it manually, after noticing the error.

I have now tried your procedure with reainstallation and finished compiling with rust

It is not clear to me which component needs rust
thanks

MichaIng, 2022-07-18:
`maturin` seems to be more a developer tool to build and publish Python module with Rust bindings and -binaries: https://pypi.org/project/maturin/

Indeed it doesn't look like something that would be required for HA, but it was pulled into your HA's pyenv environment.

Which integrations have you installed?

fpMiguel, 2022-08-05:
I have encountered the same problem when reinstalling Home Assistant on ``DietPi v8.7.1``. I had previously installed Home Assistant (and it was upgraded to the latest version) but needed a fresh start.
I uninstalled Home Assistant and removed the ``/home/homeassistant/`` folder.
When trying to install it again it was complaining about ``maturin``.
So no additional HA integrations (to the best of my knowledge it should be a clean install)

I have followed @MichaIng's recommendation to install Rust (executed this in a different terminal at the start of the homeassistant install procress)
> ```shell
> sudo -u homeassistant dash -c "$(curl -sSfL 'https://sh.rustup.rs')" -- -y --profile minimal
> ```

Which means my **Raspberry Pi 1 Model B+** has been compiling for almost a day, obviously hitting the system limits.
I've also added the [consoleoutput](https://github.com/MichaIng/DietPi/files/9264954/consoleoutput.log)

fpMiguel, 2022-08-05:
Doing a [normal standalone installation](https://www.home-assistant.io/installation/raspberrypi/#install-home-assistant-core) also throws errors for `maturin`
```
Collecting PyJWT==2.4.0
  Downloading https://www.piwheels.org/simple/pyjwt/PyJWT-2.4.0-py3-none-any.whl (18 kB)
Collecting cryptography==36.0.2
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-36.0.2-cp39-cp39-linux_armv6l.whl (2.7 MB)
     |████████████████████████████████| 2.7 MB 52 kB/s
Collecting orjson==3.7.8
  Downloading orjson-3.7.8.tar.gz (640 kB)
     |████████████████████████████████| 640 kB 1.3 MB/s
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /srv/homeassistant/bin/python3 /srv/homeassistant/lib/python3.9/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-fvx19n9n/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple/ -- 'maturin>=0.13,<0.14'
       cwd: None
  Complete output (55 lines):
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/, https://www.piwheels.org/simple/
  Collecting maturin<0.14,>=0.13
    Downloading maturin-0.13.1.tar.gz (143 kB)
    Installing build dependencies: started
    Installing build dependencies: still running...
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting tomli>=1.1.0
    Using cached https://www.piwheels.org/simple/tomli/tomli-2.0.1-py3-none-any.whl (12 kB)
  Building wheels for collected packages: maturin
    Building wheel for maturin (PEP 517): started
    Building wheel for maturin (PEP 517): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /srv/homeassistant/bin/python3 /tmp/tmpso9en6nq_in_process.py build_wheel /tmp/tmp1y4k9khr
         cwd: /tmp/pip-install-9jlly_7l/maturin_bdbf862a686546b09eb6f48991ca1765
    Complete output (32 lines):
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-armv6l-cpython-39
    creating build/lib.linux-armv6l-cpython-39/maturin
    copying maturin/import_hook.py -> build/lib.linux-armv6l-cpython-39/maturin
    copying maturin/__main__.py -> build/lib.linux-armv6l-cpython-39/maturin
    copying maturin/__init__.py -> build/lib.linux-armv6l-cpython-39/maturin
    running egg_info
    creating maturin.egg-info
    writing maturin.egg-info/PKG-INFO
    writing dependency_links to maturin.egg-info/dependency_links.txt
    writing requirements to maturin.egg-info/requires.txt
    writing top-level names to maturin.egg-info/top_level.txt
    writing manifest file 'maturin.egg-info/SOURCES.txt'
    reading manifest file 'maturin.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'maturin.egg-info/SOURCES.txt'
    running build_ext
    running build_rust
    error: can't find Rust compiler

    If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.

    To update pip, run:

        pip install --upgrade pip

    and then retry package installation.

    If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
    ----------------------------------------
    ERROR: Failed building wheel for maturin
  Failed to build maturin
  ERROR: Could not build wheels for maturin which use PEP 517 and cannot be installed directly
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/ac/4a/43daa65b7ed984612267983039a58703c48ccbc62561abb43ec7bc8b0663/orjson-3.7.8.tar.gz#sha256=a2e824220245323bb3291bb10ccf2ba936ed0b14e5e4a6260a1d1ed048caf77e (from https://pypi.org/simple/orjson/) (requires-python:>=3.7). Command errored out with exit status 1: /srv/homeassistant/bin/python3 /srv/homeassistant/lib/python3.9/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-fvx19n9n/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple/ -- 'maturin>=0.13,<0.14' Check the logs for full command output.
Collecting homeassistant
  Downloading https://www.piwheels.org/simple/homeassistant/homeassistant-2022.8.0-py3-none-any.whl (18.5 MB)
     |████████████████████████████████| 18.5 MB 12 kB/s
  Downloading https://www.piwheels.org/simple/homeassistant/homeassistant-2022.7.7-py3-none-any.whl (18.2 MB)
     |████████████████████████████████| 18.2 MB 25 kB/s
Collecting pip<22.2,>=21.0
  Downloading https://www.piwheels.org/simple/pip/pip-22.1.2-py3-none-any.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 1.5 MB/s
Collecting orjson==3.7.5
  Downloading orjson-3.7.5.tar.gz (640 kB)
     |████████████████████████████████| 640 kB 1.5 MB/s
  Installing build dependencies ... |
```

However, as you can see, it does continue the installation process.
I'll update this comment to indicate if it is succesfull or not.

Joulinar, 2022-08-05:
There are a couple of reports on the HA forum related to issues with `maturin`. Could it have become part of the standard package?

MichaIng, 2022-08-06:
Hmm, I get now other errors as if required dependencies are not defined for the homeassistant Python module:
```
ModuleNotFoundError: No module named 'sqlalchemy'
...
ModuleNotFoundError: No module named 'fnvhash'
```
After installing these manually into pyenv, it starts.

https://github.com/home-assistant/core/issues/76296

But next is that first run setup on web UI doesn't work. Last setup step fails with "Failed to save: undefined" error in browser. Also selecting something for each field doesn't help.

MichaIng, 2022-08-06:
However, after manually installing those, `maturin` is not installed, so it is still not part of HA core. @maury77 which integrations do you have enabled?

messense, 2022-08-21:
`maturin` developer here, so the issue is that the `orjson` dependency does not have `armv6l` wheels on pypi and [piwheels](https://github.com/piwheels/packages/issues/237).

For `maturin` I've manually uploaded a `linux_armv6l` wheel to pypi in [v0.13.2](https://pypi.org/project/maturin/0.13.2/#files), hope it helps.

MichaIng, 2022-09-10:
Many thanks. The other issues has been fixed as well in the meantime. Hope the `orjson` wheel built on piwheels will be fixed soon as well, to have 32-bit ARM wheels available also after updates.

However, I'll mark it as closed here.

closed 2022-09-10 by MichaIng

JzBrevis, 2023-11-19:
Having this issue on Android

YarikMsafiri, 2024-01-05:
> Having this issue on Android

Same in pydroid 3