MichaIng/DietPi #5633: set GPIO (for specified PIN) level on boot (or power on)

almirus, 2022-07-18:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I want to set pin 37 (GPIOAO_9) when booting into HIGH level. There is connected switcher for HDMI display (default is OFF) and not showing booting process.

MichaIng, 2022-07-22:
Which SBC and Debian version are you using?

almirus, 2022-07-22:
radxa zero 
Bullseye

MichaIng, 2022-07-31:
Please try with libgpio:
```sh
apt install gpiod
```

almirus, 2022-08-01:
yes
but how to set gpio before load OS (or while booting)?

WolfganP, 2022-08-01:
> yes but how to set gpio before load OS (or while booting)?

Before loading the OS you obviously have almost nothing running other than the initialization firmware of your board.
OS booting is a multistep process, so "while booting" means you need to define exactly at which step your gpio action is needed (ie define precedence, need it before X Y and Z occurs).
Did you try issuing the gpio command in /etc/rc.local or via a custom systemd entry?

MichaIng, 2022-09-18:
udev rules should be a clever way to assign GPIO states as fast as the device is detected by the kernel.

almirus, 2022-09-18:
@MichaIng hello, do you have any example for this?

MichaIng, 2022-09-18:
Like so, setting GPIO 21 of chip 0, as fast as GPIO chip 0 is added/detected:
```sh
echo 'SUBSYSTEM=="gpio", KERNEL=="gpiochip0", ACTION=="add", RUN+="/usr/bin/gpioset gpiochip0 21=1"' > /etc/udev/rules.d/99-gpio.rules
```

MichaIng, 2022-11-24:
Did this work for you?

MichaIng, 2022-12-30:
Marking as closed due to outstanding reply. Please reopen when you have feedback on the suggested udev rule.

closed 2022-12-30 by MichaIng