MichaIng/DietPi #5634: Odroid booting from SSD only

hstarcm, 2022-07-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.123-meson64 #22.05.3 SMP PREEMPT Wed Jun 22 07:23:04 utc 2022 aarch64 GNU/Linux
- SBC model | Odroid N2 (aarch64)
- Power supply used | Official HardKernel 12V2A 
- SD card used | None
- SSD used: Inland Professional SATA 480GB with UGreen USB 3.0 to Sata adapter

#### Additional Information (if applicable)
- Petitboot  dev 20220317
- Was the software title installed freshly or updated/migrated? | Fresh
- Can this issue be replicated on a fresh installation of DietPi? | Yes
<-- If you sent a "dietpi-bugreport", please paste the ID here -->
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
I just bought Odroid N2+ and I am trying all kinds of way to try to boot dietpi from SSD only without SD or eMMC. I managed to boot from SSD when in Petitboot editor by editing the following boot options for the SSD partition:

```
Device tree: /usr/var/petitboot/mnt/dev/sda1/boot/dtb-5.10.123-meson64/amlogic/meson-g12b-odroid-n2-plus.dtb
Boot arguments: root=/dev/sda1 (leave the rest of the parameters untouched)
```

The problem is these parameters can only work one time. I have to modify the Petitboot every time when it boot up. Is there any way or option to get the correct parameters from the boot.scr or save these options in Petitboot without using SD or eMMC?

Thank 

I read that dietpi does not support direct booting from SSD for Odroid N2+ but i hope this issue can be resolved easily.

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2022-07-22:
I need to play around with this. Best IMO would be if the U-Boot from the image itself would allow booting from SSD. Otherwise, if I'm not mistaken, our U-Boot config should work with latest petitboot when leaving out one of the variables. I'll have a look when I find time.

hstarcm, 2022-07-24:
Thanks for the reply and support.  

If I modify the boot.cmd in /boot
```
setenv rootdev "/dev/sda1"
setenv fdtfile "amlogic/meson-g12b-odroid-n2-plus.dtb"
```

and save the file. And then run the command
```
mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
```
In this case, it can boot up directly from SSD using petitboot.

This is for your information. Hope it will be helpful. 

Additional information:
I try using uboot to boot from SSD but it does not work.
I also try using Armbian 22.05 Jammy to boot using Uboot or Petitboot and it does not work in its original unmodified image.

MichaIng, 2022-07-25:
So `fdtfile` is not set or not set as expected (filename only, based on N2 or N2+) by petitboot. I wonder whether it works without setting it if the dtbs are left in /usr/lib default path.

MichaIng, 2022-09-26:
@Power-onoff
Pinging you here where your post fits better: USB boot did only work with the boot switch set to SPI previously, right? The MMC toggle should only allow eMMC and SD card boot. So the issue is probably the same as reported here, that petitboot (from SPI) does not work since it does not set the `fdtfile` variable, or sets it differently.

However, it sets the `variant` variable, I remember, as printed on your bootloader output: https://github.com/hardkernel/u-boot/blob/545adf4/board/hardkernel/odroidn2/odroidn2.c#L419
We can use it to set `fdtfile` correctly.

And we can test to link the `dtb` directory to the standard path below `/usr/lib` and see whether petitboot then has the `fdtfile` set correctly already.

Guys, can you please try this:
```sh
sed -i '/^setenv overlay_prefix/a\if test "${variant}" = "n2_plus"; then setenv fdtfile "amlogic/meson-g12b-odroid-n2-plus.dtb"; fi' /boot/boot.cmd
sed -i '/^setenv overlay_prefix/a\if test "${variant}" = "n2-plus"; then setenv fdtfile "amlogic/meson-g12b-odroid-n2-plus.dtb"; fi' /boot/boot.cmd
sed -i '/^setenv overlay_prefix/a\if test "${variant}" = "n2"; then setenv fdtfile "amlogic/meson-g12b-odroid-n2.dtb"; fi' /boot/boot.cmd
sed -i '/^setenv overlay_prefix/a\setenv fdtfile_orig "${fdtfile}"' /boot/boot.cmd
sed -i '/^setenv bootargs/s/"$/ fdtfile_orig=${fdtfile_orig}"; fi' /boot/boot.cmd
mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
```
If it boots:
```sh
cat /proc/cmdline
```

hstarcm, 2022-10-04:
Executing this command
`sed -i '/^setenv bootargs/s/"$/ fdtfile_orig=${fdtfile_orig}"; fi' /boot/boot.cmd`
show this error
```
sed: -e expression #1, char 56: unterminated `s\' command
```
I am noob at linux so please advise how to solve the error?

MichaIng, 2022-10-04:
Mistake my end, please use:
```sh
sed -i '/^setenv bootargs/s/"$/ fdtfile_orig=${fdtfile_orig}"/' /boot/boot.cmd
```

hstarcm, 2022-10-05:
[image]

As you can see above, it boot halfway and stop.

Power-onoff, 2022-10-07:
> 

I don't quite understand your answer

What I'm trying to do is not a USB boot

It's just booting with a MicroSD card

If you connect an external HDD after booting with a MicroSD card, it can recognize and mount normally

However, if an external HDD is connected while booting with a MicroSD card, the following error occurs

[image]

Power-onoff, 2022-10-07:
> @Power onoff 게시물이 더 잘 맞는 곳에서 핑 핑 : USB 부팅은 이전에 SPI로 설정된 부팅 스위치에서만 작동했습니다. MMC 토글은 eMMC 및 SD 카드 부팅만 허용해야 합니다. 따라서 문제는 아마도 여기에보고 된 것과 같을 수 있으며, petitboot (SPI에서)는 변수를 설정하지 않거나 다르게 설정하지 않기 때문에 작동하지 않습니다.`fdtfile`
> 
> 그러나 부트 로더 출력에 인쇄 된대로 변수를 설정합니다 : https://github.com/hardkernel/u-boot/blob/545adf4/board/hardkernel/odroidn2/odroidn2.c#L419 올바르게 설정하는 데 사용할 수 있습니다.`variant``fdtfile`
> 
> 그리고 디렉토리를 아래의 표준 경로에 링크하고 petitboot가 이미 올바르게 설정되었는지 여부를 테스트 할 수 있습니다.`dtb``/usr/lib``fdtfile`
> 
> 얘들 아, 너는 이것을 시도 할 수있다 :
> 
> ```shell
> sed -i '/^setenv overlay_prefix/a\if test "${variant}" = "n2_plus"; then setenv fdtfile "amlogic/meson-g12b-odroid-n2-plus.dtb"; fi' /boot/boot.cmd
> sed -i '/^setenv overlay_prefix/a\if test "${variant}" = "n2-plus"; then setenv fdtfile "amlogic/meson-g12b-odroid-n2-plus.dtb"; fi' /boot/boot.cmd
> sed -i '/^setenv overlay_prefix/a\if test "${variant}" = "n2"; then setenv fdtfile "amlogic/meson-g12b-odroid-n2.dtb"; fi' /boot/boot.cmd
> sed -i '/^setenv overlay_prefix/a\setenv fdtfile_orig "${fdtfile}"' /boot/boot.cmd
> sed -i '/^setenv bootargs/s/"$/ fdtfile_orig=${fdtfile_orig}"/' /boot/boot.cmd
> mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
> ```
> 
> 부팅되는 경우:
> 
> ```shell
> cat /proc/cmdline
> ```

> 

I don't quite understand your answer

What I'm trying to do is not a USB boot

It's just booting with a MicroSD card

If you connect an external HDD after booting with a MicroSD card, it can recognize and mount normally

However, if an external HDD is connected while booting with a MicroSD card, the following error occurs

[image]

Power-onoff, 2022-10-07:
> [image]
> 
> As you can see above, it boot halfway and stop. i also run `cat /proc/cmdline` for your reference

I don't know why, but I failed to boot directly from an external HDD without a MicroSD card or eMMC It also failed to change the root file system to an external HDD after booting to a MicroSD card or eMMC (I copied the image after fdisk operations to avoid duplicate UUIDs)

Therefore, an attempt was made to boot from an external HDD using petitboot from SPI flash

As a result, I succeeded

I don't know why, but it took nearly a minute for petitboot to recognize the external HDD After recognition, there was a symptom of a momentary USB power-off during boot (probably a USB driver initialization), so it took another minute to recognize during boot, and the phrase "Begin: running /script/local-block ... disn" was repeated and stopped as shown in the picture

I added "rootdelay=120" to "bootargs" and it successfully booted

MichaIng, 2022-10-07:
Ah okay, then it's a different issue.

MichaIng, 2022-11-10:
I found time to play around with this. First I updated petitboot to latest version (if it wasn't already):
```
U-Boot 2015.01 (Mar 17 2022 - 12:57:50)
```
Adding the [above](https://github.com/MichaIng/DietPi/issues/5634#issuecomment-1258440165) U-Boot settings at least does not break boot with MMC switch, so it's generally valid and correct. However, I'm unable to really test this with SPI switch. Not sure why, but it doesn't read from the USB stick I use for testing (it has an LED which flashes in I/O) until reaching a state where it waits for wake-on-lan. The USB stick LED flashes after this:
<details>

```
G12B:BL:6e7c85:2a3b91;FEAT:E0F83180:402000;POC:B;RCY:0;SPINOR:0;0.▒!,K▒▒х▒▒}▒▒▒с0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

L0:00000000
L1:00000703
L2:0000c067
L3:14000020
B2:00402000
B1:e0f83180

TE: 58141

BL2 Built : 06:17:13, Jun 28 2019. g12b gf0505d7-dirty - qi.duan@droid13

Board ID = 5
Set A53 clk to 24M
Set A73 clk to 24M
Set clk81 to 24M
A53 clk: 1200 MHz
A73 clk: 1200 MHz
CLK81: 166.6M
smccc: 00012af9
DDR driver_vesion: LPDDR4_PHY_V_0_1_14 build time: Jun 28 2019 06:17:09
board id: 5
Load FIP HDR from SPI, src: 0x00010000, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SPI, src: 0x00030000, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SPI, src: 0x0002c000, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from SPI, src: 0x00014000, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from SPI, src: 0x00020000, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

R0_RxClkDly_Margin==94 ps 8
R0_TxDqDly_Margi==118 ps 10

R1_RxClkDly_Margin==0 ps 0
[... 389 lines cut ...]
i2c 0x01 read failed
[aml_i2c_xfer] error ret = -5   i2c master a current slave addr is 0x32
i2c 0x02 read failed
Read Chip : 0x0, 0x0, 0x0
failed to read LT8619C chip id
 lt8619c hdmi_to_lcd device not found.!
co-phase 0x3, tx-dly 0, clock 400000
co-phase 0x3, tx-dly 0, clock 400000
co-phase 0x3, tx-dly 0, clock 400000
emmc/sd response timeout, cmd8, status=0x1bf2800
emmc/sd response timeout, cmd55, status=0x1bf2800
emmc/sd response timeout, cmd1, status=0x1bf2800
Net:   dwmac.ff3f0000
Hit Enter or space or Ctrl+C key to stop autoboot -- :  0
SF: Detected XT25Q64 with page size 256 Bytes, erase size 4 KiB, total 8 MiB
SF: 7274496 bytes @ 0x110000 Read: OK
### CRAMFS LOAD ERROR<0> for boot.ini!
## Executing script at 10000000
Wrong image format for "source" command
odroidn2#
```
</details>
Ah the same (second) output now is done with USB or eMMC attached as well. Looks like the petitboot recovery + upgrade loop I did broke something. However, since first the older 2021-11-12 petitboot was flashed successfully, shown as such, and able to self-upgrade with upgrade image to 2022-03-17 (as shown above), it seems to have been all successful. Not sure what changed, probably the internal kernel/initrd/boot.ini was stored on "CRAMFS" previously but erased with the petitboot recovery image. Regardless, in my case SPI/petitboot was not and is not able to even detect any other bootable device and read `/boot/boot.scr` from it 🤔.

If anyone has an idea what is wrong with my petitboot, please tell me, otherwise I've no idea how to proceed at this point. The actual OS image is not involved yet at this point.

MichaIng, 2022-11-10:
It boots again with recovery petitboot, but isn't able to self-update v2022-03-17 even when putting the `spiboot.img` onto the official Hardkernel Ubuntu image FAT partition and toggle to SPI as stated [here](https://wiki.odroid.com/odroid-n2/os_images/petitboot) and [here](https://wiki.odroid.com/getting_started/petitboot/recover_or_upgrade). This version has known issues to boot from eMMC... So still stuck in a way.

MichaIng, 2022-11-11:
Back on track with latest functioning petitboot. However finally failing on getting the `/boot/dietpiEnv.txt` loaded into U-Boot environment: https://forum.odroid.com/viewtopic.php?p=359155#p359155

MichaIng, 2022-12-04:
Linking for further info: https://dietpi.com/forum/t/loaded-kernel-remains-outdated-after-reboot/14880/29
Alternative syntax to load a config file into U-Boot script: https://github.com/mdrjr/n2_bootini/blob/master/boot.ini
Would need to be tested with mainline U-Boot, and whether ini sections break loading it via `env import`.

MichaIng, 2023-01-19:
Further related info:
- https://dietpi.com/forum/t/odroid-hc4-booting-from-ssd/6322
- https://forum.odroid.com/viewtopic.php?p=364237#p364237

MichaIng, 2023-02-04:
Solved with: https://github.com/MichaIng/DietPi/commit/356c4af
I'll trigger new image builds. Let me review the result first, I'll inform you once they are ready for testing. I can also provide steps to enable this on your running system.

MichaIng, 2023-02-05:
Images are ready and boot fine here from USB flash drive. Odroid N2 and HC4 users are very welcome to test: https://dietpi.com/downloads/images/testing/

And as mentioned, if you want to add USB support to your running system image, let me know.

amibumping, 2023-02-06:
> Images are ready and boot fine here from USB flash drive. Odroid N2 and HC4 users are very welcome to test: https://dietpi.com/downloads/images/testing/
> 
> And as mentioned, if you want to add USB support to your running system image, let me know.

Great to hear these news @MichaIng!

It would be great to see those steps to modify our running system in order to avoid install all again.

Other thing, instead of USB can be used and SSD?, and is going to come back to us the "Move rootfs"?

Thank you as always!

Joulinar, 2023-02-06:
I guess a reinstall is needed as a migration of existing system will not work.

MichaIng, 2023-02-06:
Migration is possible as mentioned 🙂:
```sh
G_EXEC cp -a /boot/boot.scr{,.bak}
G_EXEC curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/U-Boot/boot.cmd' -o /boot/boot.cmd
G_EXEC sed -i '1i[main]' /boot/dietpiEnv.txt
(( $G_HW_MODEL == 15 )) && G_EXEC sed -i '/^setenv overlay_error/a\
\n# petitboot workarounds\
if test "${petitboot_active}" = "1"; then\
\tif test "${variant}" = "n2"; then\
\t\tsetenv fdtfile "amlogic/meson-g12b-odroid-n2.dtb"\
\telse\
\t\tsetenv fdtfile "amlogic/meson-g12b-odroid-n2-plus.dtb"\
\tfi\
\tsetenv rootuuid "true"\
fi' /boot/boot.cmd
(( $G_HW_MODEL == 16 )) && G_EXEC sed -i '/^setenv overlay_error/a\
\n# petitboot workarounds\
if test "${petitboot_active}" = "1"; then\
\tsetenv fdtfile "amlogic/meson-sm1-odroid-${variant}.dtb"\
\tsetenv rootuuid "true"\
fi' /boot/boot.cmd
G_EXEC sed -i '/env import/c\	if test "${petitboot_active}" = "1"; then ini main ${scriptaddr}; else env import -t ${scriptaddr} ${filesize}; fi' /boot/boot.cmd
G_EXEC sed -i '/test -e/s/fixup.scr;/fixup.scr || test "${petitboot_active}" = "1";/' /boot/boot.cmd
G_EXEC mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
```
Paste this as one code block. It does the adjustments depending on whether it is an N2 or C4/HC4 image.

Alternative, if you prefer doing it manually:
1. `/boot/dietpiEnv.txt` needs to have a new first line `[main]`
2. For Odroid N2, `/boot/boot.cmd` needs to look like this:
<details>

```
# DO NOT EDIT THIS FILE
#
# Please edit /boot/dietpiEnv.txt to set supported parameters
#
# If you must, edit /boot/boot.cmd and recompile /boot/boot.scr with:
# mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr

# Default environment
setenv rootdev "/dev/mmcblk0p1"
setenv rootfstype "ext4"
setenv consoleargs "console=tty1"
setenv verbosity "4"
setenv docker_optimizations "off"
setenv overlay_path "amlogic"
setenv overlay_prefix "meson"

# Load addresses
setenv scriptaddr "0x32000000"
setenv kernel_addr_r "0x34000000"
setenv fdt_addr_r "0x4080000"
setenv overlay_error "false"

# petitboot workarounds
if test "${petitboot_active}" = "1"; then
	if test "${variant}" = "n2"; then
		setenv fdtfile "amlogic/meson-g12b-odroid-n2.dtb"
	else
		setenv fdtfile "amlogic/meson-g12b-odroid-n2-plus.dtb"
	fi
	setenv rootuuid "true"
fi

# Load environment file
if load ${devtype} ${devnum} ${scriptaddr} ${prefix}dietpiEnv.txt; then
	if test "${petitboot_active}" = "1"; then ini main ${scriptaddr}; else env import -t ${scriptaddr} ${filesize}; fi
fi

[... 119 lines cut ...]
		load ${devtype} ${devnum} ${fdt_addr_r} ${prefix}dtb/${fdtfile}
	else
		if load ${devtype} ${devnum} ${scriptaddr} ${prefix}dtb/${overlay_path}/overlay/${overlay_prefix}-fixup.scr; then
			echo "Applying kernel provided DT fixup script ${overlay_prefix}-fixup.scr"
			source ${scriptaddr}
		fi
		if test -e ${devtype} ${devnum} ${prefix}fixup.scr || test "${petitboot_active}" = "1"; then
			if load ${devtype} ${devnum} ${scriptaddr} ${prefix}fixup.scr; then
				echo "Applying user provided fixup script fixup.scr"
				source ${scriptaddr}
			fi
		fi
	fi
fi

# Boot
booti ${kernel_addr_r} ${ramdisk_addr_r} ${fdt_addr_r}
```
</details>

4. Finally create a backup first and then rebuild `/boot/boot.scr`:
    ```sh
    cp -a /boot/boot.scr /boot/boot.scr.bak
    mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
    ```

amibumping, 2023-02-07:
@MichaIng can you explain a bit more the process?
We have to paste it as a script .sh and execute it? We need to have an usb drive inserted before executing it? If we have a usb hub with 4 usb ports, how we do select in which port is the usb drive?

Thank you!

MichaIng, 2023-02-07:
> We have to paste it as a script .sh and execute it?

You can copy&paste the code block into your console and hit ENTER. A bit more save is to copy it into a script file indeed, so you can better review that everything was pasted correctly. But then "source" it instead of executing it, since it uses our error handler:
```sh
source script.sh
```

> We need to have an usb drive inserted before executing it?

No, this won't copy the image to USB drive, it just enables it to be generally bootable from a USB drive. You'd still need to clone/dd it over. We have some guide here to cover that topic, if needed: https://dietpi.com/blog/?p=1236

> If we have a usb hub with 4 usb ports, how we do select in which port is the usb drive?

Since the script does not do the actual copy, the question is obsolete? The SPI bootloader by default loops through all attached USB drives and boots from the first bootable system it finds. So not need to define the USB port, at least if there is only one bootable system attached (including eMMC/SD which take priority). If you have multiple OS drives attached, it can be selected from the petitboot GUI. It shows up on the attached screen and would automatically boot from the first bootable drive after 10 seconds (by default), but you can also select a specific drives, which are listed in the GUI as well.

amibumping, 2023-02-07:
@MichaIng just the last doubt, when I installed latest image for HC4 I had to delete petitboot, now with the new image do i have to restore it in order to achieve usb boot, or is not mandatory?

MichaIng, 2023-02-07:
Yes petitboot has to be restored, or you can also flash mainline U-Boot to SPI, which was another way already to enable USB boot:
```sh
apt install linux-u-boot-odroidhc4-current
source /usr/lib/u-boot/platform_install.sh
write_uboot_platform_mtd "$DIR"
```
By default the C4 U-Boot package is installed which does not ship the SPI bootloader binary, hence the package install/replacement needed first. Generally the goal of petitboot support that one can boot DietPi from USB directly without the need to first hold the MASK key, erase or replace the bootloader and all such 🙂. So if you tested either our new image or your running image cloned to USB with petitboot restored, that would be best, since I have no HC4 to test with.

amibumping, 2023-02-09:
@MichaIng I did this steps:
I restore petitboot.
Make a clone of the SD with win32imager and flash it to an SSD through etcher.
I run the script.
I plug ssd to the USB port, and power on.
Petitboot starts to seek through 10 seconds and it says: Cannot read /tmp/usb0_usr_var_petitboot_mnt_dev_sda1___boot_dtb_

I cannot do it with cloning the SSD?

MichaIng, 2023-02-09:
Cloning the drive should work.

> Cannot read /tmp/usb0_usr_var_petitboot_mnt_dev_sda1___boot_dtb_

The boot script is obviously loaded, but it fails to read the device tree directory it seems. This should be a symbolic link to `/boot/dtb-<kernel_source>`. Can you check whether it is intact? With e.g. 7zip you can also browse the content of the img file.

amibumping, 2023-02-09:
> 

[image]

What do you need to see? dtb folder has content

MichaIng, 2023-02-10:
The UUID is also used, that is good as intended. Can you check whether it is somehow wrong? In petitboot menu I think it shows the UUID of all detected drives/filesystems as well.

amibumping, 2023-02-10:
> The UUID is also used, that is good as intended. Can you check whether it is correct? In petitboot menu I think it shows the UUID of all detected drives/filesystems as well.

Yes, I check it in dietpi drive manager on the old dietpi installation, and the uuid it's the same..

MichaIng, 2023-02-10:
> and that I have to select something on boot to SPI

It shouldn't be needed to select or change anything. By default, all drive types is looped through (eMMC > SD > USB) and the first valid one picked for autoboot after 10 seconds. This seems to work correctly since your output shows that the adjusted `/boot/boot.scr` is picked.

Also the kernel and initramfs has been loaded successfully, so U-Boot (the one invoked by petitboot) found the filesystem. Now with everything loaded, it is on the initramfs to actually mount the root filesystem, and this seems to fail.

There is this USB "connect-debounce failed" error which seems to be related. I wanted to ask whether you could try another USB port, but the HC4 has one only 😄.

Could you try booting from SD card again, connect the USB drive (no need to mount), and run `update-initramfs -u`. Then copy the `/boot/initrd-*` and `/boot/uInitrd-*` files over to the USB drive. At best mount temporarily via `mount /dev/sda1 /media` or so. There might be issues since it has the same UUID than the SD card.

Also the kernel is on latest version, right? Should be `6.0.13` currently:
```sh
uname -r
```

amibumping, 2023-02-10:
Nothing same output:
[image]

I also test booting with the microSD in, and I get the same result, I thought this way it would boot:
[image]

Kernel version is 6.1.7

MichaIng, 2023-02-10:
> Kernel version is 6.1.7

Ah "edge" kernel package then. However, should work regardless.

From the initramfs prompt, can you show the output of these commands:
```sh
cat /proc/cmdline
ls /dev/disk/by-uuid/
```

> I also test booting with the microSD in, and I get the same result, I thought this way it would boot:

Indeed I also expected that. Instead of the USB error, now there is one related to SD card:
> mmc0: error -110 whilst initialising SD card

Then I guess something with the boot config is the problem. Probably the `${variant}` variable is not set to what I expected. To check that, you can edit `/boot/boot.cmd`, add `${variant}` separated with a space to the `setenv bootargs` value, so it looks like this
```cmd
setenv bootargs "root=${rootdev} rootfstype=${rootfstype} rootwait ${consoleargs} loglevel=${verbosity} consoleblank=0 coherent_pool=2M usb-storage.quirks=${usbstoragequirks} ${extraargs} ${variant}"
```
and compile it again:
```sh
mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
```
and from initramfs or also when it boots from SD card:
```sh
cat /proc/cmdline
```

amibumping, 2023-02-11:
Cat command with microSD only inserted after adding $variant
[image]

Then boot with microSD:
[image]

This is cat & ls commands with SSD and 2 hard drives attached:
[image]

This is the log before petitboot starts, just after the hardkernel log splash with the SSD connected:
[image]

MichaIng, 2023-02-11:
You did run the `cat` command within the petitboot console. You'd need to wait until you land in the `(initramfs)` console and run the commands there 😉.

amibumping, 2023-02-11:
Mm.. I think it didn't let me write... will test...

MichaIng, 2023-02-11:
Did, input works on petitboot shell but not in initramfs shell. Strange, and bad. Okay this was only to fine out whether the `${variant}` variable is set to `hc4` on HC4 as expected. We can also set the correct device tree directly. We can do this anyway since Odroid C4 has no SPI flash.

In `/boot/dietpiEnv.txt`, add a new line:
```sh
fdtfile=amlogic/meson-sm1-odroid-hc4.dtb
```

If this does not work I think we need to wait for some more feedback to know whether this affects all HC4 systems. I hope the HC4 petitboot is not somehow very different from the N2 one.

amibumping, 2023-02-12:
> In /boot/dietpiEnv.txt, add a new line:
> 
> fdtfile=amlogic/meson-sm1-odroid-hc4.dtb

Nothing same output, after making those changes I made:
`mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr`
I thought I have to..

> EDIT: Just in case: With MASKROM button and SD card it does boot, right?
Yes, if I press the button at the botom it does boot

Is it safe is I update to 8.14.2?

MichaIng, 2023-02-12:
> I thought I have to..

Only needed when doing changes to `/boot/boot.cmd`.

> PD.: Is it safe is I update to 8.14.2?

We wouldn't have released it if we wouldn't think that 😃.

Okay I'm sadly out of ideas for now what the problem is. It is as if some kernel driver is missing to allow the initramfs detecting the SSD (via UUID), but the same for SD card (when booting via petitboot). But since the same kernel, initramfs and device tree are loaded for sure, I've not idea what is different. Sadly we cannot check the cmdline, as long as input at the initramfs console does not work. And since petitboot is suppressing any U-Boot output we can also not just echo it to the console.

Ah, but what may fix it in your case, if using UUIDs here is the problem, not the device drivers themselves:
- Add `rootuuid=false` to `/boot/dietpiEnv.txt`.
- This is the default while we explicitly set it to `true`, because otherwise it cannot boot from SD card or eMMC (via petitboot) as it mixes up `/dev/mmcblk0` and `/dev/mmcblk1`. However, since you want to boot from USB, this doesn't matter.

amibumping, 2023-02-12:
[image]

Nothing... Maybe I am doing something wrong.

MichaIng, 2023-02-12:
At least we see again that the script and config is used and respected, as now it shows `/dev/sda1` as root drive, correctly, but the initramfs still fails to find it...

colafloat25, 2023-02-18:
> [image]
> 
> Nothing... Maybe I am doing something wrong.

I am also having this problem on N2+ same can't input anything and stuck on initramfs

MichaIng, 2023-02-18:
@colafloat25 
Did you try to convert your image or used a fresh one? Because based on your output `setenv rootuuid true` is not set. This alone shouldn't make a different as long as you do not try with SD card or eMMC, but other parts of the config might then be missing/wrong.

Also assure to have latest petitboot, and while obviously first boot stages work, please loop through petitboot settings, assure that "any device" is enabled as boot device.

And test with all USB ports and a different PSU and/or cable if you have them.

Since it works very well here on my Odroid N2+, the only other left difference I can think of is the PCB version. Mine is `REV 0.6 20210121`.

colafloat25, 2023-02-18:
> @colafloat25 Did you try to convert your image or used a fresh one? Because based on your output `setenv rootuuid true` is not set. This alone shouldn't make a different as long as you do not try [...]

I used the fresh one and I just install the image directly to external SSD ( No Micro-SD or eMMC ) and I did pb-update before  I plugin the external ssd and I got the screen like  @amibumping

MichaIng, 2023-03-11:
It seems to mostly work now, with some more changes done (incorporated [above](https://github.com/MichaIng/DietPi/issues/5634#issuecomment-1419842361)). It is not 100% reliable, even normal MMC mode boot from some SD card variants, but it's best we can currently do and definitely an improvement to before. Marking this as closed. Individual problems can be tracked in dedicated issues.

closed 2023-03-11 by MichaIng