MichaIng/DietPi #5635: dietpi-update failed

omavoss, 2022-07-20:
#### Details:
- Date           | Wed Jul 20 18:38:41 CEST 2022
- Bug report     | afdae904-85b7-4618-83cd-3a2b39e8fad6
- DietPi version | v8.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux RasPi-3 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  base-files bash dirmngr dpkg firmware-atheros firmware-brcm80211
  firmware-iwlwifi firmware-misc-nonfree firmware-realtek gnupg gnupg-l10n
  gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm gpgv
  libavcodec58 libavformat58 libavutil56 libc-bin libc-l10n libc6 libfreetype6
  libfribidi0 libgnutls30 libpam-runtime libssl1.1 libswresample3 libwbclient0
  locales minidlna nano openssh-client openssh-server openssh-sftp-server
  openssl python3-samba samba samba-common samba-common-bin samba-libs tzdata
  webmin wireless-regdb
47 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 28.4 MB/114 MB of archives.
After this operation, 232 kB of additional disk space will be used.
Get:1 https://download.webmin.com/download/repository sarge/contrib armhf webmin all 1.997 [28.4 MB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 1010 kB in 3s (350 kB/s)
(Reading database ... 84373 files and directories currently installed.)
Preparing to unpack .../libpam-runtime_1.4.0-9+deb11u1+rpt2_all.deb ...
Unpacking libpam-runtime (1.4.0-9+deb11u1+rpt2) over (1.4.0-9+deb11u1+rpt1) ...
dpkg: error processing archive /var/cache/apt/archives/libpam-runtime_1.4.0-9+deb11u1+rpt2_all.deb (--unpack):
 unable to stat './usr/share/locale/bg/LC_MESSAGES' (which was about to be installed): Bad message
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libpam-runtime_1.4.0-9+deb11u1+rpt2_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
 What must I do?
Thank You.

Joulinar, 2022-07-20:
could it be similar to this one? https://github.com/MichaIng/DietPi/issues/5565

Looks like you have same issue again?

MichaIng, 2022-09-15:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2022-09-15 by MichaIng