MichaIng/DietPi #5637: Image | Add support for VMware/ESXi ARMv8 64-bit (or generic arm64 image)

CypZ, 2022-07-21:
### Creating an image request

Goal is to run DietPi on my rpi 4B but on top of ESXi, not on bare metal. As far as I can see only x86_64 images are available for VMware.

#### Formal device information
- Device name | VMware/ESXi ARMv8 64-bit
- Official product URL | https://flings.vmware.com/esxi-arm-edition
- GitHub resource URL (if available) | 
- Image download URL | 

#### Is the SBC officially supported by the Debian installer?
- Debian generally support arm64 (and VMware virtual environment too) : https://www.debian.org/releases/stretch/arm64/ch02s01.html.en

#### If not, is a reliable 3rd party Debian image available for this SBC?
- ...

#### If not, are there install instructions for Debian available?
- https://www.debian.org/releases/stretch/arm64/ch02s01.html.en
- https://williamlam.com/2020/10/how-to-run-raspberry-pi-os-as-a-vm-on-esxi-arm.html
- https://www.youtube.com/watch?v=JN1h2ztUHLo

MichaIng, 2023-09-12:
@bbugh
As you requested the same, moving discussion here.

Providing a VMX appliance and VMware/ESXi can go hand in hand, I think there is not much needed to change within the VMX (config) file. For the containing image, AFAIK it is only `grub-efi-arm64` and the regular `linux-image-arm64` kernel? I guess we could also go with the `linux-image-current-arm64` kernel from Armbian, but aside of some lower disk space usage, I don't think that it has much benefits.

Tasks to do:
- Decide whether to add a new hardware ID for ARMv8 VMs oder whether to review and rework some parts of our code where ID 20 is currently treated as `x86_64`. Not sure what makes more work, I'd probably check of `$G_HW_MODEL == 20 ` in our code and see how often this happens and how much work it would be to add another differentiation for `$G_HW_ARCH` in case. But I think in most cases where the architecture plays a role we do already check for the architecture, while VMs of any architecture have sufficiently much in common to handle them below the same ID.
- Generate bare `img` (or VMDK) and import it into new ESXi-Arm VM. Then export the appliance if it works. What I want to know is the VMX and/or OVF config file, to compare with the `x86_64` ones and have a template. I hope that they are mostly identical.

oxoocoffee, 2024-07-19:
Please let me know if I can assist by providing information or testing on MBP M1 Parallels 19 or UTM.

Najihel, 2024-09-08:
Hello,

Same request to host dietpi VM on Freebox Server (KVM)

Thanks

MichaIng, 2025-05-15:
They are, but no ETA. But I recently did some changes which make an implementation easier: https://github.com/MichaIng/DietPi/commit/f8002eb

MichaIng, 2026-08-23:
Okay, seems sufficiently easy with a QCOW2 then.

Installer ISOs for ARMv8 would be a topic in its own. Currently we use Clonezilla Love, with is x86_64 only.

MichaIng, 2026-08-31:
Right, this won't be needed anymore. I further integrated this into our build scripts, running a test build with a full set of ARMv8 Trixie VM images: https://github.com/MichaIng/DietPi/actions/runs/33407700538

This includes raw `.img` images, `.qcow2`, tarballs with VMX/VMDK appliances, ESXi+VirtualBox OVA appliances, an UTM appliance, and a VHDX/Hyper-V image. Not sure whether Hyper-V and VirtualBox run on Windows for ARM, and whether the VM configurations in the appliances are all compatible with the ARM VMs. If not, let me know, and we can adjust the configs for ARM or skip individual builds.

The images can be found here once done: https://dietpi.com/downloads/images/testing/

Maaaash32, 2026-09-03:
I have a problem with your last .qcow2 image. The VM didn't give itself an IP adress. So I cannot connect to it via SSH and can't interact with it. The console inside freebox OS is only a screen, I can't write inside.
The raw image of june didn't had this IP adress problem. From the start, It had an IP adress so I could interact with it via SSH.

I can't give you a log so here is a screenshot

[image]

MichaIng, 2026-09-03:
Is the screenshot from the first boot, or did you reboot once already? Because I see no `dietpi-firstboot.service` running. But also possible that something went wrong with the new `dietpi-network` interface configuration. I'll review/test.

Maaaash32, 2026-09-03:
No, I reboot it once already because it hasn't an IP on the first boot.

MichaIng, 2026-09-10:
Found it, I missed to use `dev` branch for the DietPi code on the image, but did use the `dev` branch of the `dietpi-installer`. That way `/etc/network/interfaces` was empty, which is fine for v10.7, but not for v10.6.

New build running: https://github.com/MichaIng/DietPi/actions/runs/34512886036

closed 2026-09-12 by MichaIng

Maaaash32, 2026-09-16:
Hello, your last .qcow2 image for an ARMv8 VM of Dietpi 10.7 works like a charm on my Freebox Delta. 👍
Thank you very much ! 🥳

MichaIng, 2026-09-17:
Great, many thanks for testing it again 🙂.