MichaIng/DietPi #5639: Odriod N2+ Does not Boot from eMMC

keypunch416, 2022-07-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.6`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | unknown as will not boot
- SBC model | Odroid N2+
- Power supply used | 12/5A
- eMMC card used | Label on eMMC antistatic bag "eMMC Memory Board - 16G", number below bar code "114991509", "SEED STUDIO", "M0A200110009", "01/18/20"

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated? image freshly imaged to new eMMC
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...7z x /mnt/nfs/Data/Downloads/DietPi/DietPi_OdroidN2-ARMv8-Bullseye.7z
2. ...sudo dd if=/mnt/nfs/Data/Downloads/DietPi/DietPi_OdroidN2-ARMv8-Bullseye.img of=/dev/sdf bs=2048k oflag=dsync ; time sudo sync via USB to eMMC adapter known to work with same eMMC modules (I have 5 of them ordered same time for order received 20 July 2022)
3. Connected eMMC to Odriod board no SD card inserted and switch on eMMC position and not SPI position

#### Expected behaviour

- Expected DietPi would boot to login prompt as has done with Manjaro-ARM on another one of same eMMC ordered same time imaged using same USB to eMMC adapter

#### Actual behaviour

- All sorts of output pause and repeats what looks like attempts to boot from the eMMC unsuccessfully

#### Extra details

- Cannot attach any console or logs as occurs on HDMI display
-  eMMC SanDisk 501N8BD04 16G (difficult to read as weak lettering to start likely rubbed off due to finger tip required to secure eMMC to USB adapter and to Odriod N2+
- Information on chip was as best could read and unable to read another line of letters/numbers on chip
- Same DietPi image file imaged to a 32GB Kingston Canvas MicroSD card boots to login prompt and functions like any Linux based OS

Joulinar, 2022-07-22:
I guess it is the same like this https://dietpi.com/forum/t/odroid-n2-boot-up-problem-with-emmc/13359

keypunch416, 2022-07-22:
No not same issue.  As noted in the open of this issue there screen is far from blank and ongoing retry activity for as long as power is connected all be it likewise not successful.

keypunch416, 2022-07-23:
As I am sure many have experienced while searching wit completely different search terms for completely different area of interest in search I happened upon this that seems would be helpful and indicating I am not only one:

<https://forum.armbian.com/topic/20206-odroid-n2-issues-with-recent-firmware-and-emmc-modules/>

In my case I have the red eMMC colored modules and as the above link user noted no issues using these modules with Manjaro-ARM

As side point the search I was and have been working on for past 2 days is how to enableN2+  SPI with Diet Pi.  I gave up using Manjaro-ARM after day as not able to find how to enable SPI with Manjaro-ARM.  This should not be rocket science.  Be clear this issue is about DietPi not booting from eMMC and not about my SPI challenges.  I am simply noting my search for how to enable SPI on DietPi turned up the above link that did not turn up when I was searching for solutions for inability to boot DietPi from eMMC before I opened this issue.

MichaIng, 2022-11-12:
Latest images should solve this, where we use a different U-Boot.

closed 2022-11-12 by MichaIng