MichaIng/DietPi #5642: time sync with gateway via systemd-timesyncd not working

schnitzi99, 2022-07-23:
### Creating a bug report/issue
The issue seems that the configuration of a time-server is overwritten at each boot. That leads to systemd-timesync using the standard time server. If dietpi is used as a pihole-unbound-DNS-resolver this leads to a failed time sync as the certificates are invalid due to a old system time (july 2021).

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 1
- Kernel version | `uname -a`
Linux rpi2 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 2 Model B (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
systemd-timesyncd
- Was the software title installed freshly or updated/migrated?
migrated from previous rpi / raspbian version

- Can this issue be replicated on a fresh installation of DietPi?
unknown

- Bug report ID | `echo $G_HW_UUID`
6ccb9673-7106-45cb-85ec-25bc925a96d9

#### Steps to reproduce
1. Install pihole
2. install and use unbound as dns resolver
3. reboot
4. DNS-resolving does not work
5.  pihole says something like "certificate invalid"
6. check system time: realize the date is totally wrong
7. set ntp_server to ip adress of gateway in /etc/systemd/systemd_timesyncd
8. systemctl restart systemd_timesync
9. resolving works
10. reboot
11. DNS-resolving does not work
12. check /etc/systemd/systemd_timesyncd and realize that ntp_server is unset again

#### Expected behaviour
- time server should be set to gateway ip.
- syncing time with gateway no problem as we use the ip of the gateway

#### Actual behaviour
See at "steps to reproduce"

#### Extra details

Joulinar, 2022-07-23:
did you tried using `dietpi-config` > `8  : Network Options: Misc` > `NTP Mirror`
For me this is working without issuing using my own NTP server. Anyway even using a global debian NTP server should update your system time automatically during boot and at least once a day. Can you share the logs

```
journalctl -u systemd-timesyncd.service
```

MichaIng, 2022-07-23:
Also you mean the `NTP=` setting `/etc/systemd/timesyncd.conf`, don't you? Of course any change to is should be persistent. If it is not, then the underlying filesystem/drive seems to have issues syncing the change, e.g. when it's still in filesystem cache and the server crashes or due to some filesystem corruption.

schnitzi99, 2022-08-09:
Correct, I'm taking about the NTP= setting /etc/systemd/timesyncd.conf.

Changes are definitely not persistent, i have to change the setting after each reboot. (I did several tests)

I cannot rule out drive or file system issues, but this is the only file having this kind of problem. I will try another sd card after my holidays.

MichaIng, 2022-08-11:
Any other change to this filesystem is persistent, e.g. when you create a test file just in the same directory?

Can you show:
```sh
ls -l /etc/systemd/timesyncd.conf
```

schnitzi99, 2022-08-11:
A short test yesterday showed that there is some kind of (probably hardware) issue. 
This might be the cause of the problems. 
Due to holidays I'm not able to narrow down the hardware problem or do a fresh install right now. 
Feel free to mark this issue as solved or paused or similar, i will give feedback after my holidays.

Thanks a lot for the fast support so far!

MichaIng, 2022-09-15:
I'll mark this as closed for now. Feel free to reopen if the issue persists.

closed 2022-09-15 by MichaIng

schnitzi99, 2022-09-18:
Ah sorry - I forgot to answer.
Problem was a hardware issue. Problem was gone with new hardware.
Can stay closed :)
Thanks for the support and all the work on dietpi!!!