MichaIng/DietPi #5643: Open Beta v8.7 | Please help testing and hardening our upcoming release

MichaIng, 2022-07-23:
RC version | v8.7.0
---------- | -------
v8.7.0 PR | #5640
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2022-07-30**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.7
____
### Beta v8.7.0
_(2022-07-23)_

#### New images
- NanoPi R5S | Initial support for FriendlyELEC's new router SBC has been added, with three Ethernet ports, up to 2.5 Gbps, based on the new Rockchip RK3568 SoC.

#### New software
- Readarr | The ebook manager of the Servarr family has been added as dietpi-software option.
- Navidrome | An open source web-based music collection server and streamer has been added as dietpi-software option.

#### Enhancements
- DietPi-Software | Prowlarr: Logging is now done to /var/log/prowlarr as intended. For this change to take effect, existing instances need to be reinstalled once: dietpi-software reinstall 151
- DietPi-Software | vaultwarden: DEB packages are now hosted on dietpi.com, replacing the time consuming in-place compiling. This also solves issues where builds failed due to insufficient memory.

#### Bug fixes
- DietPi-Software | Resolved an issue where "cgroup_enable=memory" was unnecessarily added multiple times on Raspberry Pi when installing or reinstalling container engines. Many thanks to @mdsjip for fixing this issue: https://github.com/MichaIng/DietPi/pull/5600
- DietPi-Software | Webmin: Resolved an issue where the service failed to start. Many thanks to @alucard87pl for reporting this issue: https://github.com/MichaIng/DietPi/issues/5594
- DietPi-Software | Raspotify: Resolved an issue where the service failed to start on ARMv6 RPi models (Raspberry Pi 1 and Zero (1)). It is not supported by latest Raspotify anymore, so we install the latest ARMv6-compatible version instead. Many thanks to @dvelluto for reporting this issue: https://github.com/MichaIng/DietPi/issues/5602
- DietPi-Software | Rclone: Resolved an issue where on ARMv6 the x86_64 package was attempted to be installed. Many thanks to @eddiermar for reporting this issue: https://github.com/MichaIng/DietPi/issues/5601
- DietPi-Software | Unbound: Resolved an issue on Debian Bookworm where the service failed to start because of missing root trust anchors. Many thanks to @smittyj for reporting this issue: https://github.com/MichaIng/DietPi/issues/5612
- DietPi-Software | Tailscale: Resolved an issue where APT updates failed after Tailscale was uninstalled. Many thanks to @SlowRaid for resolving this issue: https://github.com/MichaIng/DietPi/issues/5623

Joulinar, 2022-07-30:
@MichaIng 
can you have a look to this one bevor doing the release? https://dietpi.com/forum/t/setting-locale-was-not-correct-missing-package-libencode-locale-perl-was-the-root-cause/13838

MichaIng, 2022-07-30:
Are you able to replicate this anywhere? It works perfectly fine here without it, on VM and on Odroid N2, and if a package is really required for the main functionality of another, I'd assume it being a dependency or at least a recommendation, but it isn't.

No time to deeper investigate it until release, seems to be an edge case where we need the exact hardware/image and to be able to replicate. What is suspicious is that the previous locale seems to have been en_GB, while our default is C.UTF-8, so something has either been changed already, or it is a very old image (none that we have on our server anymore).

Joulinar, 2022-07-30:
Ok let's check it later on. Maybe it's user specific and not something global

MichaIng, 2022-07-31:
#### DietPi v8.7 has been released: #5658
Many thanks to all testers ❤️!

closed 2022-07-31 by MichaIng