MichaIng/DietPi #5646: Wrong IP on Bridged Networking VM  -- DietPi VirtualBox VM v8.6.1

hashborgir, 2022-07-25:
Diet Pi, v8.6.1
Virtualbox VM
Bridged Networking
LAN/Host is 10.x IP 

Every single time, the DietPi VM always gets an IP from 192.x

There is no 192.x DHCP server on my LAN

Where/how is DietPi getting the wrong IP from?

If I try to manually change to 10.x, the DP VM has no networking. 

Every other VM on my host has a 10.x IP on bridged as my local router/LAN is 10.x

Please advise.

Joulinar, 2022-07-25:
What is the exact IP you get. Is it a 192.169.0.0? Or something else?

hashborgir, 2022-07-25:
VM config
[image]

DietPi after fresh install
[image]

I manually changed from DHCP to Static. Then I changed ip in dietpi-config to 10.x, and rebooted.

Now I have 10.x ip with networking.

I'm still confused as to why on fresh install, it's getting an ip from 192.x instead of 10.x

Even with the wrong 192.x address, it has networking, but my Lan machines can't talk go it without forwarding. This is why I prefer Bridged over NAT. 

Even with Bridged, it's defaulting to a 192.x address, but I don't have any 192.x DHCP servers anywhere. What is DietPi doing?

Joulinar, 2022-07-25:
I did a test with latest VirtualBox + latest Bullseye image without issues. The VM was booting with Bridged mode enabled and got correct IP address assigned. What you could do, is to test DHCP and to check logs. There you should see where the IP address was assigned from. Set the VM back to DHCP and reboot. Afterwards check following

```
journalctl -u ifup@eth0.service
```

usually there should be a line like this on the output.

```
Jul 25 11:12:05 DietPi dhclient[223]: DHCPOFFER of 192.168.0.66 from 192.168.0.11
```

MichaIng, 2022-09-15:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2022-09-15 by MichaIng