MichaIng/DietPi #5647: DietPi-Software | PiHole - new installer

Joulinar, 2022-07-25:
Hi @MichaIng 

if I remember well, you told PiHole guys have released a new installer and they moved a away from `wiptail`. On the new installer, I'm not sure if we have some challenges. It seems to produce useless characters while trying to use tab to select an option 

[image]

Next, you will get a question if to install the Admin Interface. Selecting yes resulting on next question, if `lighttpd` should be installed. Even if we call the installer with ` --disable-install-webserver` option. Bad thing, selecting yes will overwrite our `lighttpd.conf`

[image]

```
  [i] Installing StevenBlack's Unified Hosts List
  [i] Installing Admin Web Interface
  [i] Installing lighttpd
  [i] Query Logging on.
  [i] Using privacy level: 0
```

Choosing no will skip that step

```
  [i] Installing StevenBlack's Unified Hosts List
  [i] Installing Admin Web Interface
  [i] Not installing lighttpd
  [i] Query Logging on.
  [i] Using privacy level: 0
```

MichaIng, 2022-07-25:
Nasty, looks like the conditions for whether the dialogs are shown or not got lost. Hmm, why does `dialog` look so ugly? It should be able to show proper UTF-8 borders instead of those ASCII placeholders.

Joulinar, 2022-07-25:
maybe you could run a test if you are able to confirm the behaviour. Just to exclude it is something on my SSH client or settings.

MichaIng, 2022-07-31:
Fixed with: #5652

closed 2022-07-31 by MichaIng