MichaIng/DietPi #5648: Node-RED installation does not install any webbrowser on a fresh DietPi installation

StephanStS, 2022-07-25:
#### Required Information
- DietPi version | 
  - G_DIETPI_VERSION_CORE=8
  - G_DIETPI_VERSION_SUB=7
  - G_DIETPI_VERSION_RC=0
  - G_GITBRANCH='dev'
  - G_GITOWNER='MichaIng'

- Distro version | bullseye
- Kernel version | `Linux r5s 5.10.110 #81 SMP Mon Jul 4 15:26:21 CST 2022 aarch64 GNU/Linux`
- SBC model | Generic Device (aarch64), NanoPi R5s

#### Additional Information (if applicable)
- Software title | Node-RED (`#122`)
- Was the software title installed freshly or updated/migrated? Freshly on a fresh DietPi installation
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Run `dietpi-software install 122`
2. Try to access your Node-RED instance via port 1880 or the webserver default page
    [image]

#### Expected behaviour

- The Node-RED page should appear:
    [image]
- I.e. a webserver installation should be triggered by the Node-RED installation
  

#### Actual behaviour

- No webpage is displayed
- In `dietpi-software`, no webserver is shown as an installed package

#### Extra details

- If a webserver is installed manually (e.g. `dietpi-software install 85`) and the system is rebooted, Node-RED can be accessed

Joulinar, 2022-07-25:
Usually NodeRed is having an own web server and there shouldn't be a need to install Apache or Nginx.

StephanStS, 2022-07-25:
> Usually NodeRed is having an own web server and there shouldn't be a need to install Apache or Nginx.

I'll check this (maybe it was a http vs. https problem).

StephanStS, 2022-07-25:
@Joulinar: Right, webserver runs also in base installation. Problem occured due to usage of https: which is not valid in a base installation.

closed 2022-07-25 by StephanStS

Joulinar, 2022-07-25:
Correct. 😄