MichaIng/DietPi #5649: Add USB to Ethernet to base image for POE to USB/Ethernet experience

recoi1er, 2022-07-27:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...Have POE to USB Ethernet as part of the base image vs trying to get it working after the fact for ease of setup

#### Describe the solution you'd like:

- ... Have USB ethernet as an option during the initial config before even booting the device where you enter in your details for WiFi or Ethernet, have USB as another option, it would make for a much easier experience as with a Pi Zero it does has not Ethernet but Ethernet is infinitely better for performance (even if only limited to 100mbps, the sdcard can't fully utilize 1gbps in my experience nor can the device itself even if you use an ssd but it would benefit from an always on connection and better ping). Especially helpful for PiHole.

#### Describe alternatives you've considered:

- ...

#### Additional context

- ... I have this device that delivers power and ethernet via POE adapter to the data USB port:
https://www.uctronics.com/tools-and-accessories/pi-accessories/uctronics-poe-adapter-to-micro-usb-ethernet-power-for-raspberry-pi-zero-fire-tv-stick-chromecast-google-mini-and-more-ieee-802-3af-compliant.html

MichaIng, 2022-08-07:
Many thanks for your request.

Isn't this USB-Ethernet adapter just showing up as interface `eth0` (e.g. when running `ip l`)? In this case it should work just OOTB on RPi Zero, respectively on any hardware which doesn't have onboard Ethernet (in which case onboard Ethernet is configured).

Or does it show up as `usb0` like with USB OTG Ethernet as here? #3253

recoi1er, 2022-08-08:
I turned off eth0 as directed in your setup guide for WiFi - how would I turn that back on or do I need to setup new again?

Joulinar, 2022-08-08:
You should be able to do this using `dietpi-config` network settings

MichaIng, 2026-08-15:
I don't think there was any feature missing here.

However, for completeness, the following PR adds support for other interface names, like `usb0` (Ethernet over USB OTG), and makes switching between WiFi and Ethernet (between interfaces/adapter in general) easier, as it warns about conflicting routes, and allows to disable the old interface and enable the new one in one go: #8251

closed 2026-08-15 by MichaIng