MichaIng/DietPi #5650: Upgrade from Node-RED V2x to V3 fails

MDAR, 2022-07-28:
Bug report Reference code: e9065ed8-76b5-47f2-9cc1-3b8849e8d6ca

Odroid C4
DietPi V8.6 (upgraded directly from V6.4)

eMMC 16GB memory

I tried running  this command  ```dietpi-software reinstall 122``` in order to go from Node-RED V2 to V3

This failed repeatedly on one machine.
Which I wiped completely and put a fresh install of DietPi on and Node-RED V3 got installed nicely, so I knew the install process was good.

Trying to update V2 to V3 on another machine and the same issue arose.

So I removed Node-RED using the menu and tried to install a fresh 

this failed.

I looked for any hit of Node-red and found a load of folders.

`/usr/local/lib/node_modules/node-red/`

*******

Now this is silly, but I think I've found part of the issue

```/etc/hosts``` did not match ```/etc/hostname```

Which I think was a cause of some other issues

(I changed them ages ago and must have missed 

But it now fails with a new error

```
  - Command: sudo -u nodered npm i --no-audit node-red                        │
│  - Exit code: 1                                                              │
│  - DietPi version: v8.7.0 (MichaIng/dev) | HW_MODEL: 16 | HW_ARCH: 3 |       │
│ DISTRO: 5                                                                    │
│  - Image creator: DietPi Core Team                                           │
│  - Pre-image: Meveric                                                        │
│  - Error log:                                                                │
│ /usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js │
│ :110                                                                         │
│ class SocksProxyAgent extends agent_base_1.Agent {                           │
│                                            ^                                 │
│                                                                              │
│ TypeError: Class extends value undefined is not a constructor or null        │
│     at Object.<anonymous>                                                    │
│ (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.j │
│ s:110:44)                                                                    │
│     at Module._compile (node:internal/modules/cjs/loader:1120:14)            │
│     at Module._extensions..js (node:internal/modules/cjs/loader:1174:10)     │
│     at Module.load (node:internal/modules/cjs/loader:998:32)                 │
│     at Module._load (node:internal/modules/cjs/loader:839:12)                │
│     at Module.require (node:internal/modules/cjs/loader:1022:19)             │
│     at require (node:internal/modules/cjs/helpers:102:18)                    │
│     at Object.<anonymous>                                                    │
│ (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js │
│ :174:29)                                                                     │
│     at Module._compile (node:internal/modules/cjs/loader:1120:14)            │
│     at Module._extensions..js (node:internal/modules/cjs/loader:1174:10)     │
│     at Module.load (node:internal/modules/cjs/loader:998:32)                 │
│     at Module._load (node:internal/modules/cjs/loader:839:12)                │
│     at Module.require (node:internal/modules/cjs/loader:1022:19)             │
│     at require (node:internal/modules/cjs/helpers:102:18)                    │
│     at Object.<anonymous>                                                    │
│ (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/remote.j │
│ s:7:18) 
```

Joulinar, 2022-07-28:
Did you tried to follow the update instructions described on our online docs https://dietpi.com/docs/software/hardware_projects/#node-red

MDAR, 2022-07-28:
> Did you tried to follow the update instructions described on our online docs https://dietpi.com/docs/software/hardware_projects/#node-red

Thanks.

To be honest, I was following previous instructions from Michael.

I might well have broken something, but I'll try the commands and see what happens.

```
systemctl stop node-red
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm up --no-audit node-red
systemctl start node-red
```

MDAR, 2022-07-29:
> Did you tried to follow the update instructions described on our online docs https://dietpi.com/docs/software/hardware_projects/#node-red

@Joulinar 

thanks for you reply.

I'm not entirely sure what has happened, but I think something has gone horribly wrong.

Trying to install Node-RED from the dietpi-software menu always ends in the same error state that is referred to in the last bug report, but it does look like it's trying to install it.

When I run the sequence of commands you directed me to, I get this response.
(Which looks to contain the failing command in the bug report)

```
root@openHAB-C4:~# cd /mnt/dietpi_userdata/node-red
root@openHAB-C4:/mnt/dietpi_userdata/node-red# sudo -u nodered npm up --no-audit node-red
/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:110
class SocksProxyAgent extends agent_base_1.Agent {
                                           ^

TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:110:44)
    at Module._compile (node:internal/modules/cjs/loader:1120:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1174:10)
    at Module.load (node:internal/modules/cjs/loader:998:32)
    at Module._load (node:internal/modules/cjs/loader:839:12)
    at Module.require (node:internal/modules/cjs/loader:1022:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:174:29)
    at Module._compile (node:internal/modules/cjs/loader:1120:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1174:10)
    at Module.load (node:internal/modules/cjs/loader:998:32)
    at Module._load (node:internal/modules/cjs/loader:839:12)
    at Module.require (node:internal/modules/cjs/loader:1022:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/remote.js:7:18)
    at Module._compile (node:internal/modules/cjs/loader:1120:14)

Node.js v18.7.0
root@openHAB-C4:/mnt/dietpi_userdata/node-red#
```

As the install from a fresh OS image worked, is there anyway I can purge any references to Node-red (or the offending part) and get the `dietpi-software ` method to work?

MDAR, 2022-07-29:
@Joulinar & @MichaIng 

please don't ask me why this works, as that level of **sorcery** is yours alone.

I now have Node-RED V3 running on the machine. !!!

[image]

My approach was a bit drastic, but in times of trouble one must try these things.

I removed

`node.js` (9)
&
`Build-Essential` (16)

using `dietpi-software` menu (not the command line method, I know it doesn't make any difference, I just wanted to do it in a way that I couldn't get wrong)

then ONLY attempted to install `Node-RED` (122) knowing that your lovely scripts would put back in `Node.js` and `Build Essentials`.

I have an archive image of a machine that ran Node-RED V2.x if you'd like me to load it up and try anything. (if you want to look further into this..... or maybe not 😉 )

(I'll leave one of you to close this, when you are happy that all the info is present or the proposed solution is sanctioned)

MichaIng, 2022-08-07:
Do you remember when you installed Node-RED the first time?

It seems there are issues with some globally installed Node modules conflicting with the ones locally installed into the Node-RED directory or so. We had similar reports already, probably when a dependency is available as global module, it is not installed locally, even when the global version is not compatible, or so.

And we installed Node-RED as global module a longer time ago, and switched to a local one in the meantime. When I find time, I'll try to replicate and check whether this `/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js` is part of the base Node installation or a dependency of Node-RED, installed locally on a fresh install.

MDAR, 2022-08-08:
> Do you remember when you installed Node-RED the first time?

That's a good question....

I don't remember.

It was "quite some time ago"

So very probably part of the combination you have described.

Is there a log I can look at to see when it was first installed?

MichaIng, 2022-08-08:
Hmm no sadly there are no `dietpi-software` logs kept.

MichaIng, 2022-09-25:
I'll mark this as closed. Difficult to investigate, and probably not worth it, given that the global > local change was done a long time ago.

closed 2022-09-25 by MichaIng