MichaIng/DietPi #5654: Failed to register regulator

keypunch416, 2022-07-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.6.1
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.123-meson64 #22.05.3 SMP PREEMPT Wed Jun 22 07:23:04 UTC 2022 aarch64 GNU/Linux
- SBC model | Odroid N2 (aarch64) (N2+ 4GB)
- Power supply used | ALLPOWER 15VDC/4A
- SD card used | 32GB Kingston CANVAS Select Plus MicroSD 100MB/s A1

#### Additional Information (if applicable)
- Software title | DietPi
- Was the software title installed freshly or updated/migrated? Freshly installed and up to date as of this issue
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | N/A

#### Steps to reproduce

1. Boot system
2. dmesg | grep -i -e volt
3. wait hours or days
4. dmesg | grep -i -e volt

#### Expected behaviour

- I would expect no power events to be found I dmesg

#### Actual behaviour

- [    1.181955] reg-fixed-voltage regulator-flash_1v8: Failed to register regulator: -517
[    1.182503] reg-fixed-voltage regulator-vcc_1v8: Failed to register regulator: -517
[    1.182573] reg-fixed-voltage regulator-vcc_3v3: Failed to register regulator: -517
[    1.182721] reg-fixed-voltage regulator-vddao_1v8: Failed to register regulator: -517
[    1.978224] reg-fixed-voltage regulator-flash_1v8: Failed to register regulator: -517
[    1.978601] reg-fixed-voltage regulator-vcc_1v8: Failed to register regulator: -517

#### Extra details

- The power supply being used is oversized on purpose to ensure SBC does not have any shortages of power.
-Until other issues and yet to be effected SBC, Linux, and application changes are made a USB Keyboard and HDMI are connected.  No other USB devices are connected.  This has been case since first started using N2+ after received about 9 days ago.
This is consistent issue on every boot and appears to be boot related based on timestamps of dmesg

MichaIng, 2022-07-30:
Many thanks for your report.

I also see these messages, but I cannot observe any related negative effect. These seem to be regulators for devices which are not attached, e.g. if you have no SD card and no power input via GPIO pins or so. I.e. nothing to worry about.

keypunch416, 2022-07-31:
SD card was being used WRT to issue occurring.

Issue does not occur on Hardkernel "official" release:

Sun Jul 31 22:13:40 UTC 2022
`Linux odroid 4.9.312-22 #1 SMP PREEMPT Mon Jun 20 17:12:03 UTC 2022 aarch64
aarch64 aarch64 GNU/Linux`
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy

               clk_freq= 285714285, clk_parent=fclk_div7, voltage=1150, keep_count=5, threshod=<100 190>, clk_sample=285
               clk_freq= 400000000, clk_parent=fclk_div5, voltage=1150, keep_count=5, threshod=<152 207>, clk_sample=400
               clk_freq= 500000000, clk_parent=fclk_div4, voltage=1150, keep_count=5, threshod=<180 220>, clk_sample=500
               clk_freq= 666666666, clk_parent=fclk_div3, voltage=1150, keep_count=5, threshod=<210 236>, clk_sample=666
               clk_freq= 800000000, clk_parent=fclk_div2p5, voltage=1150, keep_count=5, threshod=<230 255>, clk_sample=800
               clk_freq= 800000000, clk_parent=fclk_div2p5, voltage=1150, keep_count=5, threshod=<230 255>, clk_sample=800
[    3.032222] value of voltage_tolerance 0
[    3.034224] value of voltage_tolerance 0
[   11.340306] rtc-pcf8563 1-0051: low voltage detected, date/time is not reliable.

MichaIng, 2022-09-11:
Does the same errors show up when you boot the original Armbian Bullseye image? https://www.armbian.com/odroid-n2/#kernels-archive-all

And/or after you do a full kernel upgrade on Armbian:
```sh
apt update
apt full-upgrade
reboot
```

MichaIng, 2022-11-12:
The error have gone on Linux 5.19 "edge" kernel branch. Should reach "current" kernel branch soon. Since there are no actual issues related to these error messages, I'm going to close this issue.

closed 2022-11-12 by MichaIng