MichaIng/DietPi #5661: ROCKPro64: Ethernet and USBs don't work

Kreyren, 2022-07-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```console
root@DietPi:~# cat /boot/dietpi/.version 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```console
root@DietPi:~# echo $G_DISTRO_NAME $G_RASPBIAN
bullseye
```
- Kernel version | `uname -a`
```console
root@DietPi:~# uname -a
Linux DietPi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```scm
root@DietPi:~# echo $G_HW_MODEL_NAME
ROCKPro64 (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower): Lab PSU set on 12.0VDC with limit of 5A
- SD card used | (EG: SanDisk ultra): TBA

#### Steps to reproduce

1. Download the image from https://dietpi.com/downloads/images/DietPi_ROCKPro64-ARMv8-Bullseye.7z
2. Unpack it with `7z x path/to/DietPi_ROCKPro64-ARMv8-Bullseye.7z`
3. Connec the sdcard and invoke `sudo dd if=path/to/DietPi_ROCKPro64-ARMv8-Bullseye.img of=/dev/DEVICE conv=sync status=progress && sync` to install the OS on the sdcard
4. Disable SPI with connection pin 23 <-> 25
5. Boot the image and experience the highlighted issues

#### Expected behaviour
1. Working ethernet connection
2. Working USB ports
(3. working display)

#### Actual behaviour
I don't have:
* Ethernet connection
* USB connectivity (both USB2 and USB3)

#### Extra details

* This is also a problem on: Armbian, Manjaro, Debian
* New board barelly used
* dmesg:

```console
root@DietPi:~# dmesg 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.63-rockchip64 (root@runner4) (aarch64-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld 
(GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021
[    0.000000] Machine model: Pine64 RockPro64 v2.1
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x7fba3100-0x7fba4fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000007fffffff]
[... 653 lines cut ...]
[   34.912492]  generic_handle_irq+0x30/0x48
[   34.912510]  rockchip_irq_demux+0xbc/0x240
[   34.912525]  generic_handle_irq+0x30/0x48
[   34.912541]  __handle_domain_irq+0x94/0x108
[   34.912556]  gic_handle_irq+0xc0/0x140
[   34.912572]  el1_irq+0xc8/0x180
[   34.912590]  arch_cpu_idle+0x18/0x28
[   34.912607]  default_idle_call+0x44/0x1bc
[   34.912623]  do_idle+0x204/0x278
[   34.912638]  cpu_startup_entry+0x28/0x60
[   34.912655]  rest_init+0xd4/0xe4
[   34.912675]  arch_call_rest_init+0x10/0x1c
[   34.912693]  start_kernel+0x810/0x848
[   34.912702] handlers:
[   34.934013] [<000000005dd828e6>] irq_default_primary_handler threaded [<00000000c9d608b3>] regmap_irq_thread
[   34.956638] Disabling IRQ #77
[  190.020426] rk_gmac-dwmac fe300000.ethernet eth0: validation of rgmii with support 0000000,00000000,00006280 and advertisement 0000000,00000000,00006280 failed: -22
[  190.020723] rk_gmac-dwmac fe300000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -22)
[  252.585451] rk_gmac-dwmac fe300000.ethernet eth0: validation of rgmii with support 0000000,00000000,00006280 and advertisement 0000000,00000000,00006280 failed: -22
[  252.585760] rk_gmac-dwmac fe300000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -22)
[  403.312575] rk_gmac-dwmac fe300000.ethernet eth0: validation of rgmii with support 0000000,00000000,00006280 and advertisement 0000000,00000000,00006280 failed: -22
[  403.312878] rk_gmac-dwmac fe300000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -22)
```

CC @MichaIng @Joulinar

MichaIng, 2022-07-30:
Many thanks for your report.

Hmm, I see no errors about USB, but about the Ethernet port indeed, and the kernel error about IRQ 77. There are no voltage errors (besides some failed regulator registrations, expected if related devices/power inputs are not used).

Can you show:
```sh
lsusb
ip a
```
And do you have a different PSU and/or different power cable to test with, in case it is indeed voltage/power related?

Also, if you can establish a network connection via WiFi, could you try to upgrade the kernel to latest version?
```sh
apt update
apt full-upgrade
```

Kreyren, 2022-07-31:
> Many thanks for your report.
> 
> Hmm, I see no errors about USB, but about the Ethernet port indeed, and the kernel error about IRQ 77. There are no voltage errors (besides some failed regulator registrations, expected if related devices/power inputs are not used).
> 
> Can you show:
> 
> ```shell
> lsusb
> ip a
> ```
>  -- @MichaIng (https://github.com/MichaIng/DietPi/issues/5661#issuecomment-1200279169)

```console
root@DietPi:~# lsusb                                                           
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 4e:cf:b2:ea:31:49 brd ff:ff:ff:ff:ff:ff
```

> And do you have a different PSU and/or different power cable to test with, in case it is indeed voltage/power related? -- @MichaIng (https://github.com/MichaIng/DietPi/issues/5661#issuecomment-1200279169)

Tried those already with the same result

> 
> Also, if you can establish a network connection via WiFi, could you try to upgrade the kernel to latest version?
> 
> ```shell
> apt update
> apt full-upgrade
> ```

Unable, i don't have the WiFi adapter and USB also doesn't work to attempt USB modem

Kreyren, 2022-08-09:
ping @MichaIng

Kreyren, 2022-08-12:
Filled https://forum.pine64.org/showthread.php?tid=17135

MichaIng, 2022-09-18:
Sorry for the late reply. You do still face the issue? The `eth0` adapter is detected. Does
```sh
ifup --force eth0
```
not work? The interface is defined?
```sh
cat /etc/network/interfaces
```

Kreyren, 2022-09-20:
```console
# ifup --force eth0                              
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

[  133.232786] rk_gmac-dwmac fe300000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -22)
RTNETLINK answers: Invalid argument
Listening on LPF/eth0/4e:cf:b2:ea:31:49
Sending on   LPF/eth0/4e:cf:b2:ea:31:49
Sending on   Socket/fallback
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6
send_packet: Network is down
dhclient.c:2446: Failed to send 300 byte long packet over eth0 interface.
dhclient.c:2446: Failed to send 300 byte long packet over eth0 interface.
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 5
send_packet: Network is down
dhclient.c:2446: Failed to send 300 byte long packet over eth0 interface.
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9 149.112.112.112

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9 149.112.112.112
wireless-power off
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

Joulinar, 2022-09-20:
> [  133.232786] rk_gmac-dwmac fe300000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -22)

> send_packet: Network is down
dhclient.c:2446: Failed to send 300 byte long packet over eth0 interface.

Could you check the physical connection? There seems something not working correctly.

Kreyren, 2022-09-20:
Check how? I tried to replug it multiple times in case there is something obstructing the connection and the cable works fine in different system.. hardware damage?

Joulinar, 2022-09-20:
There should be a yellow and green LEDs on the LAN socket. Do they are active while connected to network + powered on?

Kreyren, 2022-09-24:
@Joulinar I don't see any LED working on the LAN socket after connection

Kreyren, 2022-10-11:
Neither of the LEDs is active on the ethernet connector both before and after connection

MichaIng, 2022-10-16:
Our image has been updated. Probably the underlying issue has been fixed with the newer kernel and/or bootloader.

MichaIng, 2022-12-04:
I'll mark this as closed, assuming it has been solved with recent kernel, as otherwise we should have seen more reports.

closed 2022-12-04 by MichaIng