MichaIng/DietPi #5662: After update to DietPi 8.7 dietpi-backup gets stuck

maartenlangeveld, 2022-08-01:


Joulinar, 2022-08-01:
Is a manually backup working? What is the target of your backup? Is this local storage or network attached?

maartenlangeveld, 2022-08-01:
Manual backup works. I did remove some old data_x folders (except the latest backup: data folder) before to cleanup. Maybe that caused backup to get stuck on dry-run?

Joulinar, 2022-08-01:
was it local storage or could it be a full file system?

MichaIng, 2022-08-07:
A missing old data_x directory btw shouldn't cause any issues. Missing ones are just recreated one by one with every iteration. When you say "stalled", do you mean it just aborted in the middle of the dry run (and how do you know that since cron doesn't log output?), or is there an insufficient free space log shown in the `.dietpi-backup_stats` file?

If it still fails with cron but works with manual execution, we'd need to forward the cron job's output to a log file to know what happened.

maartenlangeveld, 2022-08-08:
With "stalled" I mean the process did not die but did not nothing either. I saw the processes sitting idle in 'htop' for hours.
Has not happened since, however.

maartenlangeveld, 2022-08-08:
My RPi0 hanged last night, do not now the cause.

maartenlangeveld, 2022-08-08:
I think I have an idea what caused these issues.
Recently I started to make multiple rotating backups on daily basis using diepi-backup.

However once a week I make a backup (over the network, nfs-mount) to the other device using 'dietpi-update 1 <nfs-mount>' by means of a cron job. This will also trigger multiple backups and since the other device have not enough memory to store multiple backups from the other they probably will hang.

I guess I'll have to make the network backups using my own rsync command and not using dietpi-backup.

maartenlangeveld, 2022-08-08:
Ideally would be the possibility of having the option to define more backup-jobs using the dietpi-backup menu :-) !!

e.g.
- daily rotating backup making an amount of 5 backups on the device itself
and
- weekly backup to network device (amount = 1).

MichaIng, 2022-08-08:
When you change the backup location, only a single new backup is created in one run. The configured amount of "slots" is not forcefully created. When the target drive is full, rsync should error out, not hang, although I'm not 100% sure about this when NFS is used.

How do you make the weekly backup? Best would be to change the setting in `/boot/dietpi/.dietpi-backup` before the daily cron job runs, and back afterwards, so no two parallel and/or redundant backups are created.

We thought about adding another weekly rotation option by times when adding the daily one, although not with a different target location. I'll think about this, i.e. how to implement without having too many complicated looking options.

maartenlangeveld, 2022-08-08:
What I did before was a weekly cronjob using a tab in /etc/cron.d/ containing something like:

```
8 5 * * Mon  root mount /mnt/rpi-zero-w-backup-nanopi
9 5 * * Mon  root /boot/dietpi/dietpi-backup 1 /mnt/rpi-zero-w-backup-nanopi
```

As I only want to backup the latest backup remotely to another SBC on my network I have now in stead added a tab to /etc/cron.weekly:

```
/#!/bin/sh
#
# Backup latest system-backup to RPi0 (nfs-mount /mnt/rpi-zero-w-backup-nanopi/)

mount /mnt/rpi-zero-w-backup-nanopi
rsync -azq /mnt/dietpi-backup/data /mnt/rpi-zero-w-backup-nanopi/ &>/dev/nulletc/
```

MichaIng, 2022-08-17:
Ah nice, I totally forgot that the target path can be set/overridden via CLI 😅.

Indeed your second solution looks best when you want a single iteration on the remote drive only.

I'm still not sure why the backup would stall on the dry-run, especially since an actual `rsync` succeeds? This is the exact command, if you want to debug it:
```sh
cd /tmp
cat << '_EOF_' > .dietpi-backup_filter_inc_exc
# Backup data, log and config
- /mnt/rpi-zero-w-backup-nanopi/
- /boot/dietpi/.dietpi-backup_settings
# RAM dirs
- /dev/
- /proc/
- /run/
- /sys/
- /tmp/
# Swap files
- /var/swap
- .swap*
# Fake RTC timestamp
- /etc/fake-hwclock.data
# Unlinked inodes
- /lost+found/
# APT cache
- /var/cache/apt/*
_EOF_
rsync --dry-run --stats -aH --info=name0 --info=progress2 --delete-excluded --exclude-from=.dietpi-backup_filter_inc_exc / /mnt/rpi-zero-w-backup-nanopi/data/
```

maartenlangeveld, 2022-08-17:
> I'm still not sure why the backup would stall on the dry-run, especially since an actual rsync succeeds?

I presume since my rsync solution only backups latest backup iteration to remote device. The dietpi-backup function tried to backup all iterations to the remote device which had not enough memory to store 5 iterations, I guess.

MichaIng, 2022-08-18:
> The dietpi-backup function tried to backup all iterations

It does only a single backup iteration per run, so that shouldn't be the issue.

maartenlangeveld, 2022-08-22:
Then I do not know what caused the issue. This issue may be closed as far as I am concerned.

MichaIng, 2022-08-23:
Okay let's close it as you have a working solution. I'm also still not sure why the backup would hang on the dry run 🤔.

closed 2022-08-23 by MichaIng