MichaIng/DietPi #5663: USB Soundcard Not Detected Anymore

seniorm0ment, 2022-08-02:
I had it working, then it had randomly stopped.
I just updated the system (stooped working before that), to see if it would fix. Did not.

aplay -l does not list it. Also tried uninstalling alsa and reinstalling, did not fix.
I use it for mopidy. I also tried dietpi-config and selecting usb dac, when I click ok it shows none, and cli says usb-soundcard exited with error.

It's a Schiit Modi USB dac, tried two different USB ports.

MichaIng, 2022-08-22:
Many thanks for your report.

Please fill the bug report template, so we get the required information about your hardware and software. Please also add `dmesg -l emerg,alert,crit,err` or, if it doesn't contain something related, `dmesg | tail -10` some seconds after attaching the USB DAC.

seniorm0ment, 2022-08-22:
Apologies, the bug templates don't get shown when using the Fasthub app. I will logon to web github when I get time and provide this info!

MichaIng, 2022-09-15:
Any news from your end, regarding the info we'd need to help you?

seniorm0ment, 2022-09-15:
Apologies, I kept meaning to get around to this. My room is completely torn apart due to moving stuff around so it's not even hooked up right now. I will get back

MichaIng, 2022-11-09:
I'll mark it as closed for now (cleanup), please reopen when you find time to come back to this issue 🙂.

closed 2022-11-09 by MichaIng