MichaIng/DietPi #5668: PaperMC installation fault

blueyshark, 2022-08-04:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bullseye 1
- Kernel version | Linux MMSERVER01 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l) (2GB)
- Power supply used | Mains USB port, 3.1A
- SD card used | 512GB SSD drive for root, SD card for boot (sandisk i think)

#### Additional Information (if applicable)
- Software title | PaperMC
- Fresh Install (x3, testing if the issue was a one-off)
- Can this issue be replicated on a fresh installation of DietPi? Unsure, do not have capacity to try this

#### Steps to reproduce

1. After various EULA, plugin options, two dialogues appear (see attached screencaps)
2. ...

#### Expected behaviour

- PaperMC installs normally.

#### Actual behaviour

- After various EULA, plugin options, two dialogues appear (see attached screencaps) - it seems like a partial install, as the PaperMC data folder is empty, barring a "EULA" folder (see attached screencap)

Thanks. Cuberite seems very crash prone, so wanted to try PaperMC.

MichaIng, 2022-08-04:
Many thanks for your report. Let us try to replicate.

In your install output, is there a timeout shown when PaperMC is started and waited for config files to appear?

blueyshark, 2022-08-04:
Hi @MichaIng thanks much for the reply. How could I find that info? Is there a log or something to that effect?

Joulinar, 2022-08-04:
Usually it should have been shown on the SSH terminal if you would have scrolled above the blue error box

blueyshark, 2022-08-04:
ok! I will try again and send anything I see.

Joulinar, 2022-08-04:
If possible you could try to copy/paste the output directly from SSH terminal. Usually there is no need to do screen prints 🙂

blueyshark, 2022-08-04:
@Joulinar @MichaIng Thanks again for your help. Here is the output of my latest execution:

> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Step: Installing PaperMC: Highly optimised Minecraft server with plugins, written in Java
> 
> [  OK  ] DietPi-Software | cd /tmp/DietPi-Software
> [  OK  ] DietPi-Software | curl -sSfL https://papermc.io/api/v2/projects/paper/versions/1.19.1/builds/109/downloads/paper-1.19.1-109.jar -o paper-1.19.1-109.jar
> [  OK  ] DietPi-Software | mkdir -p /opt/papermc
> [  OK  ] DietPi-Software | mv paper-1.19.1-109.jar /opt/papermc/paperclip.jar
> [  OK  ] DietPi-Software | mkdir /mnt/dietpi_userdata/papermc
> [  OK  ] DietPi-Software | eval echo "eula=true" > /mnt/dietpi_userdata/papermc/eula.txt
> [  OK  ] DietPi-Software | Verifying user "papermc" to run PaperMC
> [  OK  ] DietPi-Software | useradd -rMU -d /mnt/dietpi_userdata/papermc -s /usr/sbin/nologin papermc
> [  OK  ] DietPi-Software | cd /tmp/DietPi-Software
> [  OK  ] DietPi-Software | Checking URL: https://github.com/Tiiffi/mcrcon/archive/master.tar.gz
> [ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://github.com/Tiiffi/mcrcon/archive/master.tar.gz -o master.tar.gz
> [ INFO ] DietPi-Software | APT install for: gcc libc6-dev, please wait...
> Reading package lists...
> Building dependency tree...                                                                                                                                                                             
> Reading state information...                                                                                                                                                                            
> gcc is already the newest version (4:10.2.1-1+rpi1).                                                                                                                                                    
> libc6-dev is already the newest version (2.31-13+rpt2+rpi1+deb11u3).                                                                                                                                    
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.                                                                                                                                          
> [  OK  ] DietPi-Software | APT install for: gcc libc6-dev
> [  OK  ] DietPi-Software | G_THREAD: All threads finished
> [  OK  ] DietPi-Software | tar xf master.tar.gz
> [  OK  ] DietPi-Software | rm master.tar.gz
> [  OK  ] DietPi-Software | gcc -g0 -O3 mcrcon-master/mcrcon.c -o /usr/local/bin/mcrcon
> [  OK  ] DietPi-Software | rm -R mcrcon-master
> [  OK  ] DietPi-Software | strip --remove-section=.comment --remove-section=.note /usr/local/bin/mcrcon
> [  OK  ] DietPi-Software | rm -Rf /mnt/dietpi_userdata/papermc/logs
> [  OK  ] DietPi-Software | ln -s /var/log/papermc /mnt/dietpi_userdata/papermc/logs
> [  OK  ] DietPi-Software | chown -R papermc:papermc /mnt/dietpi_userdata/papermc
> [  OK  ] DietPi-Software | systemctl daemon-reload
> [  OK  ] DietPi-Software | Starting PaperMC to pre-create config file in max 1800 seconds
> Journal file /var/log/journal/a3590e13accc44f28e9b8c28320e2d89/user-1000.journal is truncated, ignoring file.
> -- Journal begins at Thu 2022-08-04 12:09:24 BST. --
> Aug 04 12:15:05 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> Aug 04 12:37:09 MMSERVER01 systemd[1]: Started PaperMC (DietPi).
> Aug 04 12:37:09 MMSERVER01 PaperMC[7161]: Minecraft 1.18 requires running the server with Java 17 or above. Download Java 17 (or above) from https://adoptium.net/
> Aug 04 12:37:09 MMSERVER01 systemd[1]: papermc.service: Main process exited, code=exited, status=1/FAILURE
> Aug 04 12:37:09 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> Aug 04 12:37:50 MMSERVER01 systemd[1]: Started PaperMC (DietPi).
> Aug 04 12:37:51 MMSERVER01 PaperMC[7797]: Minecraft 1.18 requires running the server with Java 17 or above. Download Java 17 (or above) from https://adoptium.net/
> Aug 04 12:37:51 MMSERVER01 systemd[1]: papermc.service: Main process exited, code=exited, status=1/FAILURE
> Aug 04 12:37:51 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> Aug 04 16:56:50 MMSERVER01 systemd[1]: Started PaperMC (DietPi).
> Aug 04 16:56:50 MMSERVER01 PaperMC[16155]: Minecraft 1.18 requires running the server with Java 17 or above. Download Java 17 (or above) from https://adoptium.net/
> Aug 04 16:56:50 MMSERVER01 systemd[1]: papermc.service: Main process exited, code=exited, status=1/FAILURE
> Aug 04 16:56:50 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> [  OK  ] DietPi-Software | systemctl stop papermc
> [FAILED] DietPi-Software | Waiting for PaperMC config file failed, skipping pre-configuration

Thanks again for any help.

Joulinar, 2022-08-04:
Hmm the service is not starting at all. Can you share following

`java -version`

blueyshark, 2022-08-04:
clearly a java version that's too low - is that something I should be maintaining, or is that something covered by `dietpi-update`

> 
> openjdk version "11.0.16" 2022-07-19
> OpenJDK Runtime Environment (build 11.0.16+8-post-Raspbian-1deb11u1)
> OpenJDK Server VM (build 11.0.16+8-post-Raspbian-1deb11u1, mixed mode)

MichaIng, 2022-08-04:
Mytherious is that Bullseye ships Java v17 which is also installed as dependency. Probably the older Java version is forced as default runtime by some script you or some installer placed. Can you check:
```
ls -l "$(which java)"
```

blueyshark, 2022-08-04:
That is mysterious, and I have only installed via `dietpi-software`

Here is the output for `ls -l "$(which java)"`
> lrwxrwxrwx 1 root root 22 Jul  1 17:38 /usr/bin/java -> /etc/alternatives/java

MichaIng, 2022-08-04:
Ah another loop:
```sh
readlink -f /etc/alternatives/java
```

blueyshark, 2022-08-04:
Appreciate this help, both of you.

> /usr/lib/jvm/java-11-openjdk-armhf/bin/java

and

> ii  openjdk-11-jre-headless:armhf        11.0.16+8-1~deb11u1              armhf        OpenJDK Java runtime, using Hotspot JIT (headless)

blueyshark, 2022-08-04:
I don't know if this is any help, but I also have installed via `dietpi-software`

> 0    OpenSSH Client: Feature-rich SSH, SFTP and SCP client
> 7    FFmpeg: Audio & video codec libary and programs
> 9    Node.js: JavaScript runtime environment
> 16   Build-Essential: GNU C/C++ compiler, development libraries and headers
> 17   Git: Clone and manage Git repositories locally 
> 46   qBittorrent: BitTorrent server with web interface (C++)
> 50   Syncthing: Backup and sync server with web interface
> 85   Nginx: Lightweight webserver
> 87   SQLite: Persistent single-file database system
> 88   MariaDB: Persistent cached file-per-table database server
> 89   PHP: Hypertext Preprocessor for dynamic web content
> 90   phpMyAdmin: Optional MariaDB web interface admin tools
> 93   Pi-hole: block adverts for any device on your network
> 94   ProFTPD: Efficient, lightweight FTP server
> 103  DietPi-RAMlog: Makes /var/log a RAM disk, preserves file structure on reboot
> 105  OpenSSH Server: Feature-rich SSH server with SFTP and SCP support
> 130  Python 3: Runtime system, pip package installer and development headers
> 178  Jellyfin: FOSS web interface media streaming server
> 181  PaperMC: Highly optimised Minecraft server with plugins, written in Java
> 182  Unbound: validating, recursive, caching DNS resolver
> 188  Go: Runtime environment and package installer
> 195  youtube-dl: Download videos from YouTube and other video sites
> 196  Java JRE: OpenJDK Runtime Environment

MichaIng, 2022-08-04:
> ii openjdk-11-jre-headless:armhf 11.0.16+8-1~deb11u1 armhf OpenJDK Java runtime, using Hotspot JIT (headless)

This was the only output line? Do you have the full PaperMC install output? Java (17) should have been installed in the same run before PaperMC.

blueyshark, 2022-08-04:
Yes, the only output line.

blueyshark, 2022-08-04:
@MichaIng hi there, here is the whole output, it looks like it never checked for java 17?

> root@MMSERVER01:~# dietpi-software
> [  OK  ] DietPi-Software | Initialised database
> [  OK  ] DietPi-Software | Reading database
> [  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Step: Checking for conflicts and missing inputs
> 
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Step: Checking for prerequisite software
> 
> [ INFO ] DietPi-Software | Java JRE will be reinstalled
> [  OK  ] DietPi-Software | Checking IPv4 network connectivity
> [  OK  ] DietPi-Software | Checking DNS resolver
> [  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
> [  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
> [  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
> [ SUB1 ] DietPi-Services > unmask 
> [  OK  ] DietPi-Services | unmask : proftpd
> [  OK  ] DietPi-Services | unmask : mariadb
> [  OK  ] DietPi-Services | unmask : php7.4-fpm
> [  OK  ] DietPi-Services | unmask : lighttpd
> [  OK  ] DietPi-Services | unmask : qbittorrent
> [  OK  ] DietPi-Services | unmask : jellyfin
> [  OK  ] DietPi-Services | unmask : syncthing
> [  OK  ] DietPi-Services | unmask : cron
> [  OK  ] DietPi-Services | unmask : ttfm_bot01_shscpt
> [ SUB1 ] DietPi-Services > stop 
> [  OK  ] DietPi-Services | stop : ttfm_bot01_shscpt
> [  OK  ] DietPi-Services | stop : cron
> [  OK  ] DietPi-Services | stop : syncthing
> [  OK  ] DietPi-Services | stop : jellyfin
> [  OK  ] DietPi-Services | stop : qbittorrent
> [  OK  ] DietPi-Services | stop : lighttpd
> [  OK  ] DietPi-Services | stop : php7.4-fpm
> [  OK  ] DietPi-Services | stop : mariadb
> [  OK  ] DietPi-Services | stop : proftpd
> [ INFO ] DietPi-Software | APT update, please wait...
> Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
> Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
> Hit:3 https://repo.jellyfin.org/debian bullseye InRelease
> Reading package lists...
> [  OK  ] DietPi-Software | APT update
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Step: Installing Java JRE: OpenJDK Runtime Environment
> 
> [ INFO ] DietPi-Software | APT install for: openjdk-11-jre-headless, please wait...
> Reading package lists...
> Building dependency tree...
> Reading state information...
> openjdk-11-jre-headless is already the newest version (11.0.16+8-1~deb11u1).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> [  OK  ] DietPi-Software | APT install for: openjdk-11-jre-headless
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Step: Installing PaperMC: Highly optimised Minecraft server with plugins, written in Java
> 
> [  OK  ] DietPi-Software | cd /tmp/DietPi-Software
> [  OK  ] DietPi-Software | curl -sSfL https://papermc.io/api/v2/projects/paper/versions/1.19.1/builds/109/downloads/paper-1.19.1-109.jar -o paper-1.19.1-109.jar
> [  OK  ] DietPi-Software | mkdir -p /opt/papermc
> [  OK  ] DietPi-Software | mv paper-1.19.1-109.jar /opt/papermc/paperclip.jar
> [  OK  ] DietPi-Software | mkdir /mnt/dietpi_userdata/papermc
[... 17 lines cut ...]
> [  OK  ] DietPi-Software | gcc -g0 -O3 mcrcon-master/mcrcon.c -o /usr/local/bin/mcrcon
> [  OK  ] DietPi-Software | rm -R mcrcon-master
> [  OK  ] DietPi-Software | strip --remove-section=.comment --remove-section=.note /usr/local/bin/mcrcon
> [  OK  ] DietPi-Software | rm -Rf /mnt/dietpi_userdata/papermc/logs
> [  OK  ] DietPi-Software | ln -s /var/log/papermc /mnt/dietpi_userdata/papermc/logs
> [  OK  ] DietPi-Software | chown -R papermc:papermc /mnt/dietpi_userdata/papermc
> [  OK  ] DietPi-Software | systemctl daemon-reload
> [  OK  ] DietPi-Software | Starting PaperMC to pre-create config file in max 1800 seconds
> Journal file /var/log/journal/a3590e13accc44f28e9b8c28320e2d89/user-1000.journal is truncated, ignoring file.
> -- Journal begins at Thu 2022-08-04 12:09:24 BST. --
> Aug 04 12:37:51 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> Aug 04 16:56:50 MMSERVER01 systemd[1]: Started PaperMC (DietPi).
> Aug 04 16:56:50 MMSERVER01 PaperMC[16155]: Minecraft 1.18 requires running the server with Java 17 or above. Download Java 17 (or above) from https://adoptium.net/
> Aug 04 16:56:50 MMSERVER01 systemd[1]: papermc.service: Main process exited, code=exited, status=1/FAILURE
> Aug 04 16:56:50 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> Aug 04 16:57:47 MMSERVER01 systemd[1]: Started PaperMC (DietPi).
> Aug 04 16:57:47 MMSERVER01 PaperMC[16780]: Minecraft 1.18 requires running the server with Java 17 or above. Download Java 17 (or above) from https://adoptium.net/
> Aug 04 16:57:47 MMSERVER01 systemd[1]: papermc.service: Main process exited, code=exited, status=1/FAILURE
> Aug 04 16:57:47 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> Aug 04 20:59:18 MMSERVER01 systemd[1]: Started PaperMC (DietPi).
> Aug 04 20:59:18 MMSERVER01 PaperMC[1151]: Minecraft 1.18 requires running the server with Java 17 or above. Download Java 17 (or above) from https://adoptium.net/
> Aug 04 20:59:18 MMSERVER01 systemd[1]: papermc.service: Main process exited, code=exited, status=1/FAILURE
> Aug 04 20:59:18 MMSERVER01 systemd[1]: papermc.service: Failed with result 'exit-code'.
> [  OK  ] DietPi-Software | systemctl stop papermc
> [FAILED] DietPi-Software | Waiting for PaperMC config file failed, skipping pre-configuration

MichaIng, 2022-08-04:
Ah, you're on a Raspbian-based system (ARMv6 image) and Raspbian has no Java 17 builds, hence we install Java 11 there. So PaperMC actually shouldn't be available on theses systems, but we missed to disable it correctly.

However, since you have it installed already, let's try to install Java 17 from the Debian repository manually:
```sh
cd /tmp
curl -sSfL 'https://deb.debian.org/debian/pool/main/o/openjdk-17/openjdk-17-jre-headless_17.0.3+7-1~deb11u1_armhf.deb' -o package.deb
apt install ./package.deb
rm package.deb
java -version
```

If this works well, we could host them on our server to be pulled on ARMv7/8 capable RPi models with ARMv6 (Raspbian-based) image, so PaperMC can work.

blueyshark, 2022-08-04:
I Installed Java 17 manually as you suggested @MichaIng :

`root@MMSERVER01:/tmp# java -version
openjdk version "17.0.3" 2022-04-19
OpenJDK Runtime Environment (build 17.0.3+7-Debian-1deb11u1)
OpenJDK Server VM (build 17.0.3+7-Debian-1deb11u1, mixed mode, sharing)
root@MMSERVER01:/tmp# 
`

I will now go ahead with trying to install paperMC again :crossed_fingers:

MichaIng, 2022-08-04:
Jep, that should work.

blueyshark, 2022-08-04:
A little confusing, as it still came up with Java 17 warnings, but has worked.

Joulinar, 2022-08-04:
Could you share that warnings pls

blueyshark, 2022-08-04:
@Joulinar Hi there, very strange, I actually reinstalled as I removed it for some minecraft based issues (not to do with DietPi)
Here is the output, this time, NO WARNINGS. Haha, Curious.

```
root@MMSERVER01:~# dietpi-software install 181  
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing PaperMC: Highly optimised Minecraft server with plugins, written in Java
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | Java JRE will be reinstalled
[  OK  ] DietPi-Software | Checking IPv4 network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > unmask 
[  OK  ] DietPi-Services | unmask : proftpd
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.4-fpm
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : qbittorrent
[  OK  ] DietPi-Services | unmask : jellyfin
[  OK  ] DietPi-Services | unmask : syncthing
[  OK  ] DietPi-Services | unmask : cuberite
[  OK  ] DietPi-Services | unmask : cron
[  OK  ] DietPi-Services | unmask : ttfm_bot01_shscpt
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : ttfm_bot01_shscpt
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : cuberite
[  OK  ] DietPi-Services | stop : syncthing
[  OK  ] DietPi-Services | stop : jellyfin
[  OK  ] DietPi-Services | stop : qbittorrent
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : proftpd
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease                                                                                                                                         
Hit:3 https://repo.jellyfin.org/debian bullseye InRelease                                                                                                                                               
Reading package lists...                                                                                                                                                                                
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Java JRE: OpenJDK Runtime Environment

[ INFO ] DietPi-Software | APT install for: openjdk-11-jre-headless, please wait...
Reading package lists...
Building dependency tree...                                                                                                                                                                             
Reading state information...                                                                                                                                                                            
openjdk-11-jre-headless is already the newest version (11.0.16+8-1~deb11u1).                                                                                                                            
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.                                                                                                                                          
[  OK  ] DietPi-Software | APT install for: openjdk-11-jre-headless

[... 74 lines cut ...]

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Finalising install

[  OK  ] DietPi-Software | systemctl daemon-reload
2022-08-04 23:31:27 [ INFO ] DietPi-RAMlog | Storing /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store...
2022-08-04 23:31:28 [  OK  ] DietPi-RAMlog | Stored /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store.
[ SUB1 ] DietPi-Services > dietpi_controlled 
[  OK  ] DietPi-Services | dietpi_controlled : proftpd
[  OK  ] DietPi-Services | dietpi_controlled : mariadb
[  OK  ] DietPi-Services | dietpi_controlled : php7.4-fpm
[  OK  ] DietPi-Services | dietpi_controlled : lighttpd
[  OK  ] DietPi-Services | dietpi_controlled : qbittorrent
[  OK  ] DietPi-Services | dietpi_controlled : jellyfin
[  OK  ] DietPi-Services | dietpi_controlled : syncthing
[  OK  ] DietPi-Services | dietpi_controlled : cuberite
[  OK  ] DietPi-Services | dietpi_controlled : papermc
[  OK  ] DietPi-Services | dietpi_controlled : cron
[  OK  ] DietPi-Services | dietpi_controlled : ttfm_bot01_shscpt

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed
```

MichaIng, 2022-08-07:
Okay great. You could btw now remove the old Java 11:
```sh
apt autopurge openjdk-11-jre-headless
```

MichaIng, 2022-11-28:
Fixed with: https://github.com/MichaIng/DietPi/commit/8394606

MichaIng, 2022-11-28:
Woah is this slow, tested with Java 17 Zero JVM on RPi Zero W:
```
INFO --- org.airsonic.player.Application          : Started Application in 4899.567 seconds (JVM running for 5017.986)
```
~1.5h for service startup. Probably also because of limited heap size, of because the BCM2708 is simply too weak.

closed 2022-12-04 by MichaIng