MichaIng/DietPi #5669: Installation of Home Assistan fails "can't find Rust compiler" for maturin

fpMiguel, 2022-08-05:
#### Required Information
- DietPi version | `8.7.1`
- Distro version | `bullseye 1`
- Kernel version | `Linux DietPi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux`
- SBC model | `RPi B+ (armv6l)`
- Power supply used | n/a 
- SD card used | n/a

#### Additional Information (if applicable)
- Software title | Home Assistant
- Was the software title installed freshly or updated/migrated?
Software was installed in the past (including receiving updates). Uninstalled but unable to reinstall again now.
- Can this issue be replicated on a fresh installation of DietPi?
Unable to test
- Bug report ID | 549cf76b-515b-4b02-b55c-dad429717fa7

#### Steps to reproduce
1. Run `dietpi-software install 157`
2. Wait for installation to faul

#### Expected behaviour
- Home Assistant correctly installs all dependencies

#### Actual behaviour
- Installation fails due to error "can't find Rust compiler" while processing `maturin`

#### Extra details
- Might be related to https://github.com/MichaIng/DietPi/issues/5630
- I've also added a comment in the ticket above https://github.com/MichaIng/DietPi/issues/5630#issuecomment-1206000508

Joulinar, 2022-08-05:
Why are you opening a duplicate issue now? This is not going to speed things up.

fpMiguel, 2022-08-05:
I understand creating a duplicate will not speed up the process. However, the other bug report is about manually running an update using the `homeassistant-update.sh` script.
This bug report is about installing Home Assistant through `dietpi-software`.

However, since you added the Duplicate label, I will close this.

Joulinar, 2022-08-05:
well it seems to be the same plugin `maturin` causing the issue.

MichaIng, 2022-08-06:
Jep, failure in both cases have the same reason: New `maturin` Python module is tried to be compiled, which requires Rust. I'll close this in favour of the existing issue.

closed 2022-08-06 by MichaIng