MichaIng/DietPi #5671: Error installing openssh

sirkong2, 2022-08-05:
#### Details:
- Date           | Fri Aug  5 21:28:28 BST 2022
- DietPi version | v8.7.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -6nc 1 -W 10 2620:fe::fe`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
PING 2620:fe::fe(2620:fe::fe) 56 data bytes
From fe80::1%eth0 icmp_seq=1 Destination unreachable: No route

--- 2620:fe::fe ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

Joulinar, 2022-08-05:
You have an issue to connect to an IPv6 address. Therefore our connection test is failing. On the error menu, there should be an option to disable IPv6. Can you check this?

sirkong2, 2022-08-05:
Thanks @Joulinar I turned IPV6 off on the config menu and that solved the error.

Joulinar, 2022-08-05:
ok, I will close the issue. Feel free to reopen if needed.

closed 2022-08-05 by Joulinar