MichaIng/DietPi #5672: No more changes of the original image by dietpi-imager!

DrCWO, 2022-08-06:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Using **dietpi-imager** script the partition of the original SD-Card that I imaged got smaller. Using 'df' usage of '/' went up from 18% to 74%. My expectation is that the original SD-Card will not be changed by dietpi-imager.

#### Describe the solution you'd like:

- Resize the partition of the saved SD-card - after making the image file - back to its original size.

#### Describe alternatives you've considered:

- I did it manually with parted/resizepart and resize2fs (but this is an avoidable extra task)

#### Additional context

- I think it should be clear so far ;-)

Joulinar, 2022-08-06:
Well this is exactly the purpose of our imager script to shrink the image as much as possible. You could use our driver manager to resize the image afterwards.

DrCWO, 2022-08-06:
Thanls' for your reply.
I think we have a different understanding of what an imaging tool should do ;-)
In my understanding I have after image creation and burn on a new SD-card a 1:1 copy of the master.

In my eyes the master SD-card is a MASTER and should not be changed at all. No software shall change the MASTER!
Sure shrinking is what has to be done to create an image and keep the file small.
In the end I expect that the master was untouched and the image is exactly like the master if I copy the image on a SD-Card with Balena Etcher. 

This means the right thing would be:
1. Change the size back to the original size (if you have to modify the master)
2. Change the size if the copied image on the new SD-card to the original size right after Boot.

Until now I used **image-backup** on Raspbian and that works as I think it should. 
It doesn't change the master and at the end the partition of the SD-Card copy got expanded during reboot.

Joulinar, 2022-08-06:
Before creating the image, you simply could activate the fs realize service. This would extend the partition on first boot automatically. The imager script we use to shrink our Images to have smaller download files. If the script did not fit your expectations, simply use W32 disk imager. This will create a 1:1 copy of the master SDcard.

DrCWO, 2022-08-06:
Thank you, fs resize servcice is a good option I think :-)

MichaIng, 2022-08-06:
I agree it would be nice if the original drive wouldn't be touched but only the image file created from it. However, it is not so trivial:
- When creating an image from a drive via `dd`, it will have a size matching the whole drive size, which is not acceptable and will take ages for nowadays common SD card, HDD and SSD sizes, apart of the massive unnecessary disk writes and temporary space requirements.
- `dd` could be instructed to abort copying at the offset based on `resize2fs -P` output. But we shrink the filesystem in multiple steps in a loop as `resize2fs -M` for some reason isn't able to shrink it in a single step. So copying size etc will still be larger and at least temporarily it will be an invalid image file with oversized partition and filesystem, where I'm not sure whether `resize2fs` will be actually able to work on. Would require testing.
- Easiest solution would be probably to either re-expand the original partition and filesystem after the image has been created and/or enable the `dietpi-fs_partition_resize.service` at original drive and/or generated image. Enabling that service anyway should be what is always wanted for the shrunk image, and even that the original drive is still touched, booting it reverts the change automatically then.

Joulinar, 2022-08-06:
personally I would vote for enabling `dietpi-fs_partition_resize.service` automatically while creating an image while using `dietpi-imager`

DrCWO, 2022-08-07:
Thank you very much for you in depth explanation :-)

I will try enabling the  dietpi-fs_partition_resize.service on the original. rebooting it shall lead to a partition in size of the sd-card. Same hopefully on the copy. Will test during upcoming week.

Best DrCWO

MichaIng, 2022-08-25:
Done: https://github.com/MichaIng/DietPi/commit/6b74f2c

The service is now enabled (if now enabled yet) on the original system before any shrinking is done. So it is enabled on the original system as well as on the newly created image file.

closed 2022-08-28 by MichaIng