MichaIng/DietPi #5673: missing some app choices - feature request

surtarso, 2022-08-07:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- yacy is tad too heavy for most home users, there is no centralized network dash, only a bunch of complementary ones, there is no storage related solution (raid etc), only photo webapp is too simple and somewhat buggy.

#### Describe the solution you'd like:

- having more option for software installs

#### Describe alternatives you've considered:

- I 'd LOVE to see: Searx, PhotoPrism, Dashy and Mergerfs as options on the software menu for install

#### Additional context

- I've been using dietpi on a rpi for a long time and I love it. I am now using it on a native x86 pc and im stating to feel there could be some more improvement. BTW, searx should be a lot more lightweight than yacy!

Joulinar, 2022-08-07:
We offer over 200 software title on our catalogue. As well you are free to install own software packages manually as on every other Debian OS. If you have specific software request, fell free to open individual software request and fill the specific request form.

MichaIng, 2022-08-07:
For Searx there is an old request which could be reactivated: #1050
For Dashy there is one already: #5494

For the others, please open individual issues, as stated by Joulinar.

surtarso, 2022-08-07:
oh, ok, sorry for that! 
Btw I didn't mean in any way to downgrade dietpi as it is. It is a PHENOMENAL piece of software that pretty much runs my home ;)
I just upgraded my devices so I was after a couple of more robust solutions (of course, diet pi.. is DIET) but as an example, searx seems a lot lighter than yacy. idk about dashy, and photoprims is obviously fatter than single php gallery. But since there are some solutions with multiple software choices, I just thought I'd give my 2 cents as an avid dietpi user, to see a couple more =)
I know I can install anything in the underlying OS, but the way diet-pi keeps them nice and tidy makes me feel happy, warm and safe, that's why I made the suggestions I did. =) 
Also, I hate dealing with multiple containers, and most of those suggestions are docker related. I much rather have diet-pi manage them for me like the champ it is! =)

closed 2022-08-07 by surtarso