MichaIng/DietPi #5677: WireGuard not work on NanoPi R5S

3735943886, 2022-08-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye`
- Kernel version | `uname -a`
`Linux DietPi 5.10.110 #81 SMP Mon Jul 4 15:26:21 CST 2022 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`NanoPi R5S (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
`5V 2A`
- SD card used | (EG: SanDisk ultra)
`None`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`WireGuard`
- Was the software title installed freshly or updated/migrated?
`Installed freshly`
- Can this issue be replicated on a fresh installation of DietPi?
`Yes`

#### Steps to reproduce

1. `sudo apt install wireguard`
2. `sudo reboot`
3. create wg0.conf file
4. `sudo wg-quick up wg0`

#### Expected behaviour

- wireguard launches normally.

#### Actual behaviour

- wireguard fails to launch.
- command `ip link add wg0 type wireguard` gives an error `Error: Unknown device type.`

Joulinar, 2022-08-08:
Why not using `dietpi-software` to install Wireguard? This will setup everything needed.

MichaIng, 2022-08-08:
I'll check whether the module is actually shipped with the kernel. Would be bad if not, given that it's a router device. Otherwise we need to host the kernel headers package, which isn't a big issue.

3735943886, 2022-08-09:
> Why not using `dietpi-software` to install Wireguard? This will setup everything needed.

I'm afraid but I couldn't find wireguard from dietpi-software.
Shall I use PiVPN?
[image]

> I'll check whether the module is actually shipped with the kernel. Would be bad if not, given that it's a router device. Otherwise we need to host the kernel headers package, which isn't a big issue.

Thank you for prompt answer.

Joulinar, 2022-08-09:
Ah probably it's disabled. Can you check

`dietpi-software list | grep wire`

3735943886, 2022-08-09:
> Ah probably it's disabled. Can you check
> 
> `dietpi-software list | grep wire`

```
dietpi@DietPi:~$ sudo dietpi-software list | grep wire
ID 117 | =0 | PiVPN: openvpn/wireguard server install & management tool | +Git | https://dietpi.com/docs/software/vpn/#pivpn
ID 172 | =0 | WireGuard: an extremely simple yet fast and modern VPN | DISABLED for NanoPi R5S (aarch64) | https://dietpi.com/docs/software/vpn/#wireguard
```

It was disabled, as expected.

Joulinar, 2022-08-09:
ok can you check if the kernel module is shipped with actual kernel version?

```
dpkg-query -S '/lib/modules/*/wireguard.ko*'
modinfo wireguard
```

3735943886, 2022-08-10:
> ok can you check if the kernel module is shipped with actual kernel version?
> 
> ```
> dpkg-query -S '/lib/modules/*/wireguard.ko*'
> modinfo wireguard
> ```
```
dietpi@DietPi:~$ dpkg-query -S '/lib/modules/*/wireguard.ko*'
dpkg-query: no path found matching pattern /lib/modules/*/wireguard.ko*
dietpi@DietPi:~$ sudo modinfo wireguard
modinfo: ERROR: Module wireguard not found.
dietpi@DietPi:~$
```

I noticed that wireguard tried to install linux-image during dependency check. And I guess it's an unusual case.
```
dietpi@DietPi:~$ sudo apt install wireguard
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  cpio initramfs-tools initramfs-tools-core klibc-utils libklibc linux-base
  linux-image-5.10.0-16-rt-arm64 linux-image-rt-arm64 wireguard-tools
Suggested packages:
  libarchive1 linux-doc-5.10 debian-kernel-handbook openresolv | resolvconf
Recommended packages:
  busybox | busybox-static pigz apparmor
The following NEW packages will be installed:
  cpio initramfs-tools initramfs-tools-core klibc-utils libklibc linux-base
  linux-image-5.10.0-16-rt-arm64 linux-image-rt-arm64 wireguard
  wireguard-tools
0 upgraded, 10 newly installed, 0 to remove and 0 not upgraded.
Need to get 45.4 MB of archives.
After this operation, 255 MB of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.
dietpi@DietPi:~$
```

Joulinar, 2022-08-11:
Ok something we would need to check if and how the kernel header package is available.

Joulinar, 2022-08-13:
@MichaIng 
did you already find time to check for the headers package? There is another user on the forum looking for this package https://dietpi.com/forum/t/unable-to-install-linux-headers/13971

MichaIng, 2022-08-19:
Please try this:
```sh
cd /tmp
curl -O 'https://dietpi.com/downloads/binaries/linux-headers-nanopi5.deb'
dpkg -i linux-headers-nanopi5.deb
rm linux-headers-nanopi5.deb
sed -i 's/13]=0/76]=1/' /boot/dietpi/dietpi-software
dietpi-software install 172
```
.... okay no idea, it fails here with two strange errors:
```sh
------------------------------
Deleting module version: 1.0.20210219
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-1.0.20210219 DKMS files...
It is likely that 5.10.110 belongs to a chroot's host
Building for 5.10.110
Building initial module for 5.10.110
Error!  The /var/lib/dkms/wireguard/1.0.20210219/5.10.110/aarch64/dkms.conf for module wireguard includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.
```
I hope the new image does not have these strange issue, being seen as chroot by DKMS and thinking the kernel/arch does not match...

Joulinar, 2022-08-20:
> it fails here with two strange errors:

Yup fails on my device as well. Let's hope you are able to create a new image. 👍

HeyMeco, 2022-09-10:
> @MichaIng
> did you already find time to check for the headers package? There is another user on the forum looking for this package https://dietpi.com/forum/t/unable-to-install-linux-headers/13971

@MichaIng if you by any chance haven't seen it:
The "Debian 10(buster) Desktop" image from friendlyelec's google drive should have the right headers.deb in /opt/

else this https://github.com/friendlyarm/sd-fuse_rk3568/blob/master/test/test-build-kernel-header-deb.sh looks like it could be easily modified with the dietpi image

HeyMeco, 2022-09-10:
> ```shell
> cd /tmp
> curl -O 'https://dietpi.com/downloads/binaries/linux-headers-nanopi5.deb'
> dpkg -i linux-headers-nanopi5.deb
> rm linux-headers-nanopi5.deb
> sed -i 's/13]=0/76]=1/' /boot/dietpi/dietpi-software
> dietpi-software install 172
> ```

Nvm I just saw that you guys were already that far.

HeyMeco, 2022-09-10:
Okay I think I found the reason why it's not included: https://github.com/ariaboard-com/rockchip_rk3568_buildroot/blob/rk356x-novotech/configs/rockchip_rk3568_defconfig

doesn't have the:
"BR2_PACKAGE_WIREGUARD_LINUX_COMPAT=y
BR2_PACKAGE_WIREGUARD_TOOLS=y"
set.

I couldn't find "friendlyelec_rk3568_defconfig" but since FriendlyElec's Buildroot is based on Rockchips that error probably made it through to the kernel we use.

MichaIng, 2022-10-31:
I generated a new image: https://dietpi.com/downloads/images/testing/DietPi_NanoPiR5S-ARMv8-Bullseye.7z
This is still based on FriendlyELEC image, but the FriendlyWRT one, which contains the WireGuard module. However, I completely recreated the root filesystem via `debootstrap` and copied only the kernel modules from FriendlyWRT over.

Boots and works fine so far. What somehow does not work anymore is the Ethernet LEDs. This is probably related to the fact that I didn't add any custom firmware from here: https://github.com/friendlyarm/sd-fuse_rk3568/tree/master/prebuilt/firmware
Testing now whether I can fix this with some particular firmware files.

MichaIng, 2022-10-31:
Found it:
```sh
modprobe ledtrig-netdev
```

Joulinar, 2022-10-31:
ahh I'm away from home but with remote access to my R5S. I guess I'm not able to flash it until I'm back 🤣

MichaIng, 2022-10-31:
Good to know: All the firmware on FriendlyELEC's image match the ones from the Debian `firmware-` packages 100% in size, as far as I checked. So it's really only the kernel modules which we need to place into the rootfs, aside of plain Debian.

HeyMeco, 2022-11-05:
Is https://dietpi.com/downloads/images/testing/DietPi_NanoPiR5S-ARMv8-Bullseye.7z the newest iteration of what needs testing? I can check it on mine

MichaIng, 2022-11-05:
Yes, it would be great if you could give it a try.

HeyMeco, 2022-11-05:
Running  `dpkg-query -S '/lib/modules/*/wireguard.ko*'
modinfo wireguard` results in:

firmware-nanopi5: /lib/modules/5.10.110/wireguard.ko
filename:       /lib/modules/5.10.110/wireguard.ko
alias:          net-pf-16-proto-16-family-wireguard
alias:          rtnl-link-wireguard
version:        1.0.0
author:         Jason A. Donenfeld <Jason@zx2c4 .com>  //_space added manually by me_
description:    WireGuard secure network tunnel
license:        GPL v2
srcversion:     4198DFE47D68B6762A7E633
depends:        libcurve25519-generic,libblake2s,udp_tunnel,libchacha20poly1305,ip6_udp_tunnel
intree:         Y
name:           wireguard
vermagic:       5.10.110 SMP mod_unload modversions aarch64

and doing a `systemctl start wg-quick@...` worked. Confirmed it with speedtest-cli during which htop reported around 45% CPU Usage

MichaIng, 2022-11-05:
Great. So I can move this to stable downloads.

Btw, one issue I still have with the Ethernet LEDs:
- They all are lit on boot even if no cable is connected.
- Once I attach and detach a cable, they correctly turn off, and on again once cable is re-connected.

Based on this issue btw: https://github.com/MichaIng/DietPi/issues/5679
- The `/sys/class/leds/*_led/link` can be used to turn on/off each light, but when turning it off, it doesn't automatically turn on again when attaching a cable, only the other way round.
- Also on boot it needs to be turned on manually once to react to attaching/detaching Ethernet cables.
- What is missing is to have them listening on cable connection without forcing them lit on boot, or otherwise a way to turn off the LEDs again if no cable is connected, without breaking them to lit again once a cable is connected.

HeyMeco, 2022-11-05:
Not sure if it was a one time thing but I installed wireguard with the initial install dialogue. When that whole process was completed for some reason my R5S couldn't resolve DNS until I rebooted it.

MichaIng, 2022-11-05:
Hmm, not sure. The WireGuard server (start) shouldn't have any effect on the host systems DNS, which is defined only by `/etc/resolv.conf`. I'll try to replicate.

3735943886, 2022-11-09:
> Great. So I can move this to stable downloads.

Thank you for great work. 

> Btw, one issue I still have with the Ethernet LEDs:
> 
> - They all are lit on boot even if no cable is connected.
> 
> - Once I attach and detach a cable, they correctly turn off, and on again once cable is re-connected.
> 
> 
> 
> Based on this issue btw: https://github.com/MichaIng/DietPi/issues/5679
> 
> - The `/sys/class/leds/*_led/link` can be used to turn on/off each light, but when turning it off, it doesn't automatically turn on again when attaching a cable, only the other way round.
> 
> - Also on boot it needs to be turned on manually once to react to attaching/detaching Ethernet cables.
> 
> - What is missing is to have them listening on cable connection without forcing them lit on boot, or otherwise a way to turn off the LEDs again if no cable is connected, without breaking them to lit again once a cable is connected.
> 
> 
> 
> This issue however seems to be the same on official FriendlyELEC images, at least they do exactly the same we do.

ifplugd might be helpful?

MichaIng, 2022-11-09:
`ifplugd` can be configured to run commands if cables are connected or disconnected. We can also check for cable connectivity via `ethtool`. However, the problem really is that I'm not aware of any command which could do what we need, using the `ledtrig-netdev` kernel driver. Skipping the kernel driver (and doing everything manually with ifplugd) isn't possible either, because the driver needs to be loaded for the ability to toggle the LEDs.

It really is only needed to somehow turn off the LED once at boot, is no cable is connected, without disabling the kernel drivers autodetection, or, turning on autodetection without forcefully turning on the LED 🤔.

Joulinar, 2022-11-09:
Not a solution, but I disabled all LED completely 🤣

closed 2022-11-12 by MichaIng