MichaIng/DietPi #5680: RK3399 Based image – unable to get /boot

VladoPortos, 2022-08-10:
Hello all,
I have a strange issue. I have flashed 9x rpi 4 Diet Pi image without single issue. Every time I flash it to the USB from ISO I reseat the USB and get the boot partition where I can add my custom `dietpi.txt`. However, when I do the same with any of the RK3399 Based ISOs the boot partition will not show up.. I have tried multiple images. I'm doing this on Windows.

Joulinar, 2022-08-10:
This is as expected because the file system layout if quite different. On RPi you have a separate vfat partition that could be accessed by Windows. On other images you have ext4 only that Windows is not able to read

VladoPortos, 2022-08-10:
@Joulinar ah makes sense.

closed 2022-08-10 by VladoPortos