MichaIng/DietPi #5681: Vaultwaren update Exit code: 100

anubis-genix, 2022-08-10:
Did an update from 8.6.1 to 8.7.1 first.
Wanted to update vaultwarden to the latest version also with "dietpi-software reinstall 183".
Got this instead:

 APT install for: ./vaultwarden_aarch64.deb
│  - Command: apt-get -y --allow-change-held-packages install ./vaultwarden_aarch64.deb
│  - Exit code: 100
│  - DietPi version: v8.7.1 (MichaIng/master) | HW_MODEL: 67 | HW_ARCH: 3 | DISTRO: 5
│  - Image creator: DietPi Core Team
│  - Pre-image: Armbian
│  - Error log:
│ Reading package lists...
│ Building dependency tree...
│ Reading state information...
│ Some packages could not be installed. This may mean that you have
│ requested an impossible situation or if you are using the unstable
│ distribution that some required packages have not yet been created
│ or been moved out of Incoming.
│ The following information may help to resolve the situation:
│
 The following packages have unmet dependencies:
│  vaultwarden : Depends: bash (>= 5.1-2+deb11u1) but 5.0-4 is to be installed
│                Depends: libc6 (>= 2.31-13+deb11u3) but 2.28-10+deb10u1 is to be installed
│                Depends: openssl (>= 1.1.1n-0+deb11u3) but 1.1.1n-0+deb10u3 is to be installed
│                Depends: libssl1.1 (>= 1.1.1n-0+deb11u3) but 1.1.1n-0+deb10u3 is to be installed
│ E: Unable to correct problems, you have held broken packages.

Restoring backup as I'm writing this issue...

Joulinar, 2022-08-10:
Looks like a bug from pur side. Actually you are running Buster, however Vaultwarden package is looking for Bullseye dependency. 🤔 

Could you try following

```
cd /tmp
curl -sSfL https://dietpi.com/downloads/binaries/buster/vaultwarden_aarch64.deb -o vaultwarden_aarch64.deb
apt install ./vaultwarden_aarch64.deb -y
rm ./vaultwarden_aarch64.deb
```

anubis-genix, 2022-08-11:
Installed the live patch and performed the update.
Even though the update went fine my vaultwarden install is still broken.

DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | SQLite will be reinstalled
[  OK  ] DietPi-Software | Checking IPv4 network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : redis-server
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.3-fpm
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : vaultwarden
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : vaultwarden
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian buster InRelease
Get:2 https://deb.debian.org/debian buster-updates InRelease [56.6 kB]
Get:3 https://deb.debian.org/debian-security buster/updates InRelease [34.8 kB]
Get:4 https://deb.debian.org/debian buster-backports InRelease [51.4 kB]
Hit:5 http://mirrors.dotsrc.org/armbian-apt buster InRelease
Get:6 https://deb.debian.org/debian-security buster/updates/main arm64 Packages [332 kB]
Fetched 475 kB in 4s (134 kB/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing SQLite: Persistent single-file database system

[ INFO ] DietPi-Software | APT install for: sqlite3 php7.3-sqlite3, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
php7.3-sqlite3 is already the newest version (7.3.31-1~deb10u1).
sqlite3 is already the newest version (3.27.2-3+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
[  OK  ] DietPi-Software | APT install for: sqlite3 php7.3-sqlite3

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing vaultwarden: Unofficial Bitwarden password manager server written in Rust

[  OK  ] DietPi-Software | rm -R /opt/vaultwarden
[  OK  ] DietPi-Software | rm /etc/systemd/system/vaultwarden.service
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://dietpi.com/downloads/binaries/buster/vaultwarden_aarch64.deb -o vaultwarden_aarch64.deb
[ INFO ] DietPi-Software | APT install for: ./vaultwarden_aarch64.deb, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  vaultwarden
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
[... 698 lines cut ...]
Aug 11 21:19:08 DietPi vaultwarden[733]: [WARNING] SIGNUPS_ALLOWED
Aug 11 21:19:08 DietPi vaultwarden[733]: [DEPRECATED]: `SMTP_SSL` or `SMTP_EXPLICIT_TLS` is set. Please use `SMTP_SECURITY` instead.
Aug 11 21:19:09 DietPi vaultwarden[733]: [2022-08-11 21:19:09.277][_][WARN] Detected TLS-enabled liftoff without enabling HSTS.
Aug 11 21:19:09 DietPi vaultwarden[733]: [2022-08-11 21:19:09.277][_][WARN] Shield has enabled a default HSTS policy.
Aug 11 21:19:09 DietPi vaultwarden[733]: [2022-08-11 21:19:09.284][start][INFO] Rocket has launched from https://127.0.0.1:8001
Aug 11 21:19:21 DietPi sshd[779]: Accepted password for root from 192.168.X.X port 56815 ssh2
Aug 11 21:19:21 DietPi sshd[779]: pam_unix(sshd:session): session opened for user root by (uid=0)
Aug 11 21:19:39 DietPi systemd-timesyncd[398]: Synchronized to time server for the first time 85.220.190.246:123 (2.debian.pool.ntp.org).
Aug 11 21:19:40 DietPi systemd[1]: Stopping Network Time Synchronization...
Aug 11 21:19:40 DietPi systemd[1]: systemd-timesyncd.service: Succeeded.
Aug 11 21:19:40 DietPi systemd[1]: Stopped Network Time Synchronization.
Aug 11 21:19:45 DietPi sudo[1249]:     root : TTY=pts/0 ; PWD=/root ; USER=root ; COMMAND=/usr/bin/journalctl
Aug 11 21:19:45 DietPi sudo[1249]: pam_unix(sudo:session): session opened for user root by root(uid=0)
Aug 11 21:19:51 DietPi CRON[742]: pam_unix(cron:session): session closed for user root
Aug 11 21:20:01 DietPi CRON[1287]: pam_unix(cron:session): session opened for user www-data by (uid=0)
Aug 11 21:20:01 DietPi CRON[1286]: pam_unix(cron:session): session opened for user root by (uid=0)
Aug 11 21:20:01 DietPi CRON[1288]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Aug 11 21:20:01 DietPi CRON[1289]: (root) CMD (   PATH="$PATH:/usr/sbin:/usr/local/bin/" pihole updatechecker local)
Aug 11 21:20:02 DietPi CRON[1286]: pam_unix(cron:session): session closed for user root
Aug 11 21:20:10 DietPi CRON[1287]: pam_unix(cron:session): session closed for user www-data
Aug 11 21:20:27 DietPi sudo[1249]: pam_unix(sudo:session): session closed for user root
Aug 11 21:21:43 DietPi sudo[1315]:     root : TTY=pts/0 ; PWD=/root ; USER=root ; COMMAND=/usr/bin/journalctl vaultwarden
Aug 11 21:21:43 DietPi sudo[1315]: pam_unix(sudo:session): session opened for user root by root(uid=0)
Aug 11 21:21:43 DietPi sudo[1315]: pam_unix(sudo:session): session closed for user root
root@DietPi:~#

Joulinar, 2022-08-11:
Valtwarden is listen to localhost only

```
Rocket has launched from https://127.0.0.1:8001/
```

Therefore you are not able to reach the service. This is a change Vaultwarden guys did on one of the last versions to increase security. I guess they assume most of people will use a revers proxy.

pls try follwoing

```
GCI_PRESERVE=1 G_CONFIG_INJECT 'ROCKET_ADDRESS=' 'ROCKET_ADDRESS=0.0.0.0' /mnt/dietpi_userdata/vaultwarden/vaultwarden.env
systemctl restart vaultwarden
```

anubis-genix, 2022-08-11:
Thanks @Joulinar it worked!

I thought another service might have been broken too in the process. That's why I pasted the whole thing ;)

MichaIng, 2022-08-12:
Ah right that is another thing we need to add to the package's postinst script for upgrades from before v1.25.0, where the default has changed (and I'm still not sure whether this was intended or not).