MichaIng/DietPi #5683: Erro while installing LXDE desktop

Getafix17, 2022-08-11:
### Creating a bug report/issue

#### Required Information
#### Details:
- Date           | Thu Aug 11 09:03:11 IST 2022
- DietPi version | v8.7.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPiSSD 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `ping -6nc 1 -W 10 2620:fe::fe`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
PING 2620:fe::fe(2620:fe::fe) 56 data bytes
From fe80::eead:e0ff:fed0:5dbb%eth0 icmp_seq=1 Destination unreachable: No route

--- 2620:fe::fe ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

Joulinar, 2022-08-11:
You have an issue to connect via IPv6. Directly on the error screen, there should be an option to disable IPv6. Pls try again.

Getafix17, 2022-08-11:
Thank You! Installed it.

closed 2022-08-11 by Getafix17