MichaIng/DietPi #5687: dietpi-imager without 7z PLEASE

DrCWO, 2022-08-12:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I use the **dietpi-imager** script and dislike it creates a 7z file as the creation takes very long on a SD-card and the file cannot be used as a source for BalenaEtcher

#### Describe the solution you'd like:

- I would appreciate if there will be a new option in the dialog to select 7z/zip/OFF. Off will do no compression at all.

#### Describe alternatives you've considered:

- None, this has to be changed in the script

#### Additional context

- None

MichaIng, 2022-08-13:
You can already do:
```sh
SHRINK_ONLY=On ./dietpi-imager
```
But we can make a menu option for it. Also the variable name is misleading as it does not only shrink the filesystem/partition but creates the image file as well. I'll rename it to `SKIP_ARCHIVE`.

DrCWO, 2022-08-13:
I tried to set both to be sure. I did:
```
SHRINK_ONLY=On
SKIP_ARCHIVE=On
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"
```
But no success. 7zr was started again :-(

ravenclaw900, 2022-08-13:
If you don't put it on the same line, you have to export the variable:
```
export SHRINK_ONLY=On
```

DrCWO, 2022-08-13:
Goof me, will retry with export ;-)
Thanks'

Joulinar, 2022-08-13:
Btw the value `SKIP_ARCHIVE` did not exist yet

MichaIng, 2022-08-25:
Feature added now: https://github.com/MichaIng/DietPi/commit/6b74f2c

```sh
export SKIP_ARCHIVE=1
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
```
*Note `dev` branch here, until it has been released this Sunday night.*

```sh
export SHRINK_ONLY=1
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
```
*This will now exit already before any image file is created, just after the source image/drive filesystem and partition have been shrunk. Suitable as a preparation for cloning the system from a smaller to a larger drive with a cloning tool.*

closed 2022-08-28 by MichaIng