MichaIng/DietPi #5689: Cannot prevent services from being auto-started after boot

DrCWO, 2022-08-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye

- Kernel version | `uname -a`
Linux rooExtend 5.15.52-rockchip64 #22.05.4 SMP PREEMPT Mon Jul 4 09:40:29 CEST 2022 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Nanopi R2S

- Power supply used | (EG: 5V 1A RAVpower)
5V 2A

- SD card used | (EG: SanDisk ultra)
SanDisk ultra 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
smbd, nmbd, docker

- Was the software title installed freshly or updated/migrated?
yes

- Can this issue be replicated on a fresh installation of DietPi?
-yes

- 

#### Steps to reproduce

1. systemctl disable smbd
2. systemctl disable nmbd
3. systemctl disable docker

#### Expected behaviour

- After reboot these services should be inactive (not running)

#### Actual behaviour

- The services were started again after reboot. See systemctl --type=service

#### Extra details

- I found startup scripts in /etc/init.d also and wonder why. 
- After removing them the services still get started after reboot
- I want to have them running during development. After this I like to disable them
and create a distributes image were they are disabled but not deinstalled after boot.

Joulinar, 2022-08-13:
Simply open `dietpi-services`, select the service in question and set following

```
Mode                    : [Systemd controlled]
Include/Exclude         : [excluded]
```

DrCWO, 2022-08-13:
Great thanks'

DrCWO, 2022-08-13:
I like to tell you that your support is **WORLD CLASS**.
Quick and competent :-):-):-)

Joulinar, 2022-08-13:
Was this working? If yes, we could close the issue, right?

DrCWO, 2022-08-14:
Yes works great. Can close it. Thanks' again.

closed 2022-08-14 by Joulinar