MichaIng/DietPi #5690: v8.7.1 Instability issues on RPi 3

Snurring, 2022-08-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`root@DietPi:~# cat /boot/dietpi/.version`
`G_DIETPI_VERSION_CORE=8`
`G_DIETPI_VERSION_SUB=7`
`G_DIETPI_VERSION_RC=1`
`G_GITBRANCH='master'`
`G_GITOWNER='MichaIng'`
`G_LIVE_PATCH_STATUS[0]='not applicable'`
`root@DietPi:~# echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye 0`
`root@DietPi:~# uname -a`
`Linux DietPi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
`root@DietPi:~# echo $G_HW_MODEL_NAME`
`RPi 3 Model B (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
`STONTRONICS Switching Adapter, Model: DSA-13PFC-05, +5.1V/2.5A "Raspberry"`
- SD card used | (EG: SanDisk ultra)
`Kingston microSD HC4 SDC4/8GB`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
`YES`

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

`1. Headless setup (ssh connection)`
`2. Install newest image and upgrade to 8.7.1`
`3. Let the RPi run for a while. Unexpected behaviour would most likely appear within 24 hours.`
`Check systemd logs regularly | 'journalctl'`

#### Expected behaviour

`- In this basic configuation RPi should run stable for months`

#### Actual behaviour

`- After a while mmc0 write error messages starts appearing in the logs.`
`- Processes can no longer do write operations and starts failing.`
`- Some programs will terminate with I/O errors.`
`- Processes as ssh and networking/dhcp can (will) fail, making communication over LAN impossible.`

#### Extra details

- Tried all the different images (armv6, armv7, armv8) an they behave the same way.
- Tried with different SD-cards, different power adapters,
   different  programming software and multiple RPi´s (a RPi2 and 2 RPi3's).
- Installed Armbian and it is running stable on same SD's/RPi's. 
- Forced rebooting clears the errors and system will start working normally until the faults appears again.

Journalctl: 
Aug 04 20:31:20 DietPi kernel: mmc0: timeout waiting for hardware interrupt.
Aug 04 20:31:20 DietPi kernel: [09177b64] TCM> 823fbb7c 0
Aug 04 20:31:20 DietPi kernel: [09177b64] FDA> 823fbaa8 823fbb4c
Aug 04 20:31:20 DietPi kernel: [09177b65] DMA> 0 0
Aug 04 20:31:20 DietPi kernel: [09177b83] TSK< 823fbaa8 0
Aug 04 20:31:20 DietPi kernel: [09177b8f] TSK> 823fbaa8 0
Aug 04 20:31:20 DietPi kernel: [c9177bbf] PRD< 823fbd7c 0
Aug 04 20:31:20 DietPi kernel: [c9177bc0] PRD1 817cf840 0
Aug 04 20:31:20 DietPi kernel: [c9177bd0] PRD2 10 0
Aug 04 20:31:20 DietPi kernel: [c9177bf5] PRD3 82270c04 0
[... 523 lines cut ...]
Aug 04 22:12:17 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:17 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmcblk0: recovery failed!
Aug 04 22:12:18 DietPi kernel: blk_update_request: I/O error, dev mmcblk0, sector 1284384 op 0x0:(READ) flags 0x80700 ph                                      ys_seg 4 prio class 0
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:18 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmc0: previous command never completed.
Aug 04 22:12:19 DietPi kernel: mmcblk0: recovery failed!
Aug 04 22:12:19 DietPi kernel: blk_update_request: I/O error, dev mmcblk0, sector 1284384 op 0x0:(READ) flags 0x0 phys_s                                      eg 1 prio class 0

Joulinar, 2022-08-13:
Looks like your SD card is dying. Probably you have another one that could be used as replacement. 

```
Aug 04 22:10:20 DietPi kernel: mmcblk0: recovery failed!
Aug 04 22:10:20 DietPi kernel: blk_update_request: I/O error, dev mmcblk0, sector 1284144 op 0x0:(READ) flags 0x80700 ph ys_seg 4 prio class 0
```

Snurring, 2022-08-13:
I don´t tink so. As stated in the bug report I have tried several SD-cards and the behaviour is the same. I tested the integrity of one of them with H2testw, and it passed. Also, I have overwritten two of the cards with the Armbian image and both have been running stable on the RPi's for days.

Joulinar, 2022-08-13:
As you can see on the error message, you have I/O issues with your SD card. 

```
Aug 04 20:31:21 DietPi fstrim[639]: fstrim: /boot: FITRIM ioctl failed: Input/output error
```

At the moment you are the only one reporting such issue. Therefore, I would not assume it's a general issue and it might be dedicated. Could you check for kernel error messages.

```
dmesg -l err,crit,alert,emerg
```

Snurring, 2022-08-13:
dmesg seems to return with I/O error when fault condition is present

MichaIng, 2022-08-17:
Can you please paste the whole `dmesg -l err,crit,alert,emerg` output or even `dmesg`?

The log you pasted seems to have been caused by `fstrim`, which runs once a week to discard unused blocks. If the issue appears more often than that, can you paste again logs which show the last actions before the first filesystem or I/O error occurs?

Of course also fstrim shouldn't cause issues, and it is enabled by default on Debian, should be as well on Armbian.

As it is one of the few differences between DietPi, Raspberry Pi OS and Armbian, could you try to switch the CPU governor to `ondemand` via `dietpi-config` Performance Options?

And please run an fsck on the root filesystem, as existing filesystem corruption can cause errors over and over again:
```sh
> /forcefsck
reboot
```
After reboot you can check the results via:
```sh
journalctl -t systemd-fsck
```

Snurring, 2022-08-21:
Results after additional testing:
So far the problem seems to appear when using a certain SD-card type, the 8GB Kingston Technology micoSDHC Class 4  (SDC4/8GB).
I have a bulk of these, and have been using them for RPi and other SBC’s without any issues for years.
Five of these cards have been tested, producing the same fault.
A NOOBS card (8GB Samsung MMCTR08GUBCH-ML) was tested and worked fine.

The dmesg command usually fails after the fault has shown up, but not every time.
I was able to run dmesg -k and dmesg -l err,crit,alert,emerg at one instance.
Logs are provided 

Switching the governor did not affect the issue.

When I tried making a forcefsck file on root, bash reported back: Read-only file system.
I rebooted the RPi (using reboot —force), made the forcefsck file and rebooted again.
Fsck did not detect any errors.

[dietpi-journalctl-log_start2.txt](https://github.com/MichaIng/DietPi/files/9389284/dietpi-journalctl-log_start2.txt)
[dmesg-l.txt](https://github.com/MichaIng/DietPi/files/9389291/dmesg-l.txt)
[dmesg-k.txt](https://github.com/MichaIng/DietPi/files/9389293/dmesg-k.txt)

Snurring, 2022-08-25:
It seems like fstrim is the culprit.
Tested three different images, all using the Kingston SD-card with results as follows.

Dietpi v8.8.0 (Bullseye. Moving to dev branch didn't change things)
`root@DietPi:~# fstrim -V`
`fstrim from util-linux 2.36.1`

`root@DietPi:~# fstrim -Av`
`fstrim: /boot: FITRIM ioctl failed: Input/output error`
`fstrim: /: FITRIM ioctl failed: Input/output error`

`root@DietPi:~# systemctl status fstrim`
`● fstrim.service - Discard unused blocks on filesystems from /etc/fstab`
`     Loaded: loaded (/lib/systemd/system/fstrim.service; static)`
`     Active: inactive (dead)`
`TriggeredBy: ● fstrim.timer`
`       Docs: man:fstrim(8)`

`root@DietPi:~# systemctl status fstrim.timer`
`● fstrim.timer - Discard unused blocks once a week`
`     Loaded: loaded (/lib/systemd/system/fstrim.timer; enabled; vendor preset: enabled)`
`     Active: active (waiting) since Wed 2022-08-24 22:41:22 BST; 8min ago`
`    Trigger: Thu 2022-08-25 00:01:58 BST; 1h 11min left`
`   Triggers: ● fstrim.service`
`       Docs: man:fstrim`

DietPi v7.94 (Stretch)
`root@DietPi:~# fstrim -V`
`fstrim from util-linux 2.29.2`

`root@DietPi:~# fstrim -av`
`/boot: 4.6 MiB (4802048 bytes) trimmed`
`/: 4.5 GiB (4836601856 bytes) trimmed`

Armbian:
`rpi4b:~:# uname -a`
`Linux rpi4b 5.15.48-bcm2711 #22.05.3 SMP PREEMPT Wed Jun 22 07:37:25 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

`rpi4b:~:# fstrim -V`
`fstrim from util-linux 2.37.2`

`rpi4b:~:# fstrim -Av`
`/boot/firmware: 136.9 MiB (143520256 bytes) trimmed on /dev/mmcblk0p1`
`/: 3.5 GiB (3727032320 bytes) trimmed on /dev/mmcblk0p2`

MichaIng, 2022-08-26:
Many thanks for testing. So it works on the DietPi ARMv6 Stretch image, on the Armbian ARMv8 image but not on the DietPi ARMv7 image. Does it work on the DietPi ARMv6 and ARMv8 images?

What I can tell is that it works on my RPi 2 ARMv7 system with an older 64 GiB Kingston SD card.

On Stretch, the kernel is locked to an older v5.10, right? Are there other differences in `/boot/config.txt` or CPU scheduler (`cpu`) which might have an effect?

Joulinar, 2022-08-26:
> On Stretch, the kernel is locked to an older v5.10, right? 

Latest kernel on Stretch should be 4.14.
5.10 was the latest one on Buster.

https://en.m.wikipedia.org/wiki/Raspberry_Pi_OS

MichaIng, 2022-08-26:
Ah right, I mixed up Stretch and Buster. So yeah, quite something will have changed between those kernel and also fstrim/distro versions, just sad that it changed from functional to non-functional in this regards.

Snurring, 2022-08-31:
Thanks for showing interest.

I have used two RPi3’s and one RPi2 for testing. Most of the tests have been done with the RPi3’s using ARMv8 images, but the ARMv6 and ARMv7 have also been tested and both failed in the same way as the ARMv8 image.
I don’t have any of the Buster images so I can’t test the card with that, but I did find an old image (v7.9.3 bullseye ARMv8) athat I tested and that also failed.

For the newer stuff I tested the bullseye dev branch and the bookworm branch and they both failed. I tested a 16G Kingston card (SDC4/16) and that card didn’t have any issues.

I also discovered that the fault occurred after running fstrim on the boot partition. Trimming the root partition works until one attempts trimming the  boot partition. After that is done, fstrim on both partitions will fail.

The last thing I did was running the dietpi-installer script on an RaspiOS LT (ARMv8) image.  This actually worked.