MichaIng/DietPi #5691: Linode GRUB2-issue

mews-se, 2022-08-14:
**#### Is your feature request related to a problem? Please describe:**

Converting a Debian-distro running on a Linode VPS fails due to their wierd grub-config.

**#### Describe the solution you'd like:**
Perhaps if someone, with the knowledge, could add a "how-to" on the DietPi-docs (https://dietpi.com/docs/hardware/#make-your-own-distribution) for applicable Linode users

**#### Additional context**

I guess Linode is pretty big, so perhaps someone with "power" could get in contact with them and politely ask them to make their users life easier :)

Also, it looks like someone has already attempted this on the Linode forums: https://www.linode.com/community/questions/22405/cannot-find-grub-drive

mews-se, 2022-08-17:
Thank. I hope that the dummy-command succeeds. I tried using the custom image - however for some reason I got a few "crashes" or at least a fair amount of unstability afterwards.

I'll get back after trying a dummy command!

mews-se, 2022-08-17:
The dummy-command actually made everything work better then with a custom image!

Perhaps the custom image was user error, but this is also a way easier method - perhaps worthy of making a note of it somewhere (since I had no idea that dummy-commands even existed :) )

MichaIng, 2022-08-17:
Hehe, there is no explicit "ignore" button intentionally, to make it less easy continuing when there is an error, which can lead to hard to debug follow-up issues 😉.

I checked the `dietpi-installer` and we can handle such cases:
- We use `lsblk -npo PKNAME "$(findmnt -Ufnro SOURCE -M /)"` to get the parent device/drive of the root partition. But in case of the default Linode image, there is not partition table but the filesystem is directly located on the drive itself, i.e. `/dev/sda1` has a parent `/dev/sda`, but `/dev/sda` has no parent. So `grub-install` fails.
- In such case we could simply skip the command, or shall we show a warning instead, that GRUB cannot be flashed, but that this is known to be the case with some VPS, e.g. Linode?

mews-se, 2022-08-17:
IMO the skipped command would be better, since it might be the same issue on other VPS-hosters

MichaIng, 2022-08-17:
I'm just hesitating to skip it completely silently, since GRUB is installed but cannot be flashed, which usually means the system cannot boot. We cannot know whether there are compatible bootloaders outside of the system drive. Not sure whether a missing partition table is sufficient indication to trust that there is a compatible bootloader somewhere, without throwing at least a prompt that an unexpected situation has been found?

mews-se, 2022-08-17:
Yeah you are right.

Show the warning that "Linode and others"? :)

MichaIng, 2022-08-17:
I'll implement the partition table check directly after the device has been selected. Only for VMs and containers, a missing partition table is accepted, but a yes-no dialogue shown with this info. It might again cause confusion, but I feel better when users abort the conversion when being unsure than running into a possibly unbootable system, e.g. if there was some other bootloader (configs) installed but removed or replaced by `dietpi-installer`/GRUB.

MichaIng, 2022-08-18:
Fixed with: https://github.com/MichaIng/DietPi/commit/634e362

closed 2022-08-28 by MichaIng