MichaIng/DietPi #5694: dietpi-wifi.txt - Wrong information result in failed automated installation

wassereimer86, 2022-08-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.7.1
- Distro version | bullseye
- Kernel version | Linux DietPi 5.15.56-v8+ #1575
- SBC model | RPi 4 Model B

#### Steps to reproduce

1. Start a fresh. automated, installation
2. Set AUTO_SETUP_AUTOMATED=1 in dietpi.txt
3. Add WiFi Network to dietpi-wifi.txt with this example password (not my real one): `KrBQ-.r&=N[xP:dJa_Mfb'c8HpamH}`

#### Expected behaviour

- Connect to WiFi Network

#### Actual behaviour

- Error while installation (something with wpa_supplicant could not be started)

#### Extra details

Problem is that the documentation in the file says "Please do NOT escape any magic characters. The script does this automatically as required.". So i did exactly this. The entry in the file was `aWIFI_KEY[0]='KrBQ-.r&=N[xP:dJa_Mfb'c8HpamH}'` and this can not work because we have multiple -> ' <-. Solution was to change the line to `aWIFI_KEY[0]="KrBQ-.r&=N[xP:dJa_Mfb'c8HpamH}"`

MichaIng, 2022-08-22:
Many thanks for your report.

You are right, the single quote character `'` itself needs to be escaped by `'\''`, since the whole string is single-quoted. I'll add this info to the `dietpi-wifi.txt`.

MichaIng, 2022-08-22:
Done with: https://github.com/MichaIng/DietPi/commit/92a35b4
Do you agree with the sentence?

wassereimer86, 2022-08-22:
Perfect for me. 🙂 Thank you!

closed 2022-08-28 by MichaIng