MichaIng/DietPi #5697: gpsd.socket: Failed to receive listening socket ([::1]:2947): Input/output error

keypunch416, 2022-08-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.6.1
- Distro version | bullseye
- Kernel version | `Linux hostname 5.10.123-meson64 #22.05.3 SMP PREEMPT Wed Jun 22 07:23:04 UTC 2022 aarch64 GNU/Linux`
- SBC model | N2 (aarch64) (N2+ 4GB)
- Power supply used | ALLPOWER 15VDC/4A
- SD card used | 32GB Kingston CANVAS Select Plus MicroSD 100MB/s A1

#### Additional Information (if applicable)
- Software title | gpsd 3.22
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | N/A

#### Steps to reproduce

1. sudo apt update
2. sudo apt install gpsd gpsd-clients pps-tools
3. connected USB GPS - u-blox M8030 then lsusb
4. gpsmon /dev/gps0
5. configure /etc/default/gpsd via nano
6. sudo systemctl enable gpsd
7. sudo systemctl start gpsd
8. sudo gpsd /dev/gps0 -F /var/run/gpsd.sock
9. cgps -s /dev/gps0
10. 
11. ...

#### Expected behaviour

- Step 1 successful
- Step 2 successful
- Step 3 successful
- Step 4 successful
- Step 5 successful
- Step 6 successful
- Step 7 successful
- Step 8 successful
- Step 9 successful

#### Actual behaviour

- Step 1 successful
- Step 2 successful with version 3.22 installed
- Step 3 successful with lsusb result: Bus 001 Device 009: ID 1546:01a8 U-Blox AG [u-blox 8]
- Step 4 successful
- Step 5 successful
- Step 6 successful
- Step 7 failed : A dependency job for gpsd.service failed. See 'journalctl -xe' for details.
Aug 22 15:23:27 hostname systemd[1]: Created slice system-gpsdctl.slice.
░░ Subject: A start job for unit system-gpsdctl.slice has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit system-gpsdctl.slice has finished successfully.
░░ 
░░ The job identifier is 2675.
Aug 22 15:23:27 hostname systemd[33851]: gpsd.socket: Failed to create listening socket ([::1]:2947): Cannot assign requested address
Aug 22 15:23:27 hostname systemd[1]: gpsd.socket: Failed to receive listening socket ([::1]:2947): Input/output error
Aug 22 15:23:27 hostname systemd[1]: gpsd.socket: Failed to listen on sockets: Input/output error
Aug 22 15:23:27 hostname systemd[1]: gpsd.socket: Failed with result 'resources'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit gpsd.socket has entered the 'failed' state with result 'resources'.
Aug 22 15:23:27 hostname systemd[1]: Failed to listen on GPS (Global Positioning System) Daemon Sockets.
░░ Subject: A start job for unit gpsd.socket has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit gpsd.socket has finished with a failure.
░░ 
░░ The job identifier is 2674 and the job result is failed.
Aug 22 15:23:27 hostname systemd[1]: Dependency failed for Manage ttyACM0 for GPS daemon.
░░ Subject: A start job for unit gpsdctl@ttyACM0.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit gpsdctl@ttyACM0.service has finished with a failure.
░░ 
░░ The job identifier is 2605 and the job result is dependency.
Aug 22 15:23:27 hostname systemd[1]: gpsdctl@ttyACM0.service: Job gpsdctl@ttyACM0.service/start failed with result 'dependency'.
- Step 8 successful
- Step 9 successful

#### Extra details

- ls -ls /dev/gps*
0 lrwxrwxrwx 1 root root 7 Aug 22 15:23 /dev/gps0 -> ttyACM0

MichaIng, 2022-08-26:
Many thanks for your report.
```
Job gpsdctl@ttyACM0.service/start failed with result 'dependency'
```
Can you show the output of:
```sh
systemctl cat gpsdctl@ttyACM0
```

keypunch416, 2022-08-26:
I was out much of day and in course of the day more odd nonsense occurred.  I need to see if what I have done addresses one of the odd behaviours.  I also do not understand why all of sudden Linux changed /dev/GPSx.  i have seen this sudden with no reason USB port assignment change for no reason with Raspberry Pi OS on system that uses number of USB devices to feed weather data.  This means yet another Linux or udev related issue that I would suggest is due to Linux Kernel and/or udev.  That said this GPS via USB is real mess and I had stability I thought until something happened while I was out for about 8 hours today.  So I had added even more to the complex changes I have made to see if I at least solved one of the issues, new one related to IPV6 that occurred while I was out and was not presenting in last few days with the many changes I have effected based on my initial findings and work of number of days of testing.

keypunch416, 2022-08-27:
For reasons unknown secondary issues have arisen so I reset the gpsd systemd files back to original installed by gpsd package which was the state of the gpsd systemd files when reported this issue and the related findings.

The output you requested is:

`systemctl cat gpsdctl@ttyACM0
# /lib/systemd/system/gpsdctl@.service
[Unit]
Description=Manage %I for GPS daemon
Requires=gpsd.socket
BindsTo=dev-%i.device
After=dev-%i.device

[Service]
Type=oneshot
Environment="GPSD_SOCKET=/run/gpsd.sock"
EnvironmentFile=-/etc/default/gpsd
RemainAfterExit=yes
ExecStart=/bin/sh -c "[ \"$USBAUTO\" = true ] && /usr/sbin/gpsdctl add /dev/%I || :"
ExecStop=/bin/sh -c "[ \"$USBAUTO\" = true ] && /usr/sbin/gpsdctl remove /dev/%I || :"
`

MichaIng, 2022-09-11:
So it seems `gpsd.socket` failed. Can you check:
```sh
systemctl status gpsd.socket
systemctl cat gpsd.socket
ls -l /run/gpsd.sock
```

keypunch416, 2022-09-11:
Bear in mind there may be delays for some of the requested information at times as I am working through a number of issues, including issues I have yet to report, clusters of medical issues, and coding issues.  The coding issues that are just me issues in porting and/or improving seismic application related code I have written.  The N2P was intended to be used for recording earthquakes around the world and I have missed a number of, including the significant events in the Indonesia area in last 36 hours due to 'Unable to enable SPI /dev/spidev0.0' #5645.

As gpsd.socket failed to run there was no /run/gpsd.sock.

/usr/lib/systemd/system/gpsd.socket:
[Unit]
Description=GPS (Global Positioning System) Daemon Sockets

[Socket]
ListenStream=/run/gpsd.sock
ListenStream=[::1]:2947
ListenStream=127.0.0.1:2947
# To allow gpsd remote access, start gpsd with the -G option and
# uncomment the next two lines:
# ListenStream=[::]:2947
# ListenStream=0.0.0.0:2947
SocketMode=0600
BindIPv6Only=yes

[Install]
WantedBy=sockets.target

/usr/lib/systemd/system/gpsd.service
[Unit]
Description=GPS (Global Positioning System) Daemon
Requires=gpsd.socket
# Needed with chrony SOCK refclock
After=chronyd.service

[Service]
Type=forking
EnvironmentFile=-/etc/default/gpsd
ExecStart=/usr/sbin/gpsd $GPSD_OPTIONS $OPTIONS $DEVICES

[Install]
WantedBy=multi-user.target
Also=gpsd.socket

2022-08-22 18:27:51 +0000 UTC : 1661192871  n2p@dietpi tty0 /etc/default $ sudo systemctl start gpsd.socket
Job failed. See "journalctl -xe" for details.
2022-08-22 18:28:48 +0000 UTC : 1661192928  n2p@dietpi 62FE9A82 tty0 /etc/default $ sudo systemctl status gpsd.socket
● gpsd.socket - GPS (Global Positioning System) Daemon Sockets
     Loaded: loaded (/lib/systemd/system/gpsd.socket; enabled; vendor preset: enabled)
     Active: failed (Result: resources)
   Triggers: ● gpsd.service
     Listen: /run/gpsd.sock (Stream)
             [::1]:2947 (Stream)
             127.0.0.1:2947 (Stream)
        CPU: 1ms

Aug 22 18:08:35 dietpi systemd[1]: Failed to listen on GPS (Global Positioning System) Daemon Sockets.
Aug 22 18:27:51 dietpi systemd[1]: gpsd.socket: Failed to receive listening socket ([::1]:2947): Input/output error
Aug 22 18:27:51 dietpi systemd[1]: gpsd.socket: Failed to listen on sockets: Input/output error
Aug 22 18:27:51 dietpi systemd[1]: gpsd.socket: Failed with result 'resources'.
Aug 22 18:27:51 dietpi systemd[1]: Failed to listen on GPS (Global Positioning System) Daemon Sockets.
Aug 22 18:28:48 dietpi systemd[35008]: gpsd.socket: Failed to create listening socket ([::1]:2947): Cannot assign requested address
Aug 22 18:28:48 dietpi systemd[1]: gpsd.socket: Failed to receive listening socket ([::1]:2947): Input/output error
Aug 22 18:28:48 dietpi systemd[1]: gpsd.socket: Failed to listen on sockets: Input/output error
Aug 22 18:28:48 dietpi systemd[1]: gpsd.socket: Failed with result 'resources'.
Aug 22 18:28:48 dietpi systemd[1]: Failed to listen on GPS (Global Positioning System) Daemon Sockets.
2022-08-22 18:29:03 +0000 UTC : 1661192943  n2p@dietpi tty0 /etc/default $

2022-08-22 18:08:35 +0000 UTC : 1661191715  n2p@dietpi tty0 ~ $ journalctl -xe
[snip]
Aug 22 15:23:27 dietpi systemd[33851]: gpsd.socket: Failed to create listening socket ([::1]:2947): Cannot assign requested address
Aug 22 15:23:27 dietpi systemd[1]: gpsd.socket: Failed to receive listening socket ([::1]:2947): Input/output error
Aug 22 15:23:27 dietpi systemd[1]: gpsd.socket: Failed to listen on sockets: Input/output error
Aug 22 15:23:27 dietpi systemd[1]: gpsd.socket: Failed with result 'resources'.
░░ Subject: Unit failed
░░ Defined-By: systemd
[... 40 lines cut ...]
Aug 22 18:08:35 dietpi systemd[1]: gpsd.socket: Failed to receive listening socket ([::1]:2947): Input/output error
Aug 22 18:08:35 dietpi systemd[1]: gpsd.socket: Failed to listen on sockets: Input/output error
Aug 22 18:08:35 dietpi systemd[1]: gpsd.socket: Failed with result 'resources'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit gpsd.socket has entered the 'failed' state with result 'resources'.
Aug 22 18:08:35 dietpi systemd[1]: Failed to listen on GPS (Global Positioning System) Daemon Sockets.
░░ Subject: A start job for unit gpsd.socket has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit gpsd.socket has finished with a failure.
░░ 
░░ The job identifier is 2895 and the job result is failed.
Aug 22 18:08:35 dietpi systemd[1]: Dependency failed for GPS (Global Positioning System) Daemon.
░░ Subject: A start job for unit gpsd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit gpsd.service has finished with a failure.
░░ 
░░ The job identifier is 2826 and the job result is dependency.
Aug 22 18:08:35 dietpi systemd[1]: gpsd.service: Job gpsd.service/start failed with result 'dependency'.

MichaIng, 2022-09-18:
Do you have IPv6 enabled?
```sh
ip -6 a
```
Or is something else listening on this port?
```sh
ss -tulpn | grep ':2947'
```

MichaIng, 2022-12-04:
@keypunch416 
Does your issue persist?

mclendon99, 2022-12-28:
I had the same issue. The clue is that "ip 6 -a" has no output, i.e. IPv6 is not enabled.
So, do a clean override of the systemd stuff

sudo cp /lib/systemd/system/gpsd* /etc/systemd/system

and edit /etc/systemd/system/gpsd.socket to comment out the ipv6 config. Here's mine:

Description=GPS (Global Positioning System) Daemon Sockets

[Socket]
# Clear default config
ListemStream=
ListenStream=/var/run/gpsd.sock
#ListenStream=[::1]:2947
ListenStream=127.0.0.1:2947
# To allow gpsd remote access, start gpsd with the -G option and
# uncomment the next two lines:
# ListenStream=[::]:2947
# ListenStream=0.0.0.0:2947
SocketMode=0600
#BindIPv6Only=ipv6-only

[Install]
WantedBy=sockets.target

The system MUST be rebooted. Restarting the service won't do it, obviously.

MichaIng, 2022-12-29:
Many thanks for reporting. Makes much sense that the socket fails if IPv6 is disabled, as it is explicitly configured to listen on both, IPv4 and IPv6.

A reboot shouldn't be required, but instead:
```sh
systemctl daemon-reload
systemctl restart gpsd.socket
systemctl restart gpsd
```

MichaIng, 2022-12-30:
Marking as closed due to outstanding reply of OP. Feel free to reopen if either enabling IPv6 or disabling it in the socket does not help in your case.

closed 2022-12-30 by MichaIng

MichaIng, 2025-06-29:
Sounds like the socket starts before network has been sufficiently configured. So to not affect the way the socket works, this might be an alternative:
```sh
mkdir -p /etc/systemd/system/gpsd.socket.d
cat << '_EOF_' > /etc/systemd/system/gpsd.socket.d/after-network.conf
[Unit]
After=network.target
_EOF_
```
or even
```sh
cat << '_EOF_' > /etc/systemd/system/gpsd.socket.d/after-network.conf
[Unit]
Wants=network-online.target
After=network-online.target
_EOF_
```
To verify the order in which things are started/reached:
```sh
journalctl -u gpsd.socket -u network.target -u network-online.target
```

MichaIng, 2025-06-29:
> Hm, tried the latter but it didn't succeed. The whole network didn't came up after that.

What? This cannot affect network bring up, unless the network would depend on GPS somehow. Only the other way round: GPS is delayed after network is there, network is brought up just like before.

Do not unset `WantedBy=sockets.target`, otherwise, when disabling and re-enabling the socket, it is just never activated. `WantedBy` is what adds a unit into the boot schedule, letting it start on a target which is fixed in boot schedule, `sockets.target` instead. The effect when changing this however only shows up when disabling/enabling a unit, since this is what creates the symlinks, like in this case:
```sh
/etc/systemd/system/sockets.target.wants/gpsd.socket => /lib/systemd/system/gpsd.socket
```

> This time, systemctl was not referring to gpsd.socket, but to some networkmanager online target (if I recall correctly).

I guess you mean `network-online.target`, which is what we added to the boot schedule, since it is not a passive target that is inside in any case. However, it should be inside by default on DietPi, pulled in by the `dietpi-postboot.service`, at least if the system is configured to wait for network before showing the pre-login banner (not affecting the actual user/password input prompt) and starting update checks. Can you check this:
```sh
systemctl show -p WantedBy network-online.target
```
This target however does not affect network setup, if newly pulled in, it is just "reached" when `ifup@<interface>.service` returned, which is after complete network setup in case of Ethernet and static IP, but can be earlier if DHCP is used and first lease times out, or it WPA supplicant WiFi connection takes somewhat longer. It affects only those units which order themselves `After=` (or `Before=`) this target, well, and it can, if network setup hangs for some reason, some processing animation on the local screen, which however is visual only, since login there should be possible regardless: the user/password input prompt is an own unit `getty@tty1.service` which does not wait for the whole boot query to finish. But if one unit is hanging in foreground, it might distort the input text. However, in any case it is not an added network setup issue, but just makes it visible, if it hangs longer for some reason. You might want to check it in this case:
```sh
journalctl -u ifup@eth0
```
or with `wlan0` if you use WiFi.