MichaIng/DietPi #5698: NextCloud broken after dietpi-update

Phil1988, 2022-08-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2022-08-23:
What exactly is the error message you have?

Phil1988, 2022-08-23:
The error message regarding what?

The nextcloud does not seem to be active.
The main issue here is that the service is not properly installed and can be started:
```
root@DietPi:~# dietpi-services start nextcloud
[FAILED] DietPi-Services | Service (nextcloud) could not be found.
root@DietPi:~#
```

Checking it via my browser gives me depending on the browser different erropages.
Firefox says
`Fehlercode: SEC_ERROR_INADEQUATE_KEY_USAGE`

while brave browser says
`NET::ERR_CERT_AUTHORITY_INVALID`

Joulinar, 2022-08-23:
Of course there is no Nextcloud service because it's a web server application. Nextcloud is provided by Apache2 web server. 

The issue you have seems to be related to your SSL certificate. I found some issues related to Firefox. https://github.com/dotnet/AspNetCore.Docs/issues/20567

Maybe you could test other browsers like Chrome or Edge

Phil1988, 2022-08-23:
I thought Brave browser is similar to chrome. 
But ofc I also checked Edge and Chrome before posting here.

I got notified by the android DAVx5 app, that no sync is possible.
Then checked the android nextcloud app and then the browers.

I am still checking, but I am 100% sure that its not a browser related problem.

Maybe a redirection problem on some configs, router etc..  (I will check that and let you know)

Phil1988, 2022-08-27:
This is surely not an issue to dietpi, but was a redirection issue.
Im still not sure what triggered it (maybe the update was a part of it) but it was a router/DNS redirection problem what happen (randomly or partly related) right after the update.

closed 2022-08-27 by Phil1988

xCraf, 2022-10-25:
Hi, 
i'm having the same issue, could you please tell me how do you solved it?