MichaIng/DietPi #5699: FFMPEG doesn't work after upgrading from v8.5.1 to v8.7.1

LaurenceGough, 2022-08-23:
#### Steps to reproduce
1. Applications (Frigate NVR) works fine in DietPi v8.5.1 but after upgrading to v8.7.1 it stops working, even after rebooting and checking settings such as hardware acceleration. 
2. I did restore from backup to v8.5.1, everything was working as expected again. I confirmed there were no other changes. I repeated this twice to confirm. I tried installing FFMPEG via the software menu (option 7) and despite rebooting it still didn't work. 

#### Expected behaviour
- Applications (Frigate NVR CCTV software) that rely on FFMPEG works

#### Actual behaviour
- Applications report that FFMPEG process is crashing but I never see any video feed from my CCTV cameras as expected

#### Extra details
- Raspberry Pi 4 2GB model
-
Sorry I do not know Linux or DietPi OS that well, but this is the first issue I've ever encountered.

Thank you

Joulinar, 2022-08-23:
If you have restored your backup, could you try following

```
apt update
apt upgrade
```

does your system is still working afterwards?

LaurenceGough, 2022-08-23:
> If you have restored your backup, could you try following
> 
> ```
> apt update
> apt upgrade
> ```
> 
> does your system is still working afterwards?

Ah yes my bad, I completely forgot about trying that simple idea first!! Whoops sorry.

At first it worked even after rebooting the container for that application, but then after rebooting the Pi indeed that breaks the application too so my title is completely misleading here so I do apologise. Hmm, I'm now not sure where to go now.

Thank you

Joulinar, 2022-08-23:
ok looks like it's not an issue of the DietPi update. Seems to be more a challenge with some Debian software packages. Let's check what kind of packages are going to be updated. Before you would need to go back to original status. 😃 

```
apt update
apt list --upgradable
```

LaurenceGough, 2022-08-23:
> 

Thank you, ok restored again and rebooted - confirmed all working again. Then performed as requested, please find the output:
```
base-files/stable 11.1+deb11u4 arm64 [upgradable from: 11.1+deb11u3]
bash/stable 5.1-2+deb11u1 arm64 [upgradable from: 5.1-2+b3]
containerd.io/bullseye 1.6.7-1 arm64 [upgradable from: 1.6.4-1]
curl/stable-security 7.74.0-1.3+deb11u2 arm64 [upgradable from: 7.74.0-1.3+deb11u1]
dirmngr/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
docker-ce-cli/bullseye 5:20.10.17~3-0~debian-bullseye arm64 [upgradable from: 5:20.10.16~3-0~debian-bullseye]
docker-ce/bullseye 5:20.10.17~3-0~debian-bullseye arm64 [upgradable from: 5:20.10.16~3-0~debian-bullseye]
dpkg/stable 1.20.11 arm64 [upgradable from: 1.20.10]
firmware-atheros/stable 1:20210315-3+rpt7 all [upgradable from: 1:20210315-3+rpt5]
firmware-brcm80211/stable 1:20210315-3+rpt7 all [upgradable from: 1:20210315-3+rpt5]
firmware-iwlwifi/stable 1:20210315-3+rpt7 all [upgradable from: 1:20210315-3+rpt5]
firmware-misc-nonfree/stable 1:20210315-3+rpt7 all [upgradable from: 1:20210315-3+rpt5]
firmware-realtek/stable 1:20210315-3+rpt7 all [upgradable from: 1:20210315-3+rpt5]
gnupg-l10n/stable,stable-security 2.2.27-2+deb11u2 all [upgradable from: 2.2.27-2+deb11u1]
gnupg-utils/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
gnupg/stable,stable-security 2.2.27-2+deb11u2 all [upgradable from: 2.2.27-2+deb11u1]
gpg-agent/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
gpg-wks-client/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
gpg-wks-server/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
gpg/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
gpgconf/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
gpgsm/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
gpgv/stable,stable-security 2.2.27-2+deb11u2 arm64 [upgradable from: 2.2.27-2+deb11u1]
libc-bin/stable 2.31-13+rpt2+rpi1+deb11u3 arm64 [upgradable from: 2.31-13+rpt2+rpi1+deb11u2]
libc-l10n/stable 2.31-13+rpt2+rpi1+deb11u3 all [upgradable from: 2.31-13+rpt2+rpi1+deb11u2]
libc6/stable 2.31-13+rpt2+rpi1+deb11u3 arm64 [upgradable from: 2.31-13+rpt2+rpi1+deb11u2]
libcurl4/stable-security 7.74.0-1.3+deb11u2 arm64 [upgradable from: 7.74.0-1.3+deb11u1]
libgnutls30/stable-security 3.7.1-5+deb11u2 arm64 [upgradable from: 3.7.1-5]
libldb2/stable-security 2:2.2.3-2~deb11u2 arm64 [upgradable from: 2:2.2.3-2~deb11u1]
libpam-modules-bin/stable 1.4.0-9+deb11u1+rpt2 arm64 [upgradable from: 1.4.0-9+deb11u1+rpt1]
libpam-modules/stable 1.4.0-9+deb11u1+rpt2 arm64 [upgradable from: 1.4.0-9+deb11u1+rpt1]
libpam-runtime/stable 1.4.0-9+deb11u1+rpt2 all [upgradable from: 1.4.0-9+deb11u1+rpt1]
libpam0g/stable 1.4.0-9+deb11u1+rpt2 arm64 [upgradable from: 1.4.0-9+deb11u1+rpt1]
libsmbclient/stable-security 2:4.13.13+dfsg-1~deb11u5 arm64 [upgradable from: 2:4.13.13+dfsg-1~deb11u3]
libssl1.1/stable 1.1.1n-0+deb11u3+rpt1 arm64 [upgradable from: 1.1.1n-0+deb11u2+rpt1]
libtirpc-common/stable-security 1.3.1-1+deb11u1 all [upgradable from: 1.3.1-1]
libtirpc3/stable-security 1.3.1-1+deb11u1 arm64 [upgradable from: 1.3.1-1]
libwbclient0/stable-security 2:4.13.13+dfsg-1~deb11u5 arm64 [upgradable from: 2:4.13.13+dfsg-1~deb11u3]
locales/stable 2.31-13+rpt2+rpi1+deb11u3 all [upgradable from: 2.31-13+rpt2+rpi1+deb11u2]
nano/stable 5.4-2+deb11u1 arm64 [upgradable from: 5.4-2]
openssh-client/stable 1:8.4p1-5+deb11u1 arm64 [upgradable from: 1:8.4p1-5]
openssh-sftp-server/stable 1:8.4p1-5+deb11u1 arm64 [upgradable from: 1:8.4p1-5]
openssl/stable 1.1.1n-0+deb11u3+rpt1 arm64 [upgradable from: 1.1.1n-0+deb11u2+rpt1]
python3-ldb/stable-security 2:2.2.3-2~deb11u2 arm64 [upgradable from: 2:2.2.3-2~deb11u1]
raspberrypi-bootloader/stable 1:1.20220811-1 arm64 [upgradable from: 1:1.20220331-1]
raspberrypi-kernel/stable 1:1.20220811-1 arm64 [upgradable from: 1:1.20220331-1]
rpi-eeprom/stable 13.16-1 arm64 [upgradable from: 13.14-1]
tzdata/stable-updates 2021a-1+deb11u5 all [upgradable from: 2021a-1+deb11u3]
unzip/stable-security 6.0-26+deb11u1 arm64 [upgradable from: 6.0-26]
wireless-regdb/stable 2022.04.08-2~deb11u1 all [upgradable from: 2020.04.29-2]
```

I must admit I was expecting to see an ffmpeg package there! I am completely guessing that it's part of another package, I've no idea I'm afraid.

Joulinar, 2022-08-23:
As I understood Frigate NVR is a Docker container. Probably Docker update is causing issues. You could try to update packages one by one to see where it is failing. I see you opened as well on issue with software vendor https://github.com/blakeblackshear/frigate/issues/3705

MichaIng, 2022-10-23:
Closing it here, investigation is done upstream: https://github.com/blakeblackshear/frigate/issues/3780

closed 2022-10-23 by MichaIng