MichaIng/DietPi #5703: DietPi restore "warning uuids of the backup and current system differ"

Joulinar, 2022-08-24:
### Discussed in https://github.com/MichaIng/DietPi/discussions/5702

<div type='discussions-op-text'>

<sup>Originally posted by **LaurenceGough** August 24, 2022</sup>
Hello,

After losing months of work is there any way to workaround or reconfigure my clean installed DietPi system to accept a DietPi backup from before the clean install?

Currently I get the warning "warning uuids of the backup and current system differ". If I accept I get as expected a load of permissions errors and it seems to break all.

I've lost everything, my whole Home assistant config and home setup, my Frigate NVR container , all my other containers all gone :( The portainer backup I had was useless. 

Thank you in advance/</div>

MichaIng, 2022-08-25:
Ah that makes sense. It checks whether the target filesystem supports UNIX permissions when creating a backup, but not when restoring the backup.

Ah least you should be able to restore the individual software data from `/mnt/dietpi_userdata`, install software on top of it and `chown -R` the directories where it is not done on install anyway.

LaurenceGough, 2022-08-25:
> Ah that makes sense. It checks whether the target filesystem supports UNIX permissions when creating a backup, but not when restoring the backup.
>
> Ah least you should be able to restore the [...]

Thanks so much Michal, I did try restoring that data but I didn't try a chown on them (although with the full restore I did try chown as a desperate fix and I didn't have permissions to run chown which I guess is expected when you destroy the entire OS's permissions :D) 

Thanks again for your help - I'll try and close this now, just user error by me! And thank you for DietPi I love the GUI options and all the customization to really make it how you want it.

MichaIng, 2022-08-26:
Ah right, ported from discussions, closing it now.

closed 2022-08-26 by MichaIng