MichaIng/DietPi #5704: Problems after update to 8.7.1  - LXDE desktop via RDP

jalsco, 2022-08-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
[Wed Aug 24] root@batopi: ~$ cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
[Wed Aug 24] root@batopi: ~$ echo $G_DISTRO_NAME $G_RASPBIAN
bullseye
[Wed Aug 24] root@batopi: ~$ uname -a
Linux batopi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux
[Wed Aug 24] root@batopi: ~$ echo $G_HW_MODEL_NAME
ROCK Pi 4 (aarch64)
```
#### Actual behaviour

I was away from this system for 4+ months, and it was untouched by anyone in that time. But was running just fine 😊. All the websites were being served flawlessly (using: caddy and lighttpd) . Just one odd thing, in the last month, it would power off after 4-5 days for no obvious reason. But then - manually powering back on - would start everything running again with no errors.  So nobody had any concerns about that. 

Before updating to 8.7.1 , I confirmed - all things I use were running:  LXDE ( access by RDP ) , SSH (by Putty and WinSCP) , Caddy, MariaDB, Matomo, Python3, Pelican. ... All those were running great. I then updated by :  `dietpi-update`.  After some time (20mins ? ) it completed. There were no errors or warning messages, only 'success' and green ticks.

1. PROBLEM: Desktop is not displaying  - I cannot access desktop via RDP as before.  It just displays completely black screen
[image]
It does not get to the "XRDP  Just Connecting  login screen". Screen just goes fully black...

2. STILL GOOD: Lighttp, mariadb, python, SSH are all running fine, without any obvious problems. 

3. MY GUESS:  I had big problems with **this exact component ( LXDE via RDP )** about 9 months ago - though the symptoms were bit different. Back then the problem was also caused by a software update. How it was fixed - by @MichaIng - is fully documented in #5012 

I suspect that the changes made to fix  #5012 , did not play well with the update to 8.7.1 , or they got overwritten. I say that as **a total guess**, made purely because current problem is in the same components.    

Any help to get it working again will be GREATLY appreciated ! 🙏🏻

#### Extra details

- ...

jalsco, 2022-08-25:
I didn't mention explicitly - the `dietpi-update` I did updated all the way from `7.9` up to `8.7.1`

jalsco, 2022-08-25:
Here's the XServer log, from this command:  `cat ~/.xorgxrdp.10.log`

Hopefully it is meaningful to somebody 🤷‍♂️

```
[ 41781.182] rdpClientConSend: g_tcp_send failed(returned -1)
[ 41781.197] rdpClientConSendMsg: error in rdpup_send_msg
[ 41781.197] rdpClientConSendPending: rdpClientConSendMsg failed
[ 41781.198] rdpClientConDisconnect:
[ 41781.198] rdpRemoveClientConFromDev: removing clientCon 0xaaaad7c5b770
[ 47319.519] rdpClientConGotConnection:
[ 47319.519] rdpClientConGotConnection: g_sck_accept ok new_sck 7
[ 47319.519] rdpClientConGetConnection: idle_disconnect_timeout set to non-positive value, idle timer turned off
[ 47319.519] rdpAddClientConToDev: adding first clientCon 0xaaaad7c5b770
[ 47319.520] rdpClientConProcessMsgVersion: version 0 0 0 1
[ 47319.520] rdpClientConProcessScreenSizeMsg: set width 1920 height 1080 bpp 24
[ 47319.520] rdpClientConProcessScreenSizeMsg: shmemid 21 shmemptr 0xffff86b02000
[ 47319.520] rdpClientConProcessMsgClientInput: invalidate x 0 y 0 cx 1920 cy 1080
[ 47319.520] rdpClientConProcessMsgClientInfo:
[ 47319.520]   got client info bytes 5752
[ 47319.520]   jpeg support 0
[ 47319.520]   offscreen support 1
[ 47319.520]   offscreen size 7864320
[ 47319.520]   offscreen entries 100
[ 47319.520] rdpClientConProcessMsgClientInfo: got RFX capture
[ 47319.520]   cap_width 1920 cap_height 1088
[ 47319.520] rdpClientConProcessMsgClientInfo: shmemid 22 shmemptr 0xffff86af3000 bytes 8355840
[ 47319.520]   client can not do offscreen to offscreen blits
[ 47319.520]   client can do new(color) cursor
[ 47319.520]   client can not do multimon
[ 47319.520] rdpRRSetRdpOutputs: numCrtcs 1 numOutputs 1 monitorCount 0
[ 47319.520] rdpRRSetRdpOutputs: update output 0 left 0 top 0 width 1920 height 1080
[ 47319.520] rdpRRUpdateOutput:
[ 47319.521] rdpLoadLayout: keylayout 0x00000409 variant  display 10
[ 47319.521] rdpkeybChangeKeyboardControl:
[ 47319.521] rdpkeybChangeKeyboardControl: autoRepeat on
[ 47319.525] rdpkeybChangeKeyboardControl:
[ 47319.525] rdpkeybChangeKeyboardControl: autoRepeat on
[ 47319.589] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47319.589] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47319.589] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47319.590] rdpEglOut: resize the crc list was 0 now 510
[ 47319.622] rdpInDeferredRepeatCallback:
[ 47319.622] rdpkeybChangeKeyboardControl:
[ 47319.622] rdpkeybChangeKeyboardControl: autoRepeat off
[ 47319.625] rdpInDeferredRepeatCallback:
[ 47319.625] rdpkeybChangeKeyboardControl:
[ 47319.625] rdpkeybChangeKeyboardControl: autoRepeat off
[ 47319.698] KbdSync: toggling num lock
[ 47319.698] rdpkeybChangeKeyboardControl:
[ 47319.698] rdpkeybChangeKeyboardControl: autoRepeat off
[ 47319.699] rdpkeybChangeKeyboardControl:
[ 47319.699] rdpkeybChangeKeyboardControl: autoRepeat off
[ 47320.558] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47320.559] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47320.559] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47320.948] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47320.949] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47320.949] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47321.351] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47321.351] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47321.352] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47322.078] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47322.078] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47322.078] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47322.150] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47322.150] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47322.150] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47322.551] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47322.552] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[... 199 lines cut ...]
[ 47398.601] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47398.602] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47398.602] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47400.103] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47400.103] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47400.104] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47401.603] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47401.604] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47401.604] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[... cut ...]
[ 47407.607] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47407.607] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47407.608] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[ 47409.108] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[ 47409.108] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[ 47409.109] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
```

MichaIng, 2022-08-25:
Many thanks for your report.

Different error but also on glamor indeed:
```
(EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
```
Can you show the output of:
```
diff /etc/X11/xrdp/xorg.conf{,.dpkg-dist}
```

jalsco, 2022-08-25:
Hi @MichaIng  😊🙏🏻 and thank you so much for your **GREAT** and tireless work ! 

```
[Thu Aug 25] root@batopi: ~$ diff /etc/X11/xrdp/xorg.conf{,.dpkg-dist}
diff: /etc/X11/xrdp/xorg.conf.dpkg-dist: No such file or directory
[Thu Aug 25] root@batopi: ~$
```

Did I do the command right ?

jalsco, 2022-08-25:
here is `/etc/X11/xrdp/xorg.conf`

```

Section "ServerLayout"
    Identifier "X11 Server"
    Screen "Screen (xrdpdev)"
    InputDevice "xrdpMouse" "CorePointer"
    InputDevice "xrdpKeyboard" "CoreKeyboard"
EndSection

Section "ServerFlags"
    Option "DefaultServerLayout" "X11 Server"
    Option "DontVTSwitch" "on"
    Option "AutoAddDevices" "off"
EndSection

Section "Module"
    Load "dbe"
    Load "ddc"
    Load "extmod"
    Load "glx"
    Load "int10"
    Load "record"
    Load "vbe"
    Load "glamoregl"
    Load "xorgxrdp"
    Load "fb"
EndSection

Section "InputDevice"
    Identifier "xrdpKeyboard"
    Driver "xrdpkeyb"
EndSection

Section "InputDevice"
    Identifier "xrdpMouse"
    Driver "xrdpmouse"
EndSection

Section "Monitor"
    Identifier "Monitor"
    Option "DPMS"
    HorizSync 30-80
    VertRefresh 60-75
    ModeLine "1920x1080" 138.500 1920 1968 2000 2080 1080 1083 1088 1111 +hsync -vsync
    ModeLine "1280x720" 74.25 1280 1720 1760 1980 720 725 730 750 +HSync +VSync
    Modeline "1368x768" 72.25 1368 1416 1448 1528 768 771 781 790 +hsync -vsync
    Modeline "1600x900" 119.00 1600 1696 1864 2128 900 901 904 932 -hsync +vsync
EndSection

Section "Device"
    Identifier "Video Card (xrdpdev)"
    Driver "xrdpdev"
    Option "DRMDevice" "/dev/dri/renderD128"
    Option "DRI3" "1"
EndSection

Section "Screen"
    Identifier "Screen (xrdpdev)"
    Device "Video Card (xrdpdev)"
    Monitor "Monitor"
    DefaultDepth 24
    SubSection "Display"
        Depth 24
        Modes "640x480" "800x600" "1024x768" "1280x720" "1280x1024" "1600x900" "1920x1080"
    EndSubSection
EndSection
```

MichaIng, 2022-08-25:
Jep, okay so there was no XRDP update.

I checked further on the error an the issue needs to be earlier where programs failed to be linked (whatever this means exactly). There are no earlier logs in the file?

You did a reboot in the meantime?

Can you also check XRDP server logs:
```
journalctl -u xrdp
```
And the `cat ~/.xorgxrdp.10.log` after a restart and fresh connecting attempt:
```sh
systemctl restart xrdp
```

jalsco, 2022-08-25:
I did reboot (a few times ) after the update to 8.7.1
```

[Thu Aug 25] root@batopi: ~$ journalctl -u xrdp
Journal file /var/log/journal/cb0595162f264c0fb73bbfc15e7e6256/system.journal is truncated, ignoring file.
-- Journal begins at Wed 2022-08-24 12:04:24 EDT, ends at Thu 2022-08-25 02:17:01 EDT. --
Aug 24 12:04:27 batopi systemd[1]: Starting xrdp daemon...
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[INFO ] address [0.0.0.0] port [3389] mode 1
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[INFO ] listening to port 3389 on 0.0.0.0
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[INFO ] xrdp_listen_pp done
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[DEBUG] Closed socket 7 (AF_INET6 :: port 3389)
Aug 24 12:04:27 batopi systemd[1]: xrdp.service: Can't open PID file /run/xrdp/xrdp.pid (yet?) after start: >
Aug 24 12:04:28 batopi systemd[1]: Started xrdp daemon.
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] starting xrdp with pid 580
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] address [0.0.0.0] port [3389] mode 1
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] listening to port 3389 on 0.0.0.0
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] xrdp_listen_pp done
Aug 24 12:04:57 batopi xrdp[580]: (580)(281472896995344)[INFO ] Socket 12: AF_INET6 connection received from>
Aug 24 12:04:57 batopi xrdp[580]: (580)(281472896995344)[DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.33>
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[DEBUG] Closed socket 11 (AF_INET6 :: port 3389)
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[INFO ] Using default X.509 certificate: /etc/xrdp/c>
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[INFO ] Using default X.509 key file: /etc/xrdp/key.>
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[ERROR] Cannot read private key file /etc/xrdp/key.p>
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[DEBUG] TLSv1.3 enabled
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[DEBUG] TLSv1.2 enabled
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[DEBUG] Security layer: requested 11, selected 0
Aug 24 12:04:57 batopi xrdp[842]: (842)(281472896995344)[DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.33>
Aug 24 12:04:57 batopi xrdp[580]: (580)(281472896995344)[INFO ] Socket 12: AF_INET6 connection received from>
Aug 24 12:04:57 batopi xrdp[580]: (580)(281472896995344)[DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.33>
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[DEBUG] Closed socket 11 (AF_INET6 :: port 3389)
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[INFO ] Using default X.509 certificate: /etc/xrdp/c>
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[INFO ] Using default X.509 key file: /etc/xrdp/key.>
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[ERROR] Cannot read private key file /etc/xrdp/key.p>
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[DEBUG] TLSv1.3 enabled
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[DEBUG] TLSv1.2 enabled
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[DEBUG] Security layer: requested 0, selected 0
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[INFO ] connected client computer name: DELL-XPS15
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[INFO ] adding channel item name rdpdr chan_id 1004 >
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[INFO ] adding channel item name rdpsnd chan_id 1005>
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[INFO ] adding channel item name cliprdr chan_id 100>
Aug 24 12:04:57 batopi xrdp[843]: (843)(281472896995344)[INFO ] adding channel item name drdynvc chan_id 100>
Aug 24 12:04:58 batopi xrdp[843]: (843)(281472896995344)[INFO ] Non-TLS connection established from ::ffff:1>
Aug 24 12:04:58 batopi xrdp[843]: (843)(281472896995344)[DEBUG] xrdp_0000034b_wm_login_mode_event_00000001
Aug 24 12:04:58 batopi xrdp[843]: (843)(281472896995344)[INFO ] Loading keymap file /etc/xrdp/km-00000409.ini
Aug 24 12:04:58 batopi xrdp[843]: (843)(281472896995344)[WARN ] local keymap file for 0x00000409 found and d>
Aug 24 12:04:58 batopi xrdp[843]: (843)(281472896995344)[DEBUG] xrdp_wm_log_msg: connecting to sesman ip 127>
lines 1-41...skipping...
-- Journal begins at Wed 2022-08-24 12:04:24 EDT, ends at Thu 2022-08-25 02:17:01 EDT. --
Aug 24 12:04:27 batopi systemd[1]: Starting xrdp daemon...
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[INFO ] address [0.0.0.0] port [3389] mode 1
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[INFO ] listening to port 3389 on 0.0.0.0
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[INFO ] xrdp_listen_pp done
Aug 24 12:04:27 batopi xrdp[575]: (575)(281472896995344)[DEBUG] Closed socket 7 (AF_INET6 :: port 3389)
Aug 24 12:04:27 batopi systemd[1]: xrdp.service: Can't open PID file /run/xrdp/xrdp.pid (yet?) after start: Operation not permitted
Aug 24 12:04:28 batopi systemd[1]: Started xrdp daemon.
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] starting xrdp with pid 580
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] address [0.0.0.0] port [3389] mode 1
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] listening to port 3389 on 0.0.0.0
Aug 24 12:04:29 batopi xrdp[580]: (580)(281472896995344)[INFO ] xrdp_listen_pp done
[... cut ...]
[    64.343] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 18 not linked)
[    64.344] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 21 not linked)
[    64.344] (EE) glamor0: GL error: GL_INVALID_OPERATION in glUseProgram(program 24 not linked)
[    64.543] rdpClientConRecv: g_sck_recv failed(returned 0)
[    64.543] rdpClientConRecvMsg: error
[    64.543] rdpClientConCheck: rdpClientConGotData failed
[    64.543] rdpClientConDisconnect:
[    64.544] rdpRemoveClientConFromDev: removing clientCon 0xaaaaf7d38a00
[Thu Aug 25] root@batopi: ~$
```

jalsco, 2022-08-25:
I think last time the solution was just to disable Glamor. Can same approach work again ??  
If that will work, I still some some help with the right steps to achieve  that 🤷‍♂️

MichaIng, 2022-08-25:
Disabling Glamor didn't work last time, if I read it right, but the driver name changes from Buster to Bullseye which needed to be corrected in the XRDP xorg config.

There is indeed an error with linking the "programs" which matches the one reported here: https://github.com/neutrinolabs/xrdp/issues/1503

However, I recognised that your kernel is outdated. Can you check whether there are upgrades available + installed kernel packages:
```sh
apt update
apt full-upgrade
uname -a
dpkg -l | grep '^linux-'
```

jalsco, 2022-08-25:
```
[Thu Aug 25] root@batopi: ~$ apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease
Get:2 https://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:3 https://dl.cloudsmith.io/public/caddy/stable/deb/debian any-version InRelease [7491 B]
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Get:5 https://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Get:6 https://deb.debian.org/debian bullseye-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:7 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-08-25-0808.06-F-2022-08-25-0808.06.pdiff [196 B]
Get:7 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-08-25-0808.06-F-2022-08-25-0808.06.pdiff [196 B]
Fetched 164 kB in 6s (25.3 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
[Thu Aug 25] root@batopi: ~$ apt list --upgradable
Listing... Done
firefox-esr/stable-security 91.13.0esr-1~deb11u1 arm64 [upgradable from: 91.12.0esr-1~deb11u1]
N: There are 2 additional versions. Please use the '-a' switch to see them.
[Thu Aug 25] root@batopi: ~$ apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  firefox-esr
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 54.4 MB of archives.
After this operation, 115 kB disk space will be freed.
Do you want to continue? [Y/n] Y
Get:1 https://deb.debian.org/debian-security bullseye-security/main arm64 firefox-esr arm64 91.13.0esr-1~deb11u1 [54.4 MB]
Fetched 54.4 MB in 9s (6216 kB/s)
(Reading database ... 51163 files and directories currently installed.)
Preparing to unpack .../firefox-esr_91.13.0esr-1~deb11u1_arm64.deb ...
Leaving 'diversion of /usr/bin/firefox to /usr/bin/firefox.real by firefox-esr'
Unpacking firefox-esr (91.13.0esr-1~deb11u1) over (91.12.0esr-1~deb11u1) ...
Setting up firefox-esr (91.13.0esr-1~deb11u1) ...
Processing triggers for mailcap (3.69) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
[Thu Aug 25] root@batopi: ~$ uname -a
Linux batopi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux
[Thu Aug 25] root@batopi: ~$ dpkg -l | grep '^linux-'
[Thu Aug 25] root@batopi: ~$
```

MichaIng, 2022-08-25:
Ah sorry, it's
```sh
dpkg -l | grep 'linux-'
```

jalsco, 2022-08-25:
```
[Thu Aug 25] root@batopi: ~$ dpkg -l | grep 'linux-'
ii  binutils-aarch64-linux-gnu          2.35.2-2                        arm64        GNU binary utilities, for aarch64-linux-gnu target
ii  linux-base                          4.6                             all          Linux image base package
rc  linux-buster-root-current-rockpi-4b 21.02.2                         arm64        Armbian tweaks for buster on rockpi-4b (current branch)
ii  linux-dtb-current-rockchip64        21.08.2                         arm64        Linux DTB, version 5.10.63-rockchip64
ii  linux-image-current-rockchip64      21.08.2                         arm64        Linux kernel, version 5.10.63-rockchip64
ii  linux-libc-dev:arm64                21.08.2                         arm64        Armbian Linux support headers for userspace development
ii  linux-u-boot-rockpi-4b-current      21.08.1                         arm64        Uboot loader 2020.10
[Thu Aug 25] root@batopi: ~$
```

MichaIng, 2022-08-25:
Now I see the Armbian repo is somehow missing. Can you check:
```sh
ls -l /etc/apt/sources.list.d
dpkg -l | grep armbian
```

jalsco, 2022-08-25:
```
[Thu Aug 25] root@batopi: ~$ ls -l /etc/apt/sources.list.d
total 4
-rw-r--r-- 1 root root 319 Nov 30  2021 caddy-stable.list
[Thu Aug 25] root@batopi: ~$ ^C
[Thu Aug 25] root@batopi: ~$ dpkg -l | grep armbian
ii  armbian-firmware                    21.08.3                         all          Linux firmware
ii  htop                                3.1.0-0~armbian20.08.2+1        arm64        interactive processes viewer
[Thu Aug 25] root@batopi: ~$
```

MichaIng, 2022-08-26:
Okay not sure what happened to this list. Let's recreate it:
```sh
curl -sSfL 'https://apt.armbian.com/armbian.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-armbian.gpg --yes
echo 'deb http://apt.armbian.com/ bullseye main' > /etc/apt/sources.list.d/dietpi-armbian.list
apt update
apt list --upgradeable
```
And since I'm not sure about left config files from `linux-buster-root-current-rockpi-4b`, can you also show the output of:
```sh
cat /var/lib/dpkg/info/linux-buster-root-current-rockpi-4b.conffiles | while read -r line; do ls -l "$line"; done
ls -l /etc/kernel/post{inst,rm} /etc/initramfs/post-update.d
```
We need to assure that initramfs updates are done when the kernel is upgraded, which requires config files from `linux-buster-root-current-rockpi-4b`, or ones placed by DietPi, and I'm not sure which ones are present on your system.

jalsco, 2022-08-26:
```
[Fri Aug 26] root@batopi: ~$ curl -sSfL 'https://apt.armbian.com/armbian.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-armbian.gpg --yes
echo 'deb http://apt.armbian.com/ bullseye main' > /etc/apt/sources.list.d/dietpi-armbian.list
[Fri Aug 26] root@batopi: ~$ apt update
Hit:2 https://deb.debian.org/debian bullseye InRelease
Get:3 https://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:4 https://dl.cloudsmith.io/public/caddy/stable/deb/debian any-version InRelease [7491 B]
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Get:1 http://armbian.tnahosting.net/apt bullseye InRelease [22.4 kB]
Get:6 https://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Get:7 https://deb.debian.org/debian bullseye-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:8 http://armbian.tnahosting.net/apt bullseye/main arm64 Packages [346 kB]
Get:9 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-08-26-0812.43-F-2022-08-26-0812.43.pdiff [203 B]
Get:9 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-08-26-0812.43-F-2022-08-26-0812.43.pdiff [203 B]
Fetched 532 kB in 8s (70.4 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
6 packages can be upgraded. Run 'apt list --upgradable' to see them.
[Fri Aug 26] root@batopi: ~$ apt list --upgradeable
Listing... Done
armbian-firmware/bullseye 22.08.0-trunk.0038 all [upgradable from: 21.08.3]
linux-dtb-current-rockchip64/bullseye 22.05.4 arm64 [upgradable from: 21.08.2]
linux-image-current-rockchip64/bullseye 22.05.4 arm64 [upgradable from: 21.08.2]
linux-libc-dev/bullseye 22.05.3 arm64 [upgradable from: 21.08.2]
linux-u-boot-rockpi-4b-current/bullseye 22.05.3 arm64 [upgradable from: 21.08.1]
zlib1g/stable-security 1:1.2.11.dfsg-2+deb11u2 arm64 [upgradable from: 1:1.2.11.dfsg-2+deb11u1]
[Fri Aug 26] root@batopi: ~$ apt update
Hit:2 https://deb.debian.org/debian bullseye InRelease
Get:3 https://dl.cloudsmith.io/public/caddy/stable/deb/debian any-version InRelease [7491 B]
Hit:4 https://deb.debian.org/debian bullseye-updates InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Hit:1 http://armbian.tnahosting.net/apt bullseye InRelease
Fetched 7491 B in 4s (1760 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
6 packages can be upgraded. Run 'apt list --upgradable' to see them.
[Fri Aug 26] root@batopi: ~$ apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  armbian-firmware linux-dtb-current-rockchip64 linux-image-current-rockchip64 linux-libc-dev linux-u-boot-rockpi-4b-current zlib1g
6 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 63.1 MB of archives.
After this operation, 120 MB disk space will be freed.
Do you want to continue? [Y/n] Y
Get:6 https://deb.debian.org/debian-security bullseye-security/main arm64 zlib1g arm64 1:1.2.11.dfsg-2+deb11u2 [88.2 kB]
Get:1 http://mirrors.jevincanders.net/armbian/apt bullseye/main arm64 armbian-firmware all 22.08.0-trunk.0038 [8104 kB]
Get:2 http://armbian.tnahosting.net/apt bullseye/main arm64 linux-dtb-current-rockchip64 arm64 22.05.4 [380 kB]
Get:3 http://mirrors.jevincanders.net/armbian/apt bullseye/main arm64 linux-image-current-rockchip64 arm64 22.05.4 [52.9 MB]
Get:4 http://mirrors.jevincanders.net/armbian/apt bullseye/main arm64 linux-libc-dev arm64 22.05.3 [1182 kB]
Get:5 http://mirrors.jevincanders.net/armbian/apt bullseye/main arm64 linux-u-boot-rockpi-4b-current arm64 22.05.3 [464 kB]
Fetched 63.1 MB in 19s (3245 kB/s)
(Reading database ... 51163 files and directories currently installed.)
Preparing to unpack .../zlib1g_1%3a1.2.11.dfsg-2+deb11u2_arm64.deb ...
Unpacking zlib1g:arm64 (1:1.2.11.dfsg-2+deb11u2) over (1:1.2.11.dfsg-2+deb11u1) ...
Setting up zlib1g:arm64 (1:1.2.11.dfsg-2+deb11u2) ...
(Reading database ... 51163 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_22.08.0-trunk.0038_all.deb ...
Unpacking armbian-firmware (22.08.0-trunk.0038) over (21.08.3) ...
Preparing to unpack .../linux-dtb-current-rockchip64_22.05.4_arm64.deb ...
Unpacking linux-dtb-current-rockchip64 (22.05.4) over (21.08.2) ...
Preparing to unpack .../linux-image-current-rockchip64_22.05.4_arm64.deb ...
Unpacking linux-image-current-rockchip64 (22.05.4) over (21.08.2) ...
Preparing to unpack .../linux-libc-dev_22.05.3_arm64.deb ...
Unpacking linux-libc-dev:arm64 (22.05.3) over (21.08.2) ...
Preparing to unpack .../linux-u-boot-rockpi-4b-current_22.05.3_arm64.deb ...
Unpacking linux-u-boot-rockpi-4b-current (22.05.3) over (21.08.1) ...
Setting up linux-image-current-rockchip64 (22.05.4) ...
update-initramfs: Generating /boot/initrd.img-5.15.52-rockchip64
Setting up linux-u-boot-rockpi-4b-current (22.05.3) ...
Setting up linux-libc-dev:arm64 (22.05.3) ...
Setting up linux-dtb-current-rockchip64 (22.05.4) ...
Setting up armbian-firmware (22.08.0-trunk.0038) ...
Processing triggers for libc-bin (2.31-13+deb11u3) ...
[Fri Aug 26] root@batopi: ~$
```

jalsco, 2022-08-26:
But some problems with the second part you asked for... 

```
[Fri Aug 26] root@batopi: ~$ cat /var/lib/dpkg/info/linux-buster-root-current-rockpi-4b.conffiles | while read -r line; do ls -l "$line"; done
cat: /var/lib/dpkg/info/linux-buster-root-current-rockpi-4b.conffiles: No such file or directory
[Fri Aug 26] root@batopi: ~$
[Fri Aug 26] root@batopi: ~$
[Fri Aug 26] root@batopi: ~$ ls -l /etc/kernel/post{inst,rm} /etc/initramfs/post-update.d
ls: cannot access '/etc/kernel/postinst': No such file or directory
ls: cannot access '/etc/kernel/postrm': No such file or directory
ls: cannot access '/etc/initramfs/post-update.d': No such file or directory
[Fri Aug 26] root@batopi: ~$
```

MichaIng, 2022-08-26:
Ab sorry, the first two are:
```sh
ls -l /etc/kernel/post{inst,rm}.d
```
Please do NOT reboot, the system might not boot up again as you upgraded the kernel already (which I intentionally did not add to suggested commands 😉).

Can you also show:
```sh
dpkg -L linux-buster-root-current-rockpi-4b
ls -l /boot
grep nitrd /boot/boot.cmd
```
To me it looks like the scripts to build the U-Boot style initramfs image is missing.

jalsco, 2022-08-26:
oooops - sorry 😱
and double ooops -😱😱 I did also reboot before I read your message.
But the good news is that the system did some up - whew !

jalsco, 2022-08-26:
```

[Fri Aug 26] root@batopi: ~$ ls -l /etc/kernel/post{inst,rm}.d
/etc/kernel/postinst.d:
total 12
-rwxr-xr-x 1 root root 410 Jun 10  2021 apt-auto-removal
-rwxr-xr-x 1 root root 863 Jan 14  2021 initramfs-tools
-rwxr-xr-x 1 root root 364 Jun  8  2019 unattended-upgrades

/etc/kernel/postrm.d:
total 4
-rwxr-xr-x 1 root root 816 Jan 14  2021 initramfs-tools
[Fri Aug 26] root@batopi: ~$
```

jalsco, 2022-08-26:
```

[Fri Aug 26] root@batopi: ~$ dpkg -L linux-buster-root-current-rockpi-4b
ls -l /boot
grep nitrd /boot/boot.cmd
/etc
/etc/X11
/etc/X11/xorg.conf.d
/usr
/usr/lib
/usr/lib/chromium-browser
total 71680
lrwxrwxrwx 1 root root       26 Aug 26 10:15 Image -> vmlinuz-5.15.52-rockchip64
-rw-r--r-- 1 root root  6384987 Jul  4 03:40 System.map-5.15.52-rockchip64
-rw-r--r-- 1 root root      243 Jul 15  2021 armbianEnv.txt
-rw-r--r-- 1 root root   307322 Mar  8  2021 boot.bmp
-rw-r--r-- 1 root root     3113 Feb  4  2021 boot.cmd
-rw-rw-r-- 1 root root     3185 Feb  4  2021 boot.scr
-rw-r--r-- 1 root root   231602 Jul  4 03:40 config-5.15.52-rockchip64
drwxr-xr-x 4 root root     4096 Aug 24 07:49 dietpi
-rw-rw-r-- 1 root root    18092 Feb 28  2021 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root    13610 Feb 28  2021 dietpi-README.md
-rw-rw-r-- 1 root root    13512 Aug 24 07:49 dietpi.txt
lrwxrwxrwx 1 root root       22 Aug 26 10:15 dtb -> dtb-5.15.52-rockchip64
drwxr-xr-x 6 root root     4096 Aug 26 10:14 dtb-5.15.52-rockchip64
-rw-r--r-- 1 root root 12443085 Aug 24 07:49 initrd.img-5.10.63-rockchip64
-rw-r--r-- 1 root root 11340172 Aug 26 10:15 initrd.img-5.15.52-rockchip64
lrwxrwxrwx 1 root root       26 Oct 19  2021 uInitrd -> uInitrd-5.10.63-rockchip64
-rw-r--r-- 1 root root 12066433 Oct 19  2021 uInitrd-5.10.63-rockchip64
-rw-r--r-- 1 root root 30540288 Jul  4 03:40 vmlinuz-5.15.52-rockchip64
load ${devtype} ${devnum} ${ramdisk_addr_r} ${prefix}uInitrd
[Fri Aug 26] root@batopi: ~$
```

MichaIng, 2022-08-26:
Okay, please do this:
```sh
mkdir -p /etc/kernel/post{inst,rm}.d /etc/initramfs/post-update.d
curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/U-Boot/dietpi-initramfs_cleanup' -o /etc/kernel/postinst.d/dietpi-initramfs_cleanup
chmod +x /etc/kernel/postinst.d/dietpi-initramfs_cleanup
ln -s /etc/kernel/post{inst,rm}.d/dietpi-initramfs_cleanup
curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/U-Boot/99-dietpi-uboot' -o /etc/initramfs/post-update.d/99-dietpi-uboot
chmod +x /etc/initramfs/post-update.d/99-dietpi-uboot
update-initramfs -u
/etc/kernel/postinst.d/dietpi-initramfs_cleanup
apt autopurge linux-buster-root-current-rockpi-4b
```

jalsco, 2022-08-27:
@MichaIng I have done all those steps - really really appreciate 💖your assistance with this ! 🙏🏻
```
[Fri Aug 26] root@batopi: ~$ mkdir -p /etc/kernel/post{inst,rm}.d /etc/initramfs/post-update.d
[Fri Aug 26] root@batopi: ~$ curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/U-Boot/dietpi-initramfs_cleanup' -o /etc/kernel/postinst.d/dietpi-initramfs_cleanup
[Fri Aug 26] root@batopi: ~$ chmod +x /etc/kernel/postinst.d/dietpi-initramfs_cleanup
[Fri Aug 26] root@batopi: ~$ ln -s /etc/kernel/post{inst,rm}.d/dietpi-initramfs_cleanup
[Fri Aug 26] root@batopi: ~$ curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/U-Boot/99-dietpi-uboot' -o /etc/initramfs/post-update.d/99-dietpi-uboot
chmod +x /etc/initramfs/post-update.d/99-dietpi-uboot
[Fri Aug 26] root@batopi: ~$ update-initramfs -u
/etc/kernel/postinst.d/dietpi-initramfs_cleanupupdate-initramfs -u
/etc/kernel/postinst.d/dietpi-initramfs_cleanup
update-initramfs: Generating /boot/initrd.img-5.15.52-rockchip64
update-initramfs: Converting to U-Boot format
/etc/initramfs/post-update.d//99-dietpi-uboot: 3: mkimage: not found
'/boot/uInitrd' -> 'uInitrd-5.15.52-rockchip64'
-bash: /etc/kernel/postinst.d/dietpi-initramfs_cleanupupdate-initramfs: No such file or directory
Removing obsolete initramfs images
removed '/boot/uInitrd-5.10.63-rockchip64'
removed '/boot/initrd.img-5.10.63-rockchip64'
[Fri Aug 26] root@batopi: ~$ apt autopurge linux-buster-root-current-rockpi-4b
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  linux-buster-root-current-rockpi-4b*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 51515 files and directories currently installed.)
Purging configuration files for linux-buster-root-current-rockpi-4b (21.02.2) ...
dpkg: warning: while removing linux-buster-root-current-rockpi-4b, directory '/usr/lib/chromium-browser' not empty so not removed
[Fri Aug 26] root@batopi: ~$
```

jalsco, 2022-08-27:
Next step ?

MichaIng, 2022-08-27:
Ah something was missing:
```sh
apt install u-boot-tools
update-initramfs -u
# And flashing the new U-Boot image
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
```

jalsco, 2022-08-27:
```
[Sat Aug 27] root@batopi: ~$ apt install u-boot-tools
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Recommended packages:
  device-tree-compiler libubootenv-tool
The following NEW packages will be installed:
  u-boot-tools
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 171 kB of archives.
After this operation, 656 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 u-boot-tools arm64 2021.01+dfsg-5 [171 kB]
Fetched 171 kB in 2s (73.8 kB/s)
Selecting previously unselected package u-boot-tools.
(Reading database ... 51514 files and directories currently installed.)
Preparing to unpack .../u-boot-tools_2021.01+dfsg-5_arm64.deb ...
Unpacking u-boot-tools (2021.01+dfsg-5) ...
Setting up u-boot-tools (2021.01+dfsg-5) ...
[Sat Aug 27] root@batopi: ~$ update-initramfs -u
update-initramfs: Generating /boot/initrd.img-5.15.52-rockchip64
update-initramfs: Converting to U-Boot format
Image Name:   uInitrd
Created:      Sat Aug 27 04:19:17 2022
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    11340179 Bytes = 11074.39 KiB = 10.81 MiB
Load Address: 00000000
Entry Point:  00000000
'/boot/uInitrd' -> 'uInitrd-5.15.52-rockchip64'
[Sat Aug 27] root@batopi: ~$ # And flashing the new U-Boot image
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
[Sat Aug 27] root@batopi: ~$
```

jalsco, 2022-08-27:
@MichaIng  did I do those last 2 steps correctly ? I mean the steps after the comment.

MichaIng, 2022-08-27:
Okay last look into `ls -l /boot` but should be all fine now for a reboot, and hopefully with the new kernel the issue is solved.

jalsco, 2022-08-27:
```
[Sat Aug 27] root@batopi: ~$ ls -l /boot
total 58820
lrwxrwxrwx 1 root root       26 Aug 26 10:15 Image -> vmlinuz-5.15.52-rockchip64
-rw-r--r-- 1 root root  6384987 Jul  4 03:40 System.map-5.15.52-rockchip64
-rw-r--r-- 1 root root      243 Jul 15  2021 armbianEnv.txt
-rw-r--r-- 1 root root   307322 Mar  8  2021 boot.bmp
-rw-r--r-- 1 root root     3113 Feb  4  2021 boot.cmd
-rw-rw-r-- 1 root root     3185 Feb  4  2021 boot.scr
-rw-r--r-- 1 root root   231602 Jul  4 03:40 config-5.15.52-rockchip64
drwxr-xr-x 4 root root     4096 Aug 24 07:49 dietpi
-rw-rw-r-- 1 root root    18092 Feb 28  2021 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root    13610 Feb 28  2021 dietpi-README.md
-rw-rw-r-- 1 root root    13512 Aug 24 07:49 dietpi.txt
lrwxrwxrwx 1 root root       22 Aug 26 10:15 dtb -> dtb-5.15.52-rockchip64
drwxr-xr-x 6 root root     4096 Aug 26 10:14 dtb-5.15.52-rockchip64
-rw-r--r-- 1 root root 11340179 Aug 27 04:19 initrd.img-5.15.52-rockchip64
lrwxrwxrwx 1 root root       26 Aug 27 04:19 uInitrd -> uInitrd-5.15.52-rockchip64
-rw-r--r-- 1 root root 11340243 Aug 27 04:19 uInitrd-5.15.52-rockchip64
-rw-r--r-- 1 root root 30540288 Jul  4 03:40 vmlinuz-5.15.52-rockchip64
[Sat Aug 27] root@batopi: ~$
```

Am I ok to reboot now ?

MichaIng, 2022-08-27:
Yes the two steps were right. They do not produce any output. And `/boot` also is fine, ready to reboot.

jalsco, 2022-08-27:
😭
reboot done.
system came back up.
exact same symptoms - RDP screen is still just black.... 😥

jalsco, 2022-08-27:
I think I will find a physical screen to attach - and just confirm that the LXDE desktop is showing when booted while attached to physical.

jalsco, 2022-08-27:
It did not boot into the LXDE desktop when a physical screen attached. Only booted up into terminal mode.   And before the terminal login prompt there were a small bunch of messages - photo below:

[image]

jalsco, 2022-08-27:
```
[Sat Aug 27] root@batopi: ~$ startx
xauth: (stdin):2:  unknown command "87852dfcfe7d5d5e713bf8babb38ae2e"

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux batopi 5.15.52-rockchip64 #22.05.4 SMP PREEMPT Mon Jul 4 09:40:29 CEST 2022 aarch64
Kernel command line: root=UUID=dfa3ab95-cede-4159-9841-4e31bafe853e rootwait rootfstype=ext4  consoleblank=0 loglevel=4 ubootpart=b670ed71-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u
Build Date: 05 August 2022  08:00:36AM
xorg-server 2:1.20.11-1+deb11u2 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Sat Aug 27 05:28:50 2022
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
xf86: found device 0
```
Is there any log file or any test that I do provide that would be useful ??

MichaIng, 2022-08-27:
Please try this to solve the xauth error:
```sh
rm ~/.Xauthority
```
Please also try to reinstall LXDE:
```sh
dietpi-software reinstall 23
```
It's mostly a noop but e.g. disables DPMS and screen blanking again (AFAIK we removed the related condig file when dealing with your issue the last time), which, who knows, might blank XRDP sessions as well.

jalsco, 2022-08-27:
```
[Sat Aug 27] root@batopi: ~$ rm ~/.Xauthority
rm: remove regular file '/root/.Xauthority'? y
[Sat Aug 27] root@batopi: ~$ dietpi-software reinstall 23
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated reinstall

[  OK  ] DietPi-Software | Reinstalling LXDE: ultra lightweight desktop
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | ALSA will be reinstalled
[ INFO ] DietPi-Software | X.Org X Server will be reinstalled
[  OK  ] DietPi-Software | Checking IPv4 network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_us                                                                                                erdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /va                                                                                                r/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mn                                                                                                t/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/do                                                                                                wnloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_                                                                                                userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.4-fpm
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm
[  OK  ] DietPi-Services | stop : mariadb
[ INFO ] DietPi-Software | APT update, please wait...
Hit:2 https://deb.debian.org/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease
Get:4 https://dl.cloudsmith.io/public/caddy/stable/deb/debian any-version InR                                                                                                elease [7491 B]
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:1 http://armbian.chi.auroradev.org/apt bullseye InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Fetched 7491 B in 4s (1723 B/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing ALSA: Advanced Linux Sound Architecture

[ SUB1 ] DietPi-Set_hardware > soundcard (default)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
[  OK  ] DietPi-Set_hardware | rm /etc/asound.conf
[  OK  ] DietPi-Set_hardware | rm /var/lib/alsa/asound.state
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "simple-card" "" "" "" ""
Hardware is initialized using a generic method
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: CONFIG_SOUNDCARD=default
[  OK  ] DietPi-Set_hardware | alsactl -g store
[  OK  ] soundcard default | Completed

 DietPi-Software
─────────────────────────────────────────────────────
[... 110 lines cut ...]
[  OK  ] DietPi-AutoStart | systemctl daemon-reload

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed

[  OK  ] DietPi-Survey | Sending survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : php7.4-fpm
[  OK  ] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : cron
[Sat Aug 27] root@batopi: ~$ update-alternatives --config x-session-manager
There are 3 choices for the alternative x-session-manager (providing /usr/bin/x-session-manager).

  Selection    Path                      Priority   Status
------------------------------------------------------------
* 0            /usr/bin/startlxde         50        auto mode
  1            /usr/bin/lxsession         49        manual mode
  2            /usr/bin/openbox-session   40        manual mode
  3            /usr/bin/startlxde         50        manual mode

Press <enter> to keep the current choice[*], or type selection number:
[Sat Aug 27] root@batopi: ~$
```

jalsco, 2022-08-27:
still not working, i.e:  - RDP to LXDE from my laptop - same symptoms.... just a black screen.  
I tried both before and after a  reboot 😥

jalsco, 2022-08-27:
Hi @MichaIng  - i also reinstalled xrdp using: `dietpi-software reinstall 29`
That also did not fix the problem 😥

jalsco, 2022-08-28:
hi @michalng it is now fixed  😊  - I will provide update with details soon

jalsco, 2022-08-28:
In summary: seems like this problem was (unfortunately) mostly just a False Alarm. And the underlying cause was not Dietpi but was in my RDP client 😱

I am a bit shy and embarrassed to say it 🙏🏻, and also sorry that it caused @MichaIng to waste time to investigate it. But, it was reasonable (although wrong) to suspect DietPi because - (1) it was working immediately before the Update, then after update was not working,  and (2) my problems with #5012 with same components, which was caused by an earlier update. 

Anyway - I thank @MichaIng for investigating - and all those updates and config changes have likely fixed other problems on my system - now and future 🙏🏻. So my system is better than before, thanks to @MichaIng  

How it is working now:  I reported  "attempting to connect by RDP, from windows gives a black screen" - apparently that is a common problem with windows RDP.  Before reporting my problem here, I **DID** try connecting to other windows and linux systems - and they worked as normal (ie no black screen). Only the DietPi gave the black screen...  Anyway - this morning - as per this article: 

Described here: [https://techdirectarchive.com/2022/01/15/how-to-resolve-microsoft-rdp-connection-black-screen/](url)

I toggled and tinkered with  "display size" and **"Persistent bitmap caching"** in my RDP client. And suddenly it magically started working again 😊✨👍   Then - in an effort to find which exact setting caused the black screen - I tried to make the black screen appear again. But - I cannot - no matter which of those settings I change now, it just continues to work... 🤷🏻‍♂️  So that's what I did....

closed 2022-08-28 by jalsco

MichaIng, 2022-09-11:
Great that it works again. Mysterious indeed. Looks like there are server-side desktop resolutions which do not work well on all RDP clients or so. I just checked my RDP settings, and those are both set (max size and bitmap caching), so it seems to be the default. But might be a good idea to add this to our XRDP docs to a troubleshooting section: https://github.com/MichaIng/DietPi-Docs/issues/770