MichaIng/DietPi #5705: Someone share their source.list please.

Wikel, 2022-08-24:
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
```

`cat /etc/apt/source.list`

Joulinar, 2022-08-24:
Could you give some background on your request

Wikel, 2022-08-24:
By mistake I deleted it.

Joulinar, 2022-08-24:
What SBC you are running?

MichaIng, 2022-08-25:
```
deb https://deb.debian.org/debian/ bullseye main contrib non-free
deb https://deb.debian.org/debian/ bullseye-updates main contrib non-free
deb https://deb.debian.org/debian-security/ bullseye-security main contrib non-free
deb https://deb.debian.org/debian/ bullseye-backports main contrib non-free
```

mews-se, 2022-08-27:
just remember its sourceS.list :)

Wikel, 2022-08-27:
Thank you all

closed 2022-08-27 by Wikel