MichaIng/DietPi #5706: Chromium mv command fails when updating from 8.2.0 -> 8.7.1

robbie-hunt, 2022-08-25:
#### Details:
- Date           | Thu Aug 25 12:03:15 IST 2022
- DietPi version | v8.2.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux ctpclient1 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `mv /etc/chromium.d/custom_flags /etc/chromium-browser/customizations/dietpi`
- Exit code      | 1
- Software title | DietPi-Patch
#### Steps to reproduce:
1. DietPi 8.2.0 installed on Raspberry Pi 4B 4GB, Chromium Version 89.0.4389.114 (Developer Build)
2. run `dietpi-update` as root
#### Expected behaviour:
- DietPi should update to latest version
#### Actual behaviour:
- An error of `mv: cannot move '/etc/chromium.d/custom_flags' to '/etc/chromium-browser/customizations/dietpi': No such file or directory` gets thrown up
#### Extra details:

- ...
#### Additional logs:
```
mv: cannot move '/etc/chromium.d/custom_flags' to '/etc/chromium-browser/customizations/dietpi': No such file or directory
```

Joulinar, 2022-08-25:
Hi,

can you share following

```
ls -la /etc/chromium.d/
ls -la /etc/chromium-browser/customizations/
```

robbie-hunt, 2022-08-25:
```
root@ctpclient1:~# ls -laF /etc/chromium.d/
total 28
drwxr-xr-x  2 root root 4096 Aug  5  2021 ./
drwxr-xr-x 87 root root 4096 Aug 11 18:08 ../
-rw-r--r--  1 root root   80 Feb  4  2021 README
-rw-r--r--  1 root root  292 Feb  4  2021 apikeys
-rw-r--r--  1 root root  255 Aug  5  2021 custom_flags
-rw-r--r--  1 root root  609 Feb  4  2021 default-flags
-rw-r--r--  1 root root  299 Feb  4  2021 extensions
```

```
root@ctpclient1:~# ls -laF /etc/chromium-browser/customizations/
ls: cannot access '/etc/chromium-browser/customizations/': No such file or directory
```

robbie-hunt, 2022-08-25:
```
root@ctpclient1:~# ls -laF /etc/chromium-browser/
ls: cannot access '/etc/chromium-browser/': No such file or directory
```

```
root@ctpclient1:~# ls -laF /etc/ | grep "chromium"
drwxr-xr-x  2 root root    4096 Aug  5  2021 chromium/
drwxr-xr-x  2 root root    4096 Aug  5  2021 chromium.d/
```

Chromium was installed via `dietpi-software`

Joulinar, 2022-08-25:
As part of the update, we try to migrate Chromium configuration files. Can you try following to create the folder

`mkdir -p /etc/chromium-browser/customizations/`

Run the update again

MichaIng, 2022-08-25:
I forgot that our first Raspberry Pi 64-bit images indeed were Buster images, where this patch is applied.

Another change since that time is that there is now a `chromium-browser` package for 64-bit (as well for Buster), with full RPi hardware acceleration, while at first there was Debian's `chromium` package only, which is installed on your system.

You might want to migrate this, but be aware that some Chromium settings/userdata might not be migrated due to mismatching versions and packages:
```sh
apt purge chromium
dietpi-software install 113
```

The patch however has been fixed (effective after next release this Sunday), copying the file only if the target directory does exist, indicating that `chromium-browser` is installed indeed: https://github.com/MichaIng/DietPi/commit/08f45ad

closed 2022-08-28 by MichaIng