MichaIng/DietPi #5708: Unclutter not recognised

pastaholic, 2022-08-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
- Distro version | bullseye 1
- Kernel version | Linux DietPi 5.15.56-v7+ #1575 SMP Fri Jul 22 20:28:11 BST 2022 armv7l GNU/Linux
- SBC model | RPi Zero 2 W (armv7l)
- Power supply used | 5V 6A
- SD card used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Chromium
- Was the software title installed freshly or updated/migrated? installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Tried twice, yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Brand new installation of DietPi
2. Install Chromium via dietpi-software
3. Set Chromium to autostart
4. Install unclutter (sudo apt-get install unclutter -y)
5. Run command to set unclutter as per instructions from **[this](https://github.com/MichaIng/DietPi/issues/2575#issue-413358699)** post

#### Expected behaviour

- The cursor should not show when Chromium is running

#### Actual behaviour

- The cursor is still showing when Chromium is running

#### Extra details

- As per instructions from **[this](https://github.com/MichaIng/DietPi/issues/2575#issue-413358699)** post, when running the command

`sudo G_AGI unclutter && echo '/usr/bin/unclutter -idle 0.1 &' > /etc/chromium-browser/customizations/99-dietpi-unclutter`

I get the response

`sudo: G_AGI: command not found`

The same happens when I run the command

`G_CONFIG_INJECT 'xinit[[:blank:]]' 'xinit $FP_CHROMIUM $CHROMIUM_OPTS -- -nocursor' /var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`

I get the response

`sudo: G_CONFIG_INJECT: command not found`

ravenclaw900, 2022-08-27:
Hi,
The `G_*` commands either need to be run as the root user directly, or with `G_SUDO`.

pastaholic, 2022-08-27:
Thank you, that solved the problem.

closed 2022-08-27 by pastaholic