MichaIng/DietPi #5709: "sudo vcgencmd display_power 0" doesn't work

pastaholic, 2022-08-27:
Creating a bug report/issue
Required Information

    DietPi version | G_DIETPI_VERSION_CORE=8
    G_DIETPI_VERSION_SUB=7
    G_DIETPI_VERSION_RC=1
    G_GITBRANCH='master'
    G_GITOWNER='MichaIng'
    G_LIVE_PATCH_STATUS[0]='not applicable'
    Distro version | bullseye 1
    Kernel version | Linux DietPi 5.15.56-v7+ 

    Aria2 configuration cleared after reboot #1575 SMP Fri Jul 22 20:28:11 BST 2022 armv7l GNU/Linux
    SBC model | RPi Zero 2 W (armv7l)
    Power supply used | 5V 6A
    SD card used | SanDisk ultra

Additional Information (if applicable)

    Software title | Chromium
    Was the software title installed freshly or updated/migrated? installed freshly
    Can this issue be replicated on a fresh installation of DietPi? Tried twice, yes

    Bug report ID | echo $G_HW_UUID

Steps to reproduce

    Brand new installation of DietPi
    Run command `sudo vcgencmd display_power 0`

Expected behaviour

    The screen should turn off

Actual behaviour

    The screen stays on, and the response is `display_power=1`

pastaholic, 2022-08-29:
Anyone?

pastaholic, 2022-08-29:
Ok, sorted.For anyone else looking, you need to downgrade the kernel.

sudo apt install rpi-update -y

sudo BRANCH=stable rpi-update b976c54917e240630c05a9b383010f1492bc61b4

Answer yes. This will take some time. When done, reboot and try again.

closed 2022-08-29 by pastaholic

Joulinar, 2022-08-30:
Better to you report that issue to RPi developer to fix this issue in kernel version. Running an old kernel is not something I would recommend.

MichaIng, 2022-08-30:
Reported here: https://github.com/raspberrypi/firmware/issues/1740

@pastaholic 
This is a HDMI screen? And do you use legacy graphics, fake KMS or full KMS?
```sh
grep 'dtoverlay=vc4' /boot/config.txt
```

pastaholic, 2022-08-30:
This is a machine that does only one thing: it connects to a website and displays a page.
There are no incoming connections to it at all apart from SSH from my personal computer.
It is a HDMI screen indeed, from an old laptop via a controller.
It has been working flawlessly for 2 years.
The command `sudo vcgencmd display_power 0` has also been working just as long.
Then it just stopped working.
I downgraded the kernel and now it works fine again.

Here's my config.txt:

> # Docs: https://www.raspberrypi.com/documentation/computers/config_txt.html
> # Overlays: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README
> 
> #-------Display---------
> # Max allocated framebuffers: Set to "0" in headless mode to reduce memory usage
> # - Defaults to "2" on RPi4 and "1" on earlier RPi models
> #max_framebuffers=0
> 
> # If you get no picture, set the following to "1" to apply most compatible HDMI settings.
> #hdmi_safe=1
> 
> # Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
> # - Ranges from "0" to "11", use values above "7" only if required, e.g. with very long HDMI cable.
> # - Default on first RPi1 A/B is "2", else "5", on RPi4 this setting is ignored.
> #config_hdmi_boost=5
> 
> # Uncomment if HDMI display is not detected and composite is being outputted.
> #hdmi_force_hotplug=1
> 
> # Uncomment to disable HDMI even if plugged, e.g. to force composite output.
> #hdmi_ignore_hotplug=1
> 
> # Uncomment to force a console size. By default it will be display's size minus overscan.
> #framebuffer_width=1280
> #framebuffer_height=720
> 
> # Uncomment to enable SDTV/composite output on RPi4. This has no effect on previous RPi models.
> #enable_tvout=1
> # SDTV mode
> #sdtv_mode=0
> 
> # Uncomment to force a specific HDMI mode (this will force VGA).
> #hdmi_group=1
> #hdmi_mode=1
> 
> # Uncomment to force an HDMI mode rather than DVI. This enables HDMI audio in DMT modes.
> #hdmi_drive=2
> 
> # Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
> # With default value "0", the display shows a blank screen instead, but will not go into standby.
> # NB: Some legacy OpenMAX applications (OMXPlayer) cannot wake screens from real standby.
> hdmi_blanking=1
> 
> # Set to "1" if your display has a black border of unused pixels visible.
> disable_overscan=1
> 
> # Uncomment the following to adjust overscan.
> # Use positive numbers if console goes off screen, and negative if there is too much border.
> #overscan_left=16
> #overscan_right=16
> #overscan_top=16
> #overscan_bottom=16
> 
> # Rotation
> #display_hdmi_rotate=0
> #lcd_rotate=0
> 
> #-------RPi camera module-------
> #start_x=1
> #disable_camera_led=1
> 
> #-------GPU memory splits-------
> gpu_mem_256=64
> gpu_mem_512=64
> gpu_mem_1024=64
> 
> #-------Boot splash screen------
> disable_splash=1
> 
> #-------Onboard sound-----------
> dtparam=audio=on
> 
> #-------I2C-------------
> #dtparam=i2c_arm=off
> #dtparam=i2c_arm_baudrate=100000
> 
> #-------SPI-------------
> #dtparam=spi=off
> 
> #-------Serial/UART-----
> # NB: "enable_uart=1" will enforce "core_freq=250" on RPi models with onboard WiFi.
> enable_uart=0
> 
> #-------SD card HPD-----
> # Comment to enable SD card hot-plug detection, while booting via USB or network.
> # NB: This causes constant CPU load and kernel errors when no SD card is inserted.
> dtparam=sd_poll_once
> 
> #-------Overclock-------
> temp_limit=65
> initial_turbo=20
> 
> #over_voltage=0
> #arm_freq=1000
> #core_freq=400
> #sdram_freq=450
> 
> #over_voltage_min=0
> #arm_freq_min=300
> #core_freq_min=250
> #sdram_freq_min=400
> dtoverlay=vc4-kms-v3d
>

MichaIng, 2022-09-11:
As per above linked issue on RPi repo, it is intended that it doesn't work anymore with `dtoverlay=vc4-kms-v3d` set, i.e. modern DRM/KMS display stack. But without it, some software, including Chromium, AFAIK, doesn't work anymore at all.

`tvservice -o` doesn't work either, right?

The alternatives work from within X sessions only, not from plain console 🤔.