MichaIng/DietPi #5710: DietPi Update failed

Trail-Seeker, 2022-08-27:
#### Details:
- Date           | Sat Aug 27 16:52:45 BST 2022
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:2 https://packages.sury.org/php stretch InRelease
Err:3 https://packages.sury.org/php stretch Release
  403  Forbidden
Hit:4 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
E: The repository 'https://packages.sury.org/php stretch Release' does no longer have a Release file.
```

Joulinar, 2022-08-27:
You are running very old Debian Stretch and PHP repository has changed the keys. it's similar to this old issue https://github.com/MichaIng/DietPi/issues/4219

Trail-Seeker, 2022-08-27:
Tried the proposed solution, DietPi Update still fails

#### Details:
- Date           | Sat Aug 27 22:26:45 BST 2022
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:2 https://packages.sury.org/php stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Ign:4 https://packages.sury.org/php stretch Release
Ign:5 https://packages.sury.org/php stretch/main all Packages
Ign:6 https://packages.sury.org/php stretch/main armhf Packages
Ign:5 https://packages.sury.org/php stretch/main all Packages
Ign:6 https://packages.sury.org/php stretch/main armhf Packages
Ign:5 https://packages.sury.org/php stretch/main all Packages
Ign:6 https://packages.sury.org/php stretch/main armhf Packages
Ign:5 https://packages.sury.org/php stretch/main all Packages
Ign:6 https://packages.sury.org/php stretch/main armhf Packages
Ign:5 https://packages.sury.org/php stretch/main all Packages
Ign:6 https://packages.sury.org/php stretch/main armhf Packages
Ign:5 https://packages.sury.org/php stretch/main all Packages
Err:6 https://packages.sury.org/php stretch/main armhf Packages
  403  Forbidden
Reading package lists...
W: The repository 'https://packages.sury.org/php stretch Release' does not have a Release file.
E: Failed to fetch https://packages.sury.org/php/dists/stretch/main/binary-armhf/Packages  403  Forbidden
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2022-08-27:
Ah I see, guys hosting `sury` PHP repository have removed support for Debian Stretch. Means you would need to remove apt package configuration for it.

```
rm -f /etc/apt/sources.list.d/dietpi-php.list
rm -f /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm -f /etc/apt/preferences.d/dietpi-php
```
If you find time, I highly recommed to have a look to our blog articl, why it would make sense to upgrade your system to a newer Debian version https://dietpi.com/blog/?p=1001

Trail-Seeker, 2022-08-28:
Thanks a lot, 

update has been installed without any issues now.

Joulinar, 2022-08-28:
Ok I'm going to close the issue. Keep in mind to perform a destro upgrade as we skipped support for Debian Stretch as well and you will not receive further updates.

closed 2022-08-28 by Joulinar