MichaIng/DietPi #5713: upgraded to Debian 11 "bullseye" and dashboard no longer works

tschrist, 2022-08-28:
I used this page to update by dietpi setup to Bullseye [(https://dietpi.com/blog/?p=811 )](url)

now I get 

This site can’t be reached 192.168.x.xxx refused to connect.
Try:

Checking the connection
[Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)
ERR_CONNECTION_REFUSED

Joulinar, 2022-08-28:
It would have been helpful if you would have filled the bug report template.

Can you describe what application you like to reach? As well, can you share status of your 

`dietpi-services status`

tschrist, 2022-08-28:
trying to get to pihole dashboard in my case: http://192.168.1.251/admin

root@pihole-vm:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[FAILED] DietPi-Services | ● lighttpd.service - Lighttpd Daemon
     Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2022-08-28 06:49:48 BST; 17h ago
    Process: 1019 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=255/EXCEPTION)
        CPU: 83ms

Aug 28 06:49:47 pihole-vm systemd[1]: lighttpd.service: Control process exited, code=exited, status=255/EXCEPTION
Aug 28 06:49:47 pihole-vm systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Aug 28 06:49:47 pihole-vm systemd[1]: Failed to start Lighttpd Daemon.
Aug 28 06:49:48 pihole-vm systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
Aug 28 06:49:48 pihole-vm systemd[1]: Stopped Lighttpd Daemon.
Aug 28 06:49:48 pihole-vm systemd[1]: lighttpd.service: Start request repeated too quickly.
Aug 28 06:49:48 pihole-vm systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Aug 28 06:49:48 pihole-vm systemd[1]: Failed to start Lighttpd Daemon.
[  OK  ] DietPi-Services | docker                active (running) since Sun 2022-08-28 06:49:45 BST; 17h ago
[  OK  ] DietPi-Services | cron                  active (running) since Sun 2022-08-28 06:49:46 BST; 17h ago
[  OK  ] DietPi-Services | dropbear              active (running) since Sun 2022-08-28 06:49:43 BST; 17h ago
[  OK  ] DietPi-Services | pihole-FTL            active (running) since Sun 2022-08-28 06:49:44 BST; 17h ago
[  OK  ] DietPi-Services | unbound               active (running) since Sun 2022-08-28 06:49:44 BST; 17h ago
[ INFO ] DietPi-Services | dietpi-vpn            inactive (dead)
[ INFO ] DietPi-Services | dietpi-cloudshell     inactive (dead)
[  OK  ] DietPi-Services | dietpi-ramlog         active (exited) since Sun 2022-08-28 06:49:43 BST; 17h ago
[  OK  ] DietPi-Services | dietpi-preboot        active (exited) since Sun 2022-08-28 06:49:43 BST; 17h ago
[  OK  ] DietPi-Services | dietpi-postboot       active (exited) since Sun 2022-08-28 06:49:44 BST; 17h ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor   inactive (dead)

tschrist, 2022-08-28:
ah - did a few googles, and found
dietpi-software reinstall 84 

seemed to work after reboot, for some reason bullseye disabled lighttpd - the reinstall worked for me!

Joulinar, 2022-08-28:
Well this is part of the upgrade guide you liked above. Did you followed all steps in chapter 2.2.4?

tschrist, 2022-08-29:
i may have missed this step 

If you use Pi-hole, you should prevent lsof from being auto-removed:
apt-mark manual lsof

Joulinar, 2022-08-29:
But this was not your issue. Looks like you missed the whole software reinstall part 😉

tschrist, 2022-08-29:
I cut/paste everything on the page except for my prior mentioned comment

closed 2022-08-29 by tschrist