MichaIng/DietPi #5717: Error installing Tor Relay software, first option (safest) bridge

Wilford888, 2022-08-29:
### Creating a bug report/issue

#### Required Information

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

bullseye

Linux DietPi 5.19.0-rc1 #1 SMP PREEMPT Fri Jul 15 09:39:18 UTC 2022 aarch64 GNU/Linux

Quartz64 (aarch64)

- Power supply used | 12V 3A Pine64 power supply
- SD card used | NA using 64GB eMMC

#### Additional Information (if applicable)
- Software title | Tor Relay
- Was the software title installed freshly or updated/migrated?  new, fresh install
- Can this issue be replicated on a fresh installation of DietPi? don't know
df49eac0-3081-4fff-903f-61a1a3c625f4

#### Steps to reproduce

1. Trying to set up Tor Relay, used software install menu, set option to bridge, safest for home
2. See below for error message

#### Expected behaviour

- Should install without errors.

#### Actual behaviour

- Unpacking obfs4proxy (0.0.8-1+b6) ...
Setting up obfs4proxy (0.0.8-1+b6) ...
[  OK  ] DietPi-Software | APT install for: obfs4proxy
[FAILED] DietPi-Software | setcap cap_net_bind_service=+ep /usr/bin/obfs4proxy
[FAILED] DietPi-Software | setcap cap_net_bind_service=+ep /usr/bin/obfs4proxy
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: df49eac0-3081-4fff-903f-61a1a3c625f4

#### Extra details

- Let me know if there is something else I should try.  Thanks.

Joulinar, 2022-08-29:
Hi,

which port you have selected for `obfs4` ??

Can you share port actually used pls

```
ss -tulpn | grep LISTEN
```

Wilford888, 2022-09-01:
I wasn't give an opportunity to choose a port before the bug generated the error message.

root@DietPiTR:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      4096       127.0.0.1:9050      0.0.0.0:*    users:(("tor",pid=17550,fd=6))
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=275,fd=4))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=275,fd=5))

Joulinar, 2022-09-01:
I tried to replicate but without success. For me this is working without issues.

You are running an edge kernel  5.19.0-rc1 as far as I can see. Maybe this has a negative effect. Could you try to execute the command manually to check for the exact error message

```
cap_net_bind_service=+ep /usr/bin/obfs4proxy
```

Wilford888, 2022-09-01:
root@DietPiTR:~# cap_net_bind_service=+ep /usr/bin/obfs4proxy
2022/09/01 19:31:12 [ERROR]: obfs4proxy - must be run as a managed transport

Joulinar, 2022-09-01:
Sorry my fault, Can you try following

```
setcap cap_net_bind_service=+ep /usr/bin/obfs4proxy
```

Wilford888, 2022-09-02:
root@DietPiTR:~# setcap cap_net_bind_service=+ep /usr/bin/obfs4proxy
Failed to set capabilities on file `/usr/bin/obfs4proxy' (Operation not supported)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file

MichaIng, 2022-09-10:
Please show the output of:
```sh
lsattr /usr/bin/obfs4proxy
```

MichaIng, 2022-11-09:
I'll mark this as closed. Feel free to reopen if issue persists.

closed 2022-11-09 by MichaIng