MichaIng/DietPi #5726: dietpi-config: Add an option to enable audio output over HDMI and external / HAT DAC under Audio Options

paradix, 2022-08-31:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- On RPI4 I would like to have the option to set audio output over DAC (usb or HAT) as default and enable audio output over HDMI. In my case it is the IQAudioDAC. I run kodi and ifi tidal connect client (through docker image). Docker can in the RPI 64bit ARMv8 version output audio only to the default sound card. There is no option to set USB/HAT DAC as default and enable HDMI output.

- Choosing rpi-bcm2835-auto in audio options enables all audio outputs and sets hdmi as default. 
- Choosing iquadio-dac disables HDMI and sets the IQAudioDAC as default

#### Describe the solution you'd like:

- I would like to be able to set the default audio output manually and keep all the other outputs enabled.

#### Describe alternatives you've considered:

- ...

#### Additional context

- ...

MichaIng, 2022-09-11:
Many thanks for your request.

I thought about this already, which would grant more flexibility:
- Have a checkbox menu to toggle driver/firmware support for multiple outputs.
- The "default" audio output selection of course assures that the required driver is enabled, but does not automatically disable all others.

This could be also combined into a single menu, i.e. a checkbox menu only with an additional selection of the default output if multiple items were selected.

Relevant for RPi only, where DAC and onboard audio drivers can be toggled via dtoverlays.

What I'm not sure about is whether multiple DAC drivers/dtoverlays can be enabled concurrently. Otherwise we'd need to somehow limit it to only 1 USB DAC and 1 HAT or so, which makes it more complicated.

As a short-term solution, you know how do enable the DAC manually additionally to onboard audio, right?

paradix, 2022-09-11:
Great to hear that :)
I have managed to achieve my goal by manually editing the config.txt and setting previously the iqaudio DAC as default audio output. That works for now for my needs. 
```
enable_uart=0
# Added by DietPi:
#dtparam=audio=on
dtoverlay=rpivid-v4l2
dtoverlay=disable-wifi
dtoverlay=vc4-kms-v3d,cma-512
#hdmi_drive=2
#hdmi_drive:1=2
dtoverlay=iqaudio-dac
```

MDAR, 2023-02-17:
@MichaIng 

Apologies if this is the wrong place to put this, I just thought it was a relevant place.

I can't remember having to make any changes in order to get audio out of the HDMI port on an Odroid C4 (when using Chromium in Xinit)

Today I put a fresh Bullseye config into my main production unit and it doesn't seem to want to output audio from the HDMI.

Before I go down a rabbit hole, have you changed something ?

MichaIng, 2023-02-18:
@MDAR 
> Apologies if this is the wrong place to put this

You problem is very different from this feature request regarding Raspberry Pi, please open a new issue.