MichaIng/DietPi #5730: DietPi-Software | Amiberry: Mouse not working with RPi4

lilltonka, 2022-09-04:
### Creating a bug report/issue

#### Required Information

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.15.61-v8+ # 1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Raspberry Pi 4 Power Supply / 5.1V / 3A / USB-C
- SD card used | SanDisk Corp. Ultra Fit

#### Additional Information (if applicable)
- Software title | Amiberry
- Fresh install and uninstalled several times to reproduce.
- _Can this issue be replicated on a fresh installation of DietPi?_ I'm not sure. This is the version I got.

#### Steps to reproduce

1. Install Amiberry and boot it up. I went in from terminal.
2. Set everything up for running a disk (.adf) with mouse needed.
3. Try to move mouse around.

#### Expected behaviour
- The mouse should move around as intended.

#### Actual behaviour
- Mouse is stuck in the left corner of the screen when running a game/program. 

#### Extra details
- Tried using both USB-connected mouse, and the Logitech Wireless one. Both works in Amiberry GUI but not when game/program starts.

MichaIng, 2022-09-06:
Many thanks for your report.

Just to rule it out, can you try to reinstall Amiberry:
```sh
dietpi-software reinstall 108
```
I just cleared the cache on Cloudflare to assure it's the latest build. But I guess it is more likely to be due to the latest SDL releases. I'll test as well.

lilltonka, 2022-09-06:
Hi [MichaIng](https://github.com/MichaIng), and thank's for your reply!

I tried reinstalling it just now, and unfortunately it did no change to the scenario.

I can also add that just to rule out any problems with the programs/games; I tried the same .adf-files in WinUAE, and the mouse moved as intended.

mc1971, 2022-09-07:
I copied LIBSDL2 ver 2.0.16 to the libs folder (had this version handy, didn't try anything more recent) and the mouse now operates correctly.

MichaIng, 2022-09-07:
Okay, so we need to revert the latest SDL2 bump. I'll trigger a new build and give you a ping when it's uploaded and ready to be updated.

lilltonka, 2022-09-07:
This is great news! Thanks to the both of you!

MichaIng, 2022-09-08:
Please test this:
```sh
cd /tmp
curl -O 'https://dietpi.com/downloads/binaries/bullseye/testing/amiberry_rpi4-64-dmx.deb'
dpkg -i amiberry_rpi4-64-dmx.deb
rm amiberry_rpi4-64-dmx.deb
```

lilltonka, 2022-09-08:
Did that, and it worked like a charm! Mouse is responding to movement and left- and right-click.

corasaniti, 2022-09-10:
Do you recommend waiting for the 8.9 release of Dietpi or applying the following workaround?

```
cd / tmp
curl -O 'https://dietpi.com/downloads/binaries/bullseye/testing/amiberry_rpi4-64-dmx.deb'
dpkg -i amiberry_rpi4-64-dmx.deb
rm amiberry_rpi4-64-dmx.deb
```

If I apply the above solution, when Dietpi8.9 is released and I update will I have problems?
Thanks in advance

Joulinar, 2022-09-10:
Usually a DietPi update will not influence amiberry

corasaniti, 2022-09-10:
Sorry but has the file been deleted from the repository?
`https://dietpi.com/downloads/binaries/bullseye/testing/amiberry_rpi4-64-dmx.deb`
thank you

Joulinar, 2022-09-10:
Can we stick to one platform? doesn't make sense to chat in GitHub as well as Facebook for same topic in parallel.

MichaIng, 2022-09-10:
I moved the new packages into stable place already. It's save to apply the update now, but in your case it's the 32-bit package, right?
```
cd /tmp
curl -O 'https://dietpi.com/downloads/binaries/bullseye/amiberry_rpi4.deb'
dpkg -i amiberry_rpi4.deb
rm amiberry_rpi4.deb
```

MichaIng, 2022-09-10:
All packages updated and Cloudflare cache cleared, so it can now be safely fixed with a reinstall:
```sh
dietpi-software reinstall 108
```

MichaIng, 2022-09-15:
Fix merged and packages rebuilt: #5733
Next Amiberry release has a fix included, so that we can bump libSDL2 to v0.24.0 again together with Amiberry v5.4.

closed 2022-09-15 by MichaIng

dlasher, 2024-02-01:
Appears to be a current issue with v9.0.2 oDROID N2+ version. No mouse in startup screen/menus, but mouse works fine once emulation is started.

```
ii  libsdl2-2.0-0:arm64              2.26.5+dfsg-1                  arm64        Simple DirectMedia Layer
ii  libsdl2-dev:arm64                2.26.5+dfsg-1                  arm64        Simple DirectMedia Layer development files
ii  libsdl2-gfx-1.0-0:arm64          1.0.4+dfsg-4+b1                arm64        drawing and graphical effects extension for SDL2
ii  libsdl2-gfx-dev:arm64            1.0.4+dfsg-4+b1                arm64        development files for SDL2_gfx
ii  libsdl2-image-2.0-0:arm64        2.6.3+dfsg-1                   arm64        Image loading library for Simple DirectMedia Layer 2, libraries
ii  libsdl2-image-dev:arm64          2.6.3+dfsg-1                   arm64        Image loading library for Simple DirectMedia Layer 2, development files
ii  libsdl2-ttf-2.0-0:arm64          2.20.1+dfsg-2                  arm64        TrueType Font library for Simple DirectMedia Layer 2, libraries
ii  libsdl2-ttf-dev:arm64            2.20.1+dfsg-2                  arm64        TrueType Font library for Simple DirectMedia Layer 2, development files
```

running v5.6.5 of Amiberry
```
ii  amiberry                         5.6.5-dietpi1                  arm64        Optimized Amiga emulator for the Raspberry Pi and other ARM boards
```