MichaIng/DietPi #5731: DietPi-Update failing at Checking IPv6 network connectivity

theAkito, 2022-09-05:
#### Details:
- Date           | Mon Sep  5 12:40:21 CEST 2022
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | Generic Device (armv7l) (ID=22)
- Kernel version | `Linux hostname 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `getent hosts one.one.one.one`
- Exit code      | 2
- Software title | DietPi-Update
#### Steps to reproduce:

1. Run `sysctl -w net.ipv6.conf.all.disable_ipv6=1`, to disable IP connectivity.
2. Run `sysctl -w net.ipv6.conf.all.disable_ipv6=0`, to enable IP connectivity.
3. Legacy IP connectivity is kept intact.
4. Run `dietpi-update`.
5. Get stuck at
```
[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking IPv6 network connectivity                                                                  [FAILED] DietPi-Update | Checking DNS resolver
 - Command: getent hosts one.one.one.one
```
#### Expected behaviour:

Running `dietpi-update` should successfully continue and finish.

#### Actual behaviour:

Running `dietpi-update` as described above does not work, because of a faulty IP network connectivity test.

#### Extra details:

"Just don't disable IP!". Well, no! That's wrong.

I have run `dietpi-update` on two basically almost the same Raspberry Pi systems. No significant differences. The biggest obvious difference is that one turned of IP temporarily, whereas the other one did not.

However, it's important to note, that I tested running `getent hosts one.one.one.one` on both systems, **and it did not work on either**. (`getent hosts 1.1.1.1` worked on both.)
So, if `dietpi-update` worked on the first one, then it should also work on the one that had IP temporarily disabled.

Additionally, both were on the same DietPi version, before the first one successfully upgraded to the currently most recent one, i.e. `8.8.1`.

Not sure if this is in any way related to issues like https://github.com/MichaIng/DietPi/issues/4742, for example.

That said, you should also note that I can `curl -6` anything perfectly fine! **IP is working**!

#### Additional logs:
```
[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking IPv6 network connectivity                                                                  [FAILED] DietPi-Update | Checking DNS resolver
 - Command: getent hosts one.one.one.one
```

Joulinar, 2022-09-05:
Just to avoid a misunderstanding, we are talking about IPv6 only. IPv4 is working 😉 

could you activate IPv6 and check following

```
/boot/dietpi/func/dietpi-set_hardware enableipv6 enable
getent hosts one.one.one.one
ip a
cat /etc/resolv.conf
```

theAkito, 2022-09-05:
> Just to avoid a misunderstanding, we are talking about IPv6 only.

Yes, we are talking about IP and not about Legacy IP.

> IPv4 is working 😉

IP is working too, just as Legacy IP does. It's just that the `dietpi-update` script is doing a faulty check.

`curl -6`ing any address is working perfectly fine. Just `dietpi-update` complains about a missing connection.

> could you activate IPv6 and check following
> 
> ```
> /boot/dietpi/func/dietpi-set_hardware enableipv6 enable
> getent hosts one.one.one.one
> ip a
> cat /etc/resolv.conf
> ```

The first line resulted in `Desired setting in /etc/hosts was already set` three times and `Desired setting in /boot/dietpi.txt was already set` one time. If this log is correct, everything was already enabled.

That was expected, as `curl -6`ing any address is working perfectly fine.

The second line yielded the same result. No output, with exit code `2`.

The last line did not show any changes. It's the same as on the other host, where `dietpi-update` was successfully applied.

Running `getent hosts one.one.one.one` on the host, which successfully was updated, does yield the same result. It does not succeed. Yet, `dietpi-update` was successful on that other host. It's just this host, where this suddenly becomes a problem.

Joulinar, 2022-09-05:
The check is not a dedicated to IPv6, it's a general test to verify DNS resolution, doesn't matter what setting the user has (IPv4 only or IPv6). On a IPv4 only system, your `curl -6` will fail. If `getent` is failing, something is not working on DNS resolution. Therefore, it would be highly appreciated if you could simply post the full output of what I have ask for.

theAkito, 2022-09-05:
> On a IPv4 only system, your `curl -6` will fail.

As I pointed out multiple times, it does not fail. Everything is working fine in regards to IP.

> If `getent` is failing, something is not working on DNS resolution.

If it can resolve any domain without issues, then DNS resolution is fine. If the `getent` command fails, then something is wrong with the command. Else it wouldn't fail, because the system can obviously connect to any domain, including running `apt upgrade`, etc...

I also already pointed out multiple times, that this very same command does **not** work either on the system that successfully used `dietpi-update`, so it is updated now. This proves the point that `getent` is just the command that is failing. The DNS resolution is fine, else the other host, where `dietpi-update` worked would be just as broken.

To finally debunk the myth of not working DNS resolution:

1. There is Raspberry Pi Uno.
2. There is Raspberry Pi Dos.
3. Running `getent hosts one.one.one.one` on Raspberry Pi Uno does **NOT** work.
4. Running `getent hosts one.one.one.one` on Raspberry Pi Dos does **NOT** work.
5. Running `dietpi-update` on Raspberry Pi Uno **DID** work.
6. Running `dietpi-update` on Raspberry Pi Dos **DID NOT** work.
7. DNS resolution is provably working on Raspberry Pi Uno.
8. DNS resolution is provably working on Raspberry Pi Dos.
9. Therefore, the logical conclusion is, that the `getent hosts one.one.one.one` command is failing. Not DNS resolution, IP resolution or anything related to the network, specific to this situation.

I hope it's clear now, that the DNS resolution is fine and the command `getent hosts one.one.one.one` itself is failing.

> it would be highly appreciated if you could simply post the full output of what I have ask for.

You never asked for the output. All you did was asking me to "check". So I did.

Anyway, this is the output.

```
/boot/dietpi/func/dietpi-set_hardware enableipv6 enable
getent hosts one.one.one.one
ip a
cat /etc/resolv.conf

 DietPi-Set_hardware
─────────────────────────────────────────────────────
 Mode: enableipv6 (enable)

[  OK  ] DietPi-Set_hardware | Desired setting in /etc/hosts was already set: *** localhost ip6-localhost ip6-loopback
[  OK  ] DietPi-Set_hardware | Desired setting in /etc/hosts was already set: ****** ip6-allnodes
[  OK  ] DietPi-Set_hardware | Desired setting in /etc/hosts was already set: ****** ip6-allrouters
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: CONFIG_ENABLE_IPV6=1
[  OK  ] enableipv6 enable | Completed

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
  link/ether *****************************************************************************************************
  inet ********************************************************************* scope global dynamic eth0
    valid_lft ******************** preferred_lft ********************
  inet6  ******************** scope global dynamic mngtmpaddr
    valid_lft ******************** preferred_lft ********************
  inet6 ******************** scope link
    valid_lft forever preferred_lft forever
.............
Various irrelevant & unrelated networks, like for example Docker's....
.............

# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#     DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
# 127.0.0.53 is the systemd-resolved stub resolver.
# run "resolvectl status" to see details about the actual nameservers.

nameserver ***.***.***.***
search *************.*****
```

Joulinar, 2022-09-05:
Your system is unique. At the moment you are the only one having such issues. Therefore it's important to understand what the issue is.

Somehow I'm missing the output of `getent hosts one.one.one.one`. Doesn't seems to be in your output. Could you try it again pls. Does it have any output?

theAkito, 2022-09-05:
> Your system is unique. At the moment you are the only one having such issues. Therefore it's important to understand what the issue is.

It's only one Raspberry Pi of two. I would've discarded this whole issue, if it was just one, because I would've blamed the very old DietPi version for it. However, the first DietPi system updated fine from that old version, which is why I assumed something more out of the ordinary.

> Somehow I'm missing the output of `getent hosts one.one.one.one`. Doesn't seems to be in your output. Could you try it again pls. Does it have any output?

> The second line yielded the same result. No output, with exit code `2`.

###### [Source](https://github.com/MichaIng/DietPi/issues/5731#issuecomment-1237053981)

It does not have any output. Never had. I just asked for a `echo $?` after it, which is why I know its exit code.

Joulinar, 2022-09-05:
Out of curiosity, could you verify following? There would need to be an output. At least on my system I have.

`getent hosts google.com`

theAkito, 2022-09-05:
> getent hosts google.com

```
getent hosts google.com
2607:f8b0:4006:808::200e google.com
```

Joulinar, 2022-09-05:
Means it's failing on Cloudflare DNS only. Correct? You could adjust the host to be checked in `/boot/dietpi.txt`. You could adjust it to something like this

```
CONFIG_CHECK_CONNECTION_IP=9.9.9.9
CONFIG_CHECK_DNS_DOMAIN=dns9.quad9.net
```

theAkito, 2022-09-05:
> ou could adjust the host to be checked in `/boot/dietpi.txt`. Change it to something like this
> 
> ```
> CONFIG_CHECK_CONNECTION_IP=9.9.9.9
> CONFIG_CHECK_DNS_DOMAIN=dns9.quad9.net
> ```

```
getent hosts dns9.quad9.net
2620:fe::fe:9   dns9.quad9.net
2620:fe::9      dns9.quad9.net
```

Works perfectly, including the following execution of `dietpi-update`.

Though, `dietpi-update` worked fine on Raspberry Pi Uno, which has the same `getent` related issue and is on the same network, as Raspberry Pi Dos (the one we are talking about here).
Why did it work once?
~~Well, if this is a "temporary" issue, we cannot exclude that it worked for a minute when I ran `dietpi-update` on the Uno.~~
Forget about it. The Uno has still the same exact issue, but `dietpi-update` works there!

For my situation, this problem is now solved, but `dietpi-update` definitely still has an issue.

Joulinar, 2022-09-05:
@MichaIng 
I leave it to you to continue discussion.

MichaIng, 2022-09-06:
So it's just the hostname `one.one.one.one` which cannot be resolved for some reason. We received reports about issues with the Cloudflare DNS in the past and hence switched to Quad9 IP and host by default. Still interesting that pinging `1.1.1.1` worked.

> Forget about it. The Uno has still the same exact issue, but `dietpi-update` works there!

You mean that system cannot resolve `one.one.one.one` either, but the update works since a different hostname is used for the DNS check? Makes sense, likely it is a limitation of your ISP or router-wise firewall.

MichaIng, 2022-09-07:
Okay I mark this as closed then. An option to change that IP/hostname right from the error-handler is planned for next release, which makes working around such issues easier.

closed 2022-09-07 by MichaIng