MichaIng/DietPi #5732: how do I build a DietPI-binary in CI?

coderofsalvation, 2022-09-08:
Hi, I would like to build an AppImage which runs on DietPI.
To build AppImage-binaries, I use CI (like travis/appveyor) for this.
Basically I need to clone a repo, run cmake, and that's it.
Is there a CI-file you can point me to, which I can use as a reference?
Should I use qemu or something?

kind regards,

(and thanks for making/maintaining DietPI!)

ravenclaw900, 2022-09-08:
Hi,
DietPi is based on Debian, so any binary built for that (or just Linux in general) should work. However, depending on the device you're running it on, you may have to build for the arm architecture instead of x86.

MichaIng, 2022-09-08:
You can have a look into our CI build jobs and scripts, e.g. here for vaultwarden: https://github.com/MichaIng/DietPi/blob/master/.build/software/vaultwarden/container_build.bash

We host a set of container images for all supportes architectures and Debian versions, which we download, mount as loop device and "boot" via systemd-nspawn, in case QEMU emulated on the (x86_64) GitHub actions runner.

But a matching Debian Docker container would do as well, of course.

coderofsalvation, 2022-09-11:
thank you for that shellscript, I think it'll be a nice starting point (adding some dependencies etc) to build something inside dietpi.

closed 2022-11-24 by MichaIng