MichaIng/DietPi #5734: Raspberry Pi | Replace raspberrypi-sys-mods with own package

MichaIng, 2022-09-08:
The recent release added `raspi-config` as dependency with a bunch of child dependencies and scripts, which are all not needed on DietPi. Many other things are and were for Raspberry Pi OS first boots, only a small number of GPIO/I2C/SPI related configs and scripts are reasonable on DietPi as well: https://github.com/RPi-Distro/raspberrypi-sys-mods

maartenlangeveld, 2022-09-09:
Hi @MichaIng, noticed that too. Will the next DietPi-update revert those changes or do I have to deinstall those unneeded dependencies manually?

MichaIng, 2022-09-10:
Most likely the next update will pull a package from our server with the same name but only those parts needed on DietPi and a raised epoch version to assure it is not replaced when the one from RPi repo receives an update. The APT autoremoval will then remove the added (and possibly other) dependencies automatically.

Joulinar, 2022-09-11:
for time being, you could mark package on-hold

Hold a package: `sudo apt-mark hold <package-name>`
Remove the hold: `sudo apt-mark unhold <package-name>`
Show all packages on hold: `sudo apt-mark showhold`

maartenlangeveld, 2022-09-11:
Hi @Joulinar, thanks. I'm afraid putting them on hold is a little too late, the packages have already installed.

MichaIng, 2022-09-11:
It even pulls the `initramfs-tools` as dependency 🤔. At least no actually initramfs is build to `/boot` 😄. However, aside of some additional packages/files, it shouldn't cause any issues. No need to go through downgrading and setting on hold IMO.

maartenlangeveld, 2022-09-11:
I already understood that it does not harm, just happy the unneeded packages will be disabled with next Dietpi update. Thanks!

MichaIng, 2022-09-17:
Done, can be tested with current beta 🙂.

closed 2022-09-17 by MichaIng