MichaIng/DietPi #5735: Installing Go packages - write permissions error

tlgs, 2022-09-08:
### Creating a bug report/issue

Unable to install Go packages using `go install` due to user permissions for the module cache.

#### Required Information

DietPi version:

```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

Distro version:

```
bullseye 0
```

Kernel version:

```
Linux r.magpie 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
```

SBC model:

```
RPi 4 Model B (aarch64)
```

Power supply used: ?

SD card used: SanDisk 64GB

#### Additional Information (if applicable)

- Software title: `Go`
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. `sudo dietpi-install 188`
2. `go install github.com/tsenart/vegeta@v12.8.4`

Or any other Go package with installable executables.

#### Expected behaviour

Go should be able to compile the executable and place it in `$GOBIN`.

#### Actual behaviour

The command errors out because the current user has no write permissions to the system set `$GOPATH`:

```console
$ go install github.com/tsenart/vegeta@v12.8.4
go: github.com/tsenart/vegeta@v12.8.4: mkdir /mnt/dietpi_userdata/go/pkg/mod: permission denied
```

#### Extra details

I'm able to install Go packages using the `golang-go` package:

1. `sudo apt install golang-go`
2. `go get -u github.com/tsenart/vegeta`

Note that since `1.17`, `go install` is the [preferred approach](https://go.dev/doc/go-get-install-deprecation) to installing executables. I'm using `go get` above because upstream is still `1.15`:

```
go version go1.15.15 linux/arm64
```

Joulinar, 2022-09-08:
`$GOPATH` has write permission for root user only. Therefore you would need to start install `go install github.com/tsenart/vegeta@v12.8.4` as root user

MichaIng, 2022-09-08:
Many thanks for your report.

> Go should be able to compile the executable and place it in `$GOBIN`.

AFAIK this path is only used for Go language executables, not for compiled binaries, isn't it?

However, indeed we export a `$GOPATH` in `/etc/bashrc.d/go.sh`, which we probably shouldn't do anymore. If I'm not mistaken, by default the user's home directory is used. Of course you can use e.g. `~/.bashrc` to re-set or unset the variable, or delete `/etc/bashrc.d/go.sh`.

At least we should move `/etc/bashrc.d/go.sh` into `/root/.bashrc` or exit within the script when it's not root, so `go install` always works, but behaviour with root user remains as it was before?

tlgs, 2022-09-08:
@Joulinar Yes, that's exactly the issue; it's not standard installer behavior. Also, it seems root doesn't see the installed `go` binary:

```console
$ sudo go install github.com/tsenart/vegeta@v12.8.4
sudo: go: command not found
```

@MichaIng Thank's for the quick and thorough response. My pleasure, using DietPi has been a pleasant experience.

Yeah, setting `GOPATH` for this user would be my next step but I thought this is something that should work out-of-the-box, so I thought it warranted a report here. :)

This should be fairly easy to reproduce, but let me know if you need some more testing.

Joulinar, 2022-09-08:
`sudo` is not using the root env as it has an own one. You could check and adjust using `sudo visudo`

Running the install as real user root will work

MichaIng, 2022-09-08:
`sudo -E go` should work. `/etc/bash.bashrc` is only loaded on interactive shells, and `sudo <command>` does not run the command within an interactive shell. And `/etc/bashrc.d/go.sh` (loaded from `/etc/bash.bashrc`) does not only set `$GOPATH` but also adds `$GOBIN` to `$PATH`.

However, it doesn't make much sense to take care this sudo hassle. I agree that Go for regular users should work the regular way. Actually I think the same should be true for root, i.e. I intend to remove `$GOPATH` from that file.

Joulinar, 2022-09-08:
As workaround, did you tried root user?

tlgs, 2022-09-08:
@Joulinar As part of the machine setup I've restricted root SSH access and locked its password. Afraid that's not an option with my setup, hehe.

Its not a big deal for my setup, just a change I thought DietPi could benefit from. And like [MichaIng](https://github.com/MichaIng) suggested, setting `GOPATH` for the user would be a more appropriate way to deal with this: I'd want a user install, and not have to `sudo $app` to run any executable installed via `go install`.

Joulinar, 2022-09-09:
Did you tried to switch user using `su -`?

MichaIng, 2022-09-10:
Right, `sudo -E` passes the environment but re-sets PATH. So you need an interactive shell, e.g. `sudo -s` (which I prefer over `su`), or use `su -c 'go install abc'` which preserves the PATH as well and does not open an interactive shell.

However, better to edit `/etc/bashrc.d/go.sh`.

I'm currently thinking about how to handle existing installs, when we change this. Users might have given their login user permissions to write to `/mnt/dietpi_userdata/go` 🤔. Probably we should not touch `/etc/bashrc.d/go.sh` when it exists already but only on fresh installs.

MichaIng, 2022-09-10:
PR up: #5738

closed 2022-09-15 by MichaIng